EmuELEC/EmuELEC #784: Per game core override settings

funnymanva, 2021-12-31:
This is more of a feature request than a bug, but would be great if there was a way (even if I have to create the file using a shell) to override the default emulator core per game. As an example, and I know there are many more with regard to N64 and PSP, but I like the Libretro mame2010 core for most of the MAME games as it has additional ones the older mame2003-plus doesn't support. However, I've found that Donkey Kong and Donkey Kong, jr lock up and don't play on mame2010, but do fine on mame2003-plus. It would be great if I could have a file named dkongo.cfg and dkongjr.cfg (rom names in use with cfg as extension) somewhere that tells it to use the libretro/mame2003-plus core but leave the default for all the rest of the games in the MAME system.

miwasp, 2021-12-31:
This feature is already there.
Select the game you want to change the core for in Emulationstation, press "Game options" (normally long press on B or A) and then "Advanced Game options". There you can set a core specific to this game.

funnymanva, 2021-12-31:
Thank you. I've been googling everywhere for this.

closed 2022-01-03 by shantigilbert