EmuELEC/EmuELEC #793: support webdav mount

zhangguanzhang, 2022-01-17:
# BEFORE YOU OPEN AN ISSUE MAKE SURE IT IS NOT ALREADY RESOLVED IN THE WIKI https://github.com/EmuELEC/EmuELEC/wiki

# MAKE SURE YOUR DEVICE IS A [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices)  

# PLEASE DO NOT POST GENERAL HELP QUESTIONS! EITHER USE THE FORUM (https://emuelec.org) OR USE DISCORD FOR THAT (https://discord.gg/cbgtJTu), THIS IS ONLY FOR BUG/ISSUES REPORTING!

# DELETE EVERYTHING ABOVE THIS LINE (INCLUDING THIS LINE) BEFORE SUBMITTING TO MAKE SURE YOU'VE READ THIS!

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

```
EmuELEC:~ # mount
mount             mount.exfat       mount.exfat-fuse  mount.fuse        mount.ntfs        mountpoint
```
I hope there is a `mount.davfs` which could mount the `webdav`
**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

shantigilbert, 2022-01-17:
Sorry I do not know anything about Webdav, and I am personally not interested in adding/fixing it, as I have no use for it. I will leave this issue open in case someone else is willing to tackle this.

shantigilbert, 2022-01-17:
As al alternative, not sure it will work for you but you can try this: https://github.com/EmuELEC/EmuELEC/wiki/ROMS-on-CIFS-SAMBA-shares

zhangguanzhang, 2022-01-18:
> As al alternative, not sure it will work for you but you can try this: https://github.com/EmuELEC/EmuELEC/wiki/ROMS-on-CIFS-SAMBA-shares

I know this. But I want use webdav

zhangguanzhang, 2022-01-21:
@shantigilbert any update? 😢

shantigilbert, 2022-01-21:
As I mentioned before, this is not something I am going to work on, I do not have the knowledge, and it is not something I personally interested in.

zhangguanzhang, 2022-01-21:
😢，Is there had a command like `apt` in EmuELEC?

shantigilbert, 2022-01-21:
You can try  `installentware` then after reboot `opkg list` but be aware I have never used that and I do not know how to use it nor how to resolve issues if any.

zhangguanzhang, 2022-01-21:
> mount -t davfs -o noexec http://192.168.101.1:81 test

woo!，after `installentware` and reboot , I can use `opkg`

```
EmuELEC:~ # opkg list | grep davfs
davfs2 - 1.6.0-1 - Web Distributed Authoring and Versioning (WebDAV), an extension to the HTTP-protocol, allows authoring of resources on a remote web server.davfs2 provides the ability to access such resources like a typical filesystem, allowing for use by standard applications with no built-in support for WebDAV.  davfs2 is designed to fully integrate into the filesystem semantics of Unix-like systems (mount, umount, etc.). davfs2 makes mounting by unprivileged users as easy and secure as possible.  davfs2 does extensive caching to make the file system responsive, to avoid unnecessary network traffic and to prevent data loss, and to cope for slow or unreliable connections.  davfs2 will work with most WebDAV servers needing little or no configuration.
```

and I install the davfs2
```
EmuELEC:~ # opkg install davfs2
Installing davfs2 (1.6.0-1) to root...
Downloading http://bin.entware.net/aarch64-k3.10/davfs2_1.6.0-1_aarch64-3.10.ipk
Installing zlib (1.2.11-3) to root...
Downloading http://bin.entware.net/aarch64-k3.10/zlib_1.2.11-3_aarch64-3.10.ipk
Installing libopenssl (1.1.1l-1) to root...
Downloading http://bin.entware.net/aarch64-k3.10/libopenssl_1.1.1l-1_aarch64-3.10.ipk
Installing libexpat (2.4.1-1) to root...
Downloading http://bin.entware.net/aarch64-k3.10/libexpat_2.4.1-1_aarch64-3.10.ipk
Installing libneon (0.31.2-2) to root...
Downloading http://bin.entware.net/aarch64-k3.10/libneon_0.31.2-2_aarch64-3.10.ipk
Installing libfuse (2.9.9-1a) to root...
Downloading http://bin.entware.net/aarch64-k3.10/libfuse_2.9.9-1a_aarch64-3.10.ipk
Configuring zlib.
Configuring libopenssl.
Configuring libexpat.
Configuring libneon.
Configuring libfuse.
Configuring davfs2.
```

but 
```
$ mount -t davfs -o noexec http://192.168.101.1:8080 test
mount.davfs: group nobody does not exist
mount: mounting http://192.168.101.1:8080 on test failed: No such device
```

there is no `useradd` `groupadd` command . so I install the 
```
$ opkg install shadow-groupadd
$ groupadd -g 65535 nobody
groupadd: cannot open /opt/etc/group
$ ls -l /opt/etc/group
lrwxrwxrwx    1 root     root            10 Jan 21 12:44 /opt/etc/group -> /etc/group
```

zhangguanzhang, 2022-01-21:
```
EmuELEC:~ # grep -Ev '^\s*$|^\s*#' /opt/etc/davfs2/davfs2.conf
dav_user        root            # system wide config file only
dav_group       root            # system wide config file only
```

```
EmuELEC:~ # mount -t davfs   http://192.168.101.1:8080 test
mount.davfs: file /opt/etc/davfs2/secrets has wrong permissions
mount: mounting http://192.168.101.1:8080 on test failed: No such device
```

```
$ chmod 0600 /opt/etc/davfs2/secrets
$ mount -t davfs   http://192.168.101.1:8080 test
Please enter the username to authenticate with server
http://192.168.101.1:8080 or hit enter for none.
  Username: root 
Please enter the password to authenticate user root with server
http://192.168.101.1:8080 or hit enter for none.
  Password:  
$ ls -l test
drwx------    2 root     root             0 Jan 21 14:48 lost+found
drwxr-xr-x   26 root     root             0 Jan 21 09:54 roms
$ ls -l test/roms/
total 0
drwxr-xr-x    5 root     root             0 Jan 21 10:24 arcade
drwxr-xr-x    3 root     root             0 Jan 21 10:24 atomiswave
drwxr-xr-x   19 root     root             0 Jan 21 10:10 bezels
drwxr-xr-x   18 root     root             0 Jan 21 10:09 bios
drwxr-xr-x    7 root     root             0 Jan 21 10:06 dreamcast
drwxr-xr-x    3 root     root             0 Jan 21 10:05 famicom
drwxr-xr-x    4 root     root             0 Jan 21 10:04 fbneo
drwxr-xr-x    3 root     root             0 Jan 21 10:03 gba
drwxr-xr-x    3 root     root             0 Jan 21 10:01 genesis
drwxr-xr-x    3 root     root             0 Jan 21 10:00 megadrive
drwxr-xr-x    3 root     root             0 Jan 21 10:00 n64
drwxr-xr-x    3 root     root             0 Jan 21 09:59 naomi
drwxr-xr-x    3 root     root             0 Jan 21 09:59 nds
drwxr-xr-x    4 root     root             0 Jan 21 09:59 neocd
drwxr-xr-x    3 root     root             0 Jan 21 09:59 openbor
drwxr-xr-x    3 root     root             0 Jan 21 09:58 pcengine
drwxr-xr-x    3 root     root             0 Jan 21 09:58 pcenginecd
drwxr-xr-x    6 root     root             0 Jan 21 09:57 psp
drwxr-xr-x    4 root     root             0 Jan 21 09:57 psx
drwxr-xr-x    3 root     root             0 Jan 21 09:57 sega32x
drwxr-xr-x    3 root     root             0 Jan 21 09:57 segacd
drwxr-xr-x    3 root     root             0 Jan 21 09:56 sfc
drwxr-xr-x    3 root     root             0 Jan 21 09:56 sgfx
drwxr-xr-x    3 root     root             0 Jan 21 09:54 snes
```
😁

shantigilbert, 2022-01-21:
Glad it worked for you, I will close the issue now.

closed 2022-01-21 by shantigilbert