EmuELEC/EmuELEC #796: The monitor breaks down

SajjadSabzkar, 2022-01-22:
# Describe the bug
The monitor breaks down after leaving the game and need to restart emualtion station to fix it .
# requirement
1. RCA Monitor (connected with av connector )
2. Supported Device with Emuelec v4.3
# Steps to reproduce the behavior:
1. Change resolution to 576cvbs (not 576P50HZ)
2. Restart system or shutdown and turn on the device 
3. All working fine and starting the game or start application in setup . for example run terminal 
4. When exit the terminal with hotkey resolution back to other resulotion . i think it's return back to 1080P60HZ
if i restart device all working fine and step 3 to 4 it happens.
# Notic 
1. Emuelec v4.2 It does not have this bug and  v4.3 Emuelec have this bug (test version not checked ) .
2. For other resolution i only test 1080P60HZ it's working fine and only 576cvbs has this bug ( i tested 576cvbs) .
I working on Emuelec Source for solving this bug but i don't must be started . but i hope solved this bug.

# Informations
x96 max plus 

# ScreenShot
[image]

# Video
https://user-images.githubusercontent.com/33274732/150643076-0809a637-4c9d-44f9-9a2c-cf5f26aa9425.mp4

shantigilbert, 2022-01-22:
I have never used such a monitor, so I have no idea how to even start to debug this. if this monitor is a must for you then I would suggest you use v4.2 until someone else can help fix this issue.

shantigilbert, 2022-01-22:
@Langerz82 Do you have any idea about this?

Langerz82, 2022-01-23:
See:
https://github.com/EmuELEC/EmuELEC/pull/799

Hopefully it works ok? I'm not sure if cvbs double buffers or not, guess we will see.

SajjadSabzkar, 2022-01-23:
> See: #799
> 
> Hopefully it works ok? I'm not sure if cvbs double buffers or not, guess we will see.
> 
> edit: Might need to make it higher resolution and scale it down too. But well see if these changes work first.

I Apply your changes to my Emuelec Source but some notic : 
# notic 
1. screen breaks still persists . 
2. after exit game screen width and height is ok but screen breaks only presist. ( befor apply your changes  when exit the game , screen width and height return to defualt resolution [1080P] . this changed visible in uploaded video ) 

# Video
https://user-images.githubusercontent.com/33274732/150683726-c358675e-fa69-4939-ab35-37e4518a2967.mp4

i think this bug refer to show_splash.sh because in v4.2 when exit the game / application loading message is lower than screen size showed and v4.2 not hasn't this bug but this version when exit the game / application loading message is higher than screen size showed and i think this problem  leads to this bugs. thanks for trying solving this bug :)

Langerz82, 2022-01-23:
Can you try in the command prompt enter:
```
setres.sh 576cvbs
```
If that corrupts your screen then setres is likely the error, if not it's ES that is not returning the resolution to the default you selected.

Langerz82, 2022-01-23:
Also with version 4.3 if you leave emulationstation on the setting 1080p (1920x1080) theoretically it should work fine. The old way actually displays 1920x1080 but resizes it to 576cvbs.
If that's the case I might need to make some changes so changing to 576cvbs doesnt corrupt the screen.

SajjadSabzkar, 2022-01-23:
> Also with version 4.3 if you leave emulationstation on the setting 1080p (1920x1080) theoretically it should work fine. The old way actually displays 1920x1080 but resizes it to 576cvbs. If that's the case I might need to make some changes so changing to 576cvbs doesnt corrupt the screen.

yes when screen breaks i'm connect to Emuelec device with ssh and run your command and screen fixed :)

SajjadSabzkar, 2022-01-23:
> 

> Can you try in the command prompt enter:
> 
> ```
> setres.sh 576cvbs
> ```
> 
> If that corrupts your screen then setres is likely the error, if not it's ES that is not returning the resolution to the default you selected.

yes when screen breaks i'm connect to Emuelec device with ssh and run your command and screen fixed :)

Langerz82, 2022-01-23:
can you ssh in and copy/paste your "/storage/.config/emuelec/configs/emuelec.conf" here ?

SajjadSabzkar, 2022-01-23:
> 

> can you ssh in and copy/paste your "/storage/.config/emuelec/configs/emuelec.conf" here ?

yes sure !
**************************************************************************************************
## ------------ A - EmuELEC configuration ------------ #
## Enable SSH at boot
ee_ssh.enabled=1

## Enable Bluetooth at boot
ee_bluetooth.enabled=1

## Use maxperf, enable max settings for GPU and performance governer, this setting can be set globally, per platform or per game on the ES menu
global.maxperf=1

## Always show boot video
ee_bootvideo.enabled=0

## Show a random splash mp4 video from /storage/roms/splash/video
ee_randombootvideo.enabled=0

## Set video mode
ee_videomode=576cvbs

## What to start at boot Emulationstation or Retroarch
ee_boot=Emulationstation

## Always show Retroarch FPS
global.showFPS=0

## Enable retroarch bezels
global.bezel=0

## Enable splash screens
ee_splash.enabled=1

## set to 1 if you have long loading times after quitting a game, you will lose sound mixing in ES, this only affects S905 devices as all others use alsa.
#ee_alsa.always=0

## Force splash screens to display for X seconds
#ee_splash.delay=0

## Some external HDDs take longer to mount than ES to load, if your external ROMS do not mount in time, increase this timer
#ee_load.delay=0

## Retry mounting this many times
ee_mount.retry=1

# Audio device to use, comment for auto, other options 0,0 (default) 0,1 (for some AV outputs)
#ee_audio_device=auto

# Enable Advance mame auto gamepad configuration 1,0 (default 1)
advmame_auto_gamepad=1

# Enable Advance mame config buttom remapping 1,0 (default 0)
advmame_joy_remap=1

AdvanceMame.joy_btns=input a button,input b button,input x button,input y button,input r button,input l button,input r2 button,input l2 button

AdvanceMame.joy_btn_indexes=1,2
AdvanceMame.joy_btn_names=mk,sf

AdvanceMame.joy_btn_order0=0 1 2 3 4 5 6 7
AdvanceMame.joy_btn_order1=3 4 2 1 0 5 6 7
AdvanceMame.joy_btn_order2=3 2 5 1 0 4 6 7

# OdroidGoAdvance Brightness Level
brightness.level=100

## EmulationStation menu style
## default -> default all options menu
## none -> no menu except the game search menu
## bartop -> less menu, only needed for bartops
#system.es.menu=default

# ------------ B - Network ------------ #
## Set system hostname
system.hostname=EMUELEC
## Activate wifi (0,1)
wifi.enabled=1
## Wifi SSID (string)
wifi.ssid=AndroidAPAC04
## Wifi KEY (string)
## after rebooting the "new key" is replace by a hidden value "enc:xxxxx"
## you can edit the "enc:xxxxx" value to replace by a clear value, it will be updated again at the following reboot
## Escape your special chars (# ; $) with a backslash : $ => \$
wifi.key=vvvvvvvv

# secondary wifi (not configurable via the user interface)
#wifi2.ssid=new ssid
#wifi2.key=new key

# third wifi (not configurable via the user interface)
#wifi3.ssid=new ssid
#wifi3.key=new key

# ------------ C - Audio ------------ #
## Set the audio device (auto, hdmi, jack, speakers, headphones) mostly for OdroidGoAdvance
audio.device=auto
## Set system volume (0..100)
audio.volume=100
## Enable or disable system sounds in ES (0,1)
audio.bgmusic=1

# -------------- D - Controllers ----------------- #
# Enable support for standard bluetooth controllers
controllers.bluetooth.enabled=1

# ------------ F - Language and keyboard ------------ #
## Set the language of the system (fr_FR,en_US,en_GB,de_DE,pt_BR,es_ES,it_IT,eu_ES,tr_TR,zh_CN)
system.language=en_US
## Set you local time
## Select your timezone from : ls /usr/share/zoneinfo/ (string)
system.timezone=America/Mexico_City

# ------------ G - UPDATES ------------ #
## Automatically check for updates at start (0,1)
updates.enabled=1
# default : stable ; set to beta to get the next version currently being tested.
updates.type=stable

# ------------ H - HERE IT IS - GLOBAL EMULATOR CONFIGURATION ------------ #
## The global value will be used for all emulators, except if the value
## is redefined in the emulator

## Retroarch menu driver, ozone (default), rgui or xmb 
#global.retroarch.menu_driver=

## Shader set
## Automatically select shaders for all systems
## (none, retro, scanlines)
#global.shaderset=none

## Once enabled, your screen will be cropped, and you will have a pixel perfect image (0,1)
#global.integerscale=0

## Set gpslp shader for all emulators (prefer shadersets above). Absolute path (string)
#global.shaders=

## Set ratio for all emulators (auto,4/3,16/9,16/10,custom)
#global.ratio=4/3

## Set smooth for all emulators (0,1)
#global.smooth=1

## Set rewind for all emulators (0,1)
#global.rewind=1

## Set autosave/load savestate for all emulators (0,1)
#global.autosave=0

## Set runahead frames for all emulators (0,1,2,3,4,5,6)
global.runahead=0

## Set secondinstance for runahead for all emulators (0,1)
global.secondinstance=0

## Enable retroarchievements (0,1)
## Set your www.retroachievements.org username/password
## Escape your special chars (# ; $) with a backslash : $ => \$
global.retroachievements=0
global.retroachievements.hardcore=0
global.retroachievements.leaderboards=0
global.retroachievements.verbose=0
global.retroachievements.screenshot=0
global.retroachievements.username=
global.retroachievements.password=

## Enable RetroArch AI game translation service
#global.ai_service_enabled=0
#global.ai_service_url=
#global.ai_target_lang=

# scrapper
# Comma seperated order to prefer images, s=snapshot, b=boxart, f=fanart, a=banner, l=logo, 3b=3D boxart
#scrapper.style=s,b,f,a,l,3b

## Configurations generated by Emulationstation
audio.display_titles=1
audio.display_titles_time=10
audio.persystem=0

Langerz82, 2022-01-23:
please put the configs around these characters
"```"
code
"```"
and remove your wifi details.

Also in ssh try doing this:
```
echo "576cvbs" > "/storage/.config/EE_VIDEO_MODE"
```
And reboot to see if it makes any difference.

Langerz82, 2022-01-23:
Hey in ssh try the following:
```
cd /storage/.config/emulationstation
nano es_settings.cfg
```
copy/paste before the end config tag on previous lines. ` </config> `
```
	<bool name="FullscreenBorderless" value="true" />
	<int name="ScreenWidth" value="720" />
	<int name="ScreenWidth" value="576" />
```
then reboot.

SajjadSabzkar, 2022-01-23:
> 

> please put the configs around 3 backtick characters to start and 3 to end. ( ` ) <-- character and remove your wifi details.
> 
> Also in ssh try doing this:
> 
> ```
> echo "576cvbs" > "/storage/.config/EE_VIDEO_MODE"
> ```
> 
> And reboot to see if it makes any difference.

> Hey in ssh try the following:
> 
> ```
> cd /storage/.config/emulationstation
> nano es_settings.cfg
> ```
> 
> copy/paste before the end config tag on previous lines. `</config>`
> 
> ```
> 	<bool name="FullscreenBorderless" value="true" />
> 	<int name="ScreenWidth" value="720" />
> 	<int name="ScreenHeight" value="576" />
> ```
> 
> then reboot.

I did everything you said but The problem persists

SajjadSabzkar, 2022-01-23:
> 

> Also with version 4.3 if you leave emulationstation on the setting 1080p (1920x1080) theoretically it should work fine. The old way actually displays 1920x1080 but resizes it to 576cvbs. If that's the case I might need to make some changes so changing to 576cvbs doesnt corrupt the screen.

No, according to the last thing I remember, this does not happen and the problem is the same as the problem I mentioned at the beginning as a problem.

Langerz82, 2022-01-23:
@SajjadSabzkar ah ok. For some reason it does look like its setting to 1920 in width Emulation Station which is corrupting because of the screen size set to 720 in width. If you look carefully the corruption has a bit less than 2.5 repeating patterns, which would be the equivalent 720x3 trying to fit in 1920. So I am thinking that ES for some reason isnt respecting the 576cvbs resolution.

Langerz82, 2022-01-23:
In version 4.3 Try having it corrupt like the bug you said. Then in ssh type in:
```
setres.sh 1080p60hz
```
then see what happens.

SajjadSabzkar, 2022-01-23:
> 

> @SajjadSabzkar ah ok. For some reason it does look like its setting to 1920 in width Emulation Station which is corrupting because of the screen size set to 720 in width. If you look carefully the [...]

also i replace directory ./packages/sx05re/emuelec/  in v4.2 with ./packages/sx05re/emuelec/ with dev branch (last source)  and all things likes v4.2 and works perfectly.

SajjadSabzkar, 2022-01-23:
> In version 4.3 Try having it corrupt like the bug you said. Then in ssh type in:
> 
> ```
> setres.sh 1080p60hz
> ```
> 
> then see what happens.

I did everything you said on version 4.3 of github release and I did not see any noticeable change

# Video

https://user-images.githubusercontent.com/33274732/150691071-9af7bbdf-dc77-424f-a7fe-e95b75e9fbb3.mp4

# notic 
1. i  also try run this command  " setres.sh 576cvbs " and i saw below error in terminal but screen fix in 576cvbs :
/usr/bin/setres.sh: line 97: /sys/class/graphics/fb0/scale_width: Permission denied
/usr/bin/setres.sh: line 98: /sys/class/graphics/fb0/scale_height: Permission denied

shantigilbert, 2022-01-23:
> > In version 4.3 Try having it corrupt like the bug you said. Then in ssh type in:
> > ```
> > setres.sh 1080p60hz
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > then see what happens.
> 
> I did everything you said on version 4.3 of github release and I did not see any noticeable change
> # Video
> v4.3_Emuelec.mp4
> # notic
> 
>     1. i  also try run this command  " setres.sh 576cvbs " and i saw below error in terminal but screen fix in 576cvbs :
>        /usr/bin/setres.sh: line 97: /sys/class/graphics/fb0/scale_width: Permission denied
>        /usr/bin/setres.sh: line 98: /sys/class/graphics/fb0/scale_height: Permission denied

Sorry I have not read all the messages, but judging by the video, this happens only after coming back from setup scripts? or does it happen after coming back from any game? 

What device exactly are you using, you said "Compatible device" but no SOC.

SajjadSabzkar, 2022-01-23:
> 

> Sorry I have not read all the messages, but judging by the video, this happens only after coming back from setup scripts? or does it happen after coming back from any game?
>
> What device exactly are [...]

it  also happen after coming back from any game 

# information Device :
 X96 Max Plus 4GB 32GB Android 9.0 Amlogic S905X3 TV Box Quad Core

shantigilbert, 2022-01-23:
Then it is something with `setres.sh`, the problem is that we do not have such monitor to even debug, @Langerz82 has given you a lot of pointers on how to fix this, unfortunately without the monitor in hand it will fall unto you to tweak it into working.
 
Try to nullify `setres.sh` by adding `return 0` at the top and see if that works

Langerz82, 2022-01-23:
@SajjadSabzkar 
Can you try and copy paste this "setres.sh" into your "/packages/sx05re/emuelec/bin/setres.sh" and see if it works ok?
https://raw.githubusercontent.com/EmuELEC/EmuELEC/34a203166859d19b7e64e6191e64ef43c461687b/packages/sx05re/emuelec/bin/setres.sh

I noticed that the cvbs settings were not writing to the mode so it's possible that skipping that is causing the error.

Langerz82, 2022-01-23:
or you can get the error then enter in ssh:
```
echo 576cvbs > /sys/class/display/mode
```
and see if that fixes the scrambling.

SajjadSabzkar, 2022-01-23:
> @SajjadSabzkar Can you try and copy paste this "setres.sh" into your "/packages/sx05re/emuelec/bin/setres.sh" and see if it works ok? [...]

Wow I changed this and it works well :))
Thank you for taking the opportunity to fix it

SajjadSabzkar, 2022-01-23:
> or you can get the error then enter in ssh:
> 
> ```
> echo 576cvbs > /sys/class/display/mode
> ```
> 
> and see if that fixes the scrambling.
> 
> Let me know. I'd like to make a permanent fix for this. ^_^

The bug has been fixed. If you like, I can check this too

shantigilbert, 2022-01-23:
So this is fixed, with #799 ?

SajjadSabzkar, 2022-01-23:
> So this is fixed, with #799 ?

yes , thank you very much ^_^

closed 2022-01-23 by shantigilbert

Langerz82, 2022-01-23:
ok great, thanks for helping us diagnose the issue.