EmuELEC/EmuELEC #797: neogeo game failed

zhangguanzhang, 2022-01-22:
# BEFORE YOU OPEN AN ISSUE MAKE SURE IT IS NOT ALREADY RESOLVED IN THE WIKI https://github.com/EmuELEC/EmuELEC/wiki

# MAKE SURE YOUR DEVICE IS A [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices)  

# PLEASE DO NOT POST GENERAL HELP QUESTIONS! EITHER USE THE FORUM (https://emuelec.org) OR USE DISCORD FOR THAT (https://discord.gg/cbgtJTu), THIS IS ONLY FOR BUG/ISSUES REPORTING!

# DELETE EVERYTHING ABOVE THIS LINE (INCLUDING THIS LINE) BEFORE SUBMITTING TO MAKE SURE YOU'VE READ THIS!
[image]
[image]
[image]

```
# cat /etc/os-release 
NAME="EmuELEC"
VERSION="4.0"
ID="coreelec"
VERSION_ID="9.2"
PRETTY_NAME="EmuELEC (Official - EmuELEC is FREE! NOT FOR SALE!): 4.0"
HOME_URL="https://github.com/EmuELEC"
BUG_REPORT_URL="https://github.com/EmuELEC/EmuELEC"
BUILD_ID="597b8729000c4d53cc036e36b015aae1a66693da"
OPENELEC_ARCH="Amlogic.aarch64"
LIBREELEC_ARCH="Amlogic.aarch64"
COREELEC_ARCH="Amlogic.aarch64"
COREELEC_BUILD="Official - EmuELEC is FREE! NOT FOR SALE!"
COREELEC_PROJECT="Amlogic"
BUILDER_VERSION="4.0"
```
game file:
[neobombe.zip](https://github.com/EmuELEC/EmuELEC/files/7918737/neobombe.zip)

zhangguanzhang, 2022-01-22:
[neobombe.zip](https://github.com/EmuELEC/EmuELEC/files/7918747/neobombe.zip)
I use the file will be ok

closed 2022-01-22 by zhangguanzhang

shantigilbert, 2022-01-23:
Sorry was I not clear last time? DO NOT POST ROMS!!!!!