EmuELEC/EmuELEC #798: mame mage "grdians" cannot play

zhangguanzhang, 2022-01-22:
# BEFORE YOU OPEN AN ISSUE MAKE SURE IT IS NOT ALREADY RESOLVED IN THE WIKI https://github.com/EmuELEC/EmuELEC/wiki

# MAKE SURE YOUR DEVICE IS A [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices)  

# PLEASE DO NOT POST GENERAL HELP QUESTIONS! EITHER USE THE FORUM (https://emuelec.org) OR USE DISCORD FOR THAT (https://discord.gg/cbgtJTu), THIS IS ONLY FOR BUG/ISSUES REPORTING!

# DELETE EVERYTHING ABOVE THIS LINE (INCLUDING THIS LINE) BEFORE SUBMITTING TO MAKE SURE YOU'VE READ THIS!
[image]
game rom file:
[grdians.zip](https://github.com/EmuELEC/EmuELEC/files/7918775/grdians.zip)

shantigilbert, 2022-01-22:
Please, do not open this kind of issues here, use discord or the forum, and PLEASE do not share ROMS!

closed 2022-01-22 by shantigilbert

zhangguanzhang, 2022-01-22:
😢ok