EmuELEC/EmuELEC #814: Incorrect resolution Mupen64 in Gameforce

SamuEDL, 2022-02-16:
**Bug:**
MUPEN64 Standalone (Both in rice and in Glide64mk2) is not adapted to the Gameforce resolution on 4.4-TEST-02152022. From what I see in the log, try to be shown in 1080p (Core: Setting 16-bit video mode: 1920x1080).

**Informations:**
Version:  4.4-TEST-02152022
Hardware Platform: Gameforce Chi

**Log:**
[emuelec.log.txt](https://github.com/EmuELEC/EmuELEC/files/8081683/emuelec.log.txt)

**Test image**
[image]
.

Langerz82, 2022-03-14:
You have to change the resolution to 640x480 for example in SS:
/emuelec/configs/mupen64plussa/mupen64plus.cfg
```
[Video-General]

# Width of output window or fullscreen width
ScreenWidth = 640
# Height of output window or fullscreen height
ScreenHeight = 480
# Use fullscreen mode if True, or windowed mode if False
Fullscreen = True
# If true, activate the SDL_GL_SWAP_CONTROL attribute
VerticalSync = True
# Rotate screen contents: 0=0 degree, 1=90 degree, 2 = 180 degree, 3=270 degree
Rotate = 0
```

Langerz82, 2022-03-14:
For testing see:
https://github.com/EmuELEC/EmuELEC/pull/849

commit 5381a711 (2022-03-17): Mupen64plus: Add handheld resolutions

fixes #814