EmuELEC/EmuELEC #817: [BUG]Drastic does not make the installation

SamuEDL, 2022-02-16:
**Bug:**
The "Install Drastic" script it does not work correctly and returns to Emulationstation. 

**Informations:**
Version: 4.4-TEST-02152022
Hardware Platform: Gameforce Chi and Amlogic Box

**Logs:**
[ee_script.log](https://github.com/EmuELEC/EmuELEC/files/8082379/ee_script.log)

shantigilbert, 2022-02-26:
Can you try again, but this time press the "A" button on the gamepad when the "OK" button is selected on the window?

PikachuEXE, 2022-02-26:
I am using OGS (from #819) and there is no "OK" button for me

Here are my logs
[emuelec.log](https://github.com/EmuELEC/EmuELEC/files/8146133/emuelec.log)
[ee_script.log](https://github.com/EmuELEC/EmuELEC/files/8146134/ee_script.log)

Video for running `Install Drastic`

https://user-images.githubusercontent.com/1018543/155835129-78447ca1-1df2-4308-a8c2-603813f0bcb0.mp4

shantigilbert, 2022-02-26:
unzip this file in `/emuelec/bin` (or `configfiles/emuelec/bin` if using samba) and try again. 

[text_viewer.zip](https://github.com/EmuELEC/EmuELEC/files/8147595/text_viewer.zip)

PikachuEXE, 2022-02-27:
`text_viewer` uploaded to `/emuelec/bin` with permission changed to `755`
Now both works and drastic can run NDS games

closed 2022-03-19 by shantigilbert