EmuELEC/EmuELEC #823: Xbox 360 wired controller exit back to Emulation Station

rtodd23456, 2022-02-22:
even though i have config the controller in Emualtion Station i set the back button as the hotkey 

so when i want to exit back to menu from game play i normally press back and start at same time twice .

but for some reason i have to press the big middle xbox button instead so it thinks the xbox middle button is the back button

does not do this on 4.3 stable

HK1 Android Box

AlienatedDiego, 2022-02-22:
well i'm not sure. there were many changes in 4.4

From your report it looks like maybe just an ID change of your controller ; afaik those are recognized as "Microsoft x-box 360 pad" which EmuELEC uses as default in es_input.conf and may now be recognized with another id, maybe "xbox360 controller" but keep using the other one

in short. if you have upgraded from 4.3 to 4.4 just try to reset scripts and try again (remap it)
if nothing has changed then try editing the es_input.conf file deleting the default input and remapping it again might work

Langerz82, 2022-03-09:
Hello,
Using the latest version of gptokeyb in its package.mk (packages/sx05re/sysutils/gptokeyb) should allow you to either use the back button or the guide button as the hotkey.

In package.mk change to:
```
PKG_VERSION="0303b36b5376a9b25cf82a53ed4242509daf14e9"
```

See:
https://github.com/EmuELEC/gptokeyb/commit/0303b36b5376a9b25cf82a53ed4242509daf14e9

closed 2022-04-18 by shantigilbert