EmuELEC/EmuELEC #824: [OGABE] Menu does not appear in AdvMame.

british-choi, 2022-02-23:
Device : OGABE
Version : v4.4-TEST-02152022

If you press the tab key in AdvMame, the Menu screen should appear.
However, it is converted from graphic screen to text screen. (Game Sound is output normally.)

british-choi, 2022-02-26:
> 

yes, it is SDL 2.0.16(version) issue

miwasp, 2022-03-06:
Menus do not appear in all latest TEST-releases up to current dev on all handheld devices (OGA, OGS and GF).
It works fine on Amlogic devices.

shantigilbert, 2022-03-07:
Good news and bad news... 

Bad news is that this probably won't get fixed as it seems its an issue with the newer SDL and advmame with small screens.
 
Good news is that there is now a key-map function for advmame in Emulationstation, so the only reason to open the menu (key mapping) can be done in ES directly, this with the new gamepad auto configuration (up to 8 players) makes the menu unnecessary. 

I might remove the menu button function so that it is not pressed by mistake.

I will keep this issue open in case someone comes along with a fix.

miwasp, 2022-03-08:
Setting "misc_smp no" in advmame.rc will show the menus.
I don't know if this can be called a 'fix' as i don't know what this means to performance.

commit 055c3e79 (2022-03-08): advancemame: use `misc_smp` no for GF and OGA/OGS

Fixed #824 thanks to @miwasp
commit 740820a4 (2022-03-08): advancemame: use `misc_smp no` for GF and OGA/OGS

Fixed #824 thanks to @miwasp
closed 2022-03-08 by shantigilbert