EmuELEC/EmuELEC #827: Odroid N2+,Xbox360 wireless controller, quit to ES on AdvanceMAME & FBNeoSA

dewenhsu, 2022-02-26:
EmuELEC 4.4 Test-0215-2022, Odroid N2+, Xbox360 Wireless controller :
when lanuch AdvanceMAME or FBNeoSA, it will show purple emuelec loading screen then quit back to ES screen. 
Check log for AdvMame, it shows "Invalid argument in input_map[p1_button7"
( as the attached picture ),
[image]
https://www.dropbox.com/s/jvds3crormpc0bh/emuelec_advmame.log?dl=0

for FBNeoSA, it just loading and quit like AdvMame, but the log file shows no clue
https://www.dropbox.com/s/f8nn4qqila6pc7i/emuelec_fbneosa.log?dl=0

Thank you !

miwasp, 2022-02-26:
@Langerz82 : Can you please have a look at this?
The script set_advmame_joy.sh creates faulty entries in advmame.rc which prevent advmame from starting.

british-choi, 2022-02-27:
@shantigilbert 
fbneoSA core is not compiled in EmuELEC

Langerz82, 2022-02-27:
The analog index 2 and 5 don't have values assigned to them, I'll see if I can create a fix.

Langerz82, 2022-02-27:
Hey, try the PR fix with the latest dev build and see if it works ok.

miwasp, 2022-02-27:
I don't have a xbox controller but i think this is a more general problem. I have a similar problem with a Sony controller and your PR does not change anything with this controller.

This is the output of "set_advmame_joy.sh"
> N2+:~/.advance # set_advmame_joy.sh
BTN_CFG=0 1 2 3 4 5 6 7
CONTROLLER: 1 js1 030000004c050000cc09000011810000 Sony Interactive Entertainment Wireless Controller
GC_CONFIG=030000004c050000cc09000011810000,Sony Interactive Entertainment Wireless Controller,platform:Linux,,dpleft:h0.8,rightx:a3,dpright:h0.2,rightshoulder:b5,dpdown:h0.4,righty:a4,leftshoulder:b4,,y:b3,x:b2,b:b0,a:b1,dpup:h0.1,back:b8,leftstick:b11,start:b9,,lefty:a1,guide:b10,lefttrigger:b6,righttrigger:b7,,rightstick:b12,leftx:a0,

This is (part of) the generated "advmame.rc"
> ...
input_map[p1_button7] joystick_button[sony_interactive_entertainment_wireless_controller,button8]
input_map[p1_button8] joystick_button[sony_interactive_entertainment_wireless_controller,button7]
input_map[p1_up] joystick_button[sony_interactive_entertainment_wireless_controller,stick4,y,up] or joystick_digital[sony_interactive_entertainment_wireless_controller,stick,y,up]
input_map[p1_down] joystick_button[sony_interactive_entertainment_wireless_controller,stick4,y,down] or joystick_digital[sony_interactive_entertainment_wireless_controller,stick,y,down]
...

This is the emuelec.log after trying to start advancemame:
> N2+:~/.advance # more /emuelec/logs/emuelec.log
EmuELEC Run Log
##############################################
\#                 EmuELEC                    #
\#    Retro emulation for Amlogic Devices    #
\#             Based on CoreELEC              #
##############################################
EmuELEC (Community - EmuELEC is FREE! NOT FOR SALE!): 4.4-Nexus_devel_20220216073739 (Amlogic-ng.aarch64)
PLATFORM: mame
ROM NAME: /storage/roms/mame/dkong.zip
BASE ROM NAME: dkong.zip
USING CONFIG: /storage/.config/retroarch/retroarch.cfg
1st Argument: /storage/roms/mame/dkong.zip
2nd Argument: -Pmame
3rd Argument: --core=AdvanceMame
4th Argument: --emulator=AdvanceMame
Full arguments: /storage/roms/mame/dkong.zip -Pmame --core=AdvanceMame --emulator=AdvanceMame --controllers=-p1index 0 -p1guid 030000004c050000cc09000011810000
Run Command is:
/usr/bin/bash advmame.sh /storage/roms/mame/dkong.zip
Emulator Output is:
Invalid argument 'joystick_button[sony_interactive_entertainment_wireless_controller,stick4,y,up] or joystick_digital[sony_interactive_entertainment_wireless_controller,stick,y,up]' for option 'input_map[p1_up]'.
Invalid joystick button 'stick4,y,up'.
Valid format is keyboard[BOARD,KEY]/joystick_button[JOYSTICK,BUTTON]/mouse_button[MOUSE,BUTTON].
exit 1

Langerz82, 2022-02-27:
@miwasp Hello, your error is a bit different because it's concerning hat values on the dpad. When I switch inputs to SDL hat values dont get picked up by advmame. I've reverted it to use raw input, and hopefully the hat values will work ok. Try with the latest changes on my patch and see if it works. If it doesnt work you may have to change the hat stick value in emuelec.conf from 4 to whichever number works for your controller.

miwasp, 2022-02-28:
Your latest changes fixed it for the Sony controller. I didn't had to make changes to emuelec.conf.
Thank you.

Langerz82, 2022-02-28:
@miwasp ok great, good to hear. Can you also run flycastSA and DolphinSA without issues and the dpad working fine?

@dewenhsu Hey man can you apply my PR and test all is working ok for you as well?

miwasp, 2022-02-28:
dpad is also working fine in flycastSA. I don't use DolphinSA, so i can't say.

dewenhsu, 2022-02-28:
> @miwasp ok great, good to hear. Can you also run flycastSA and DolphinSA without issues and the dpad working fine?
> 
> @dewenhsu Hey man can you apply my PR and test all is working ok for you as well?

PR-fix set_advmame_joy.sh output :
EMUELEC:~ # EMUELEC:~ # EMUELEC:~ # /emuelec/bin/set_advmame_joy.sh
BTN_CFG=0 1 2 3 4 5 6 7
CONTROLLER: 1 js0 030000005e040000a102000007010000 Xbox 360 Wireless Receiver (XBOX)
GC_CONFIG=030000005e040000a102000007010000,360 Wireless Receiver (XBOX),dpleft:b11,rightx:a3,dpright:b12,rightshoulder:b5,dpdown:b14,righty:a4,leftshoulder:b4,y:b2,x:b3,b:b0,a:b1,dpup:b13,back:b6,leftstick:b9,start:b7,lefty:a1,guide:b6,lefttrigger:a2,righttrigger:a5,rightstick:b10,leftx:a0,platform:Linux,

Generated advmame.rc
https://www.dropbox.com/s/1i86c4ce0yjtwxd/pr-fix-advmame.rc?dl=0

Advmame still quit to ES. log file
https://www.dropbox.com/s/757tz7eyo8814ng/PR-fix-advmame-emuelec.log?dl=0

BTW, I can run GameCube ( dolphin ) controller map OK by default, except I can not quit to ES ( no hot key to quit )

thank you.

Langerz82, 2022-03-01:
@dewenhsu Hello, try and use the latest patch I added:
https://github.com/EmuELEC/EmuELEC/pull/828/commits/6c004767169d5f378d615092700c1a197904dff4

It will rename the "joystick" to "joystick_digital", use "raw" instead of "sdl", and should create valid values for buttons 7 and 8.

dewenhsu, 2022-03-01:
> @dewenhsu Hello, try and use the latest patch I added: [6c00476](https://github.com/EmuELEC/EmuELEC/commit/6c004767169d5f378d615092700c1a197904dff4) edit: fixed link.
> 
> It will rename the "joystick" to "joystick_digital", use "raw" instead of "sdl", and should create valid values for buttons 7 and 8.

dewenhsu, 2022-03-01:
> > @dewenhsu Hello, try and use the latest patch I added: [6c00476](https://github.com/EmuELEC/EmuELEC/commit/6c004767169d5f378d615092700c1a197904dff4) edit: fixed link.
> > It will rename the "joystick" to "joystick_digital", use "raw" instead of "sdl", and should create valid values for buttons 7 and 8.

dewenhsu, 2022-03-01:
@Langerz82 
thank for the help ! I think it is nearly solved...

test result :
new set_advmame_joy.sh
https://www.dropbox.com/s/q9k16fbsq6tk48c/set_advmame_joy-v2.sh?dl=0

putty output
https://www.dropbox.com/s/hg735ntf3h0tc4r/output.txt?dl=0

new created advmame.rc
https://www.dropbox.com/s/ehg28hro438exks/advmame-v2.rc?dl=0

log file ( Invalid argument 'joystick_digital[xbox_360_wireless_receiver_xbox,stick,rz]' for option 'input_map[p1_button7]'.
Missing , in 'joystick_digital'.
https://www.dropbox.com/s/0skxafdqqkbbqfd/emuelec-v2.log?dl=0

Langerz82, 2022-03-01:
I'll try and test later today, with my xbox series x controller if I get some time.

Langerz82, 2022-03-01:
@dewenhsu 
Ok, first try disabling the auto gamepad in emuelec.conf by going into ssh and changing in 
/storage/.config/emuelec/configs/emuelec.conf
```
advmame_auto_gamepad=0
```
save then close.
Then manually adjust the advmame.rc by adding down in the end bit, /storage/.advance/advmame.rc
```
input_map[p1_button7] joystick_digital[xbox_360_wireless_receiver_xbox,stick,rz,down]
input_map[p1_button8] joystick_digital[xbox_360_wireless_receiver_xbox,stick,z,down]
```
And see if that loads, if it does, in advmame go into input config for the game and set any button to one of the triggers and let me know the value it gives. This will confirm if the button setting above is correct.

Langerz82, 2022-03-02:
@dewenhsu 
I checked values in advj on my desktop and I think it should be:
```
input_map[p1_button7] joystick_digital[xbox_360_wireless_receiver_xbox,stick,rz,right]
input_map[p1_button8] joystick_digital[xbox_360_wireless_receiver_xbox,stick,z,right]
```

dewenhsu, 2022-03-02:
@Langerz82 I tried rz,down or rz,right, but still not work
Log file :
https://www.dropbox.com/s/91uk9ih53n6etp5/emuelec.log?dl=0

Then I tried delete button 7 & 8 config in advmame.rc,
enter advmame success and manual config 
button 7 for Right Trigger and button 8 for Left Trigger
[image]

the button 7 & 8 data in advmame.rc
`input_map[p1_button7] joystick_digital[xbox_360_wireless_receiver_xbox,2,1,0] joystick_digital[xbox_360_wireless_receiver_xbox,2,1,1]
input_map[p1_button8] joystick_digital[xbox_360_wireless_receiver_xbox,1,0,0] joystick_digital[xbox_360_wireless_receiver_xbox,1,0,1]`

I don't know if it can help ? But hope it will :-)

Langerz82, 2022-03-02:
@dewenhsu 
Hello, great thanks for your detailed reporting. I have created a new patch which will require testing, if you can it would be great for you to test as I do not have a xbox usb dongle handy. If so let me know how it went.
See New Patch:
https://github.com/EmuELEC/EmuELEC/pull/828/files

Cheers.

Langerz82, 2022-03-02:
Great to hear. Yeah the power issue might be a bit above my expertise. I don't think it would be too hard but I'm not familiar with the code. I know SDL has an event when controllers re-connect and that can be picked up and re-initialize the controller, but it depends what the emulator is using as an input driver.

dewenhsu, 2022-03-02:
sp should I close the comment ?

shantigilbert, 2022-03-08:
Fixed in: https://github.com/EmuELEC/EmuELEC/commit/740820a4ad45d143e52de74764983038ec28cd49

closed 2022-03-08 by shantigilbert