EmuELEC/EmuELEC #834: This problem occurs when compiling 4.3, I don't know how to solve it, I hope experts can give me some pointers.

xtgrrk, 2022-03-06:
>>> opentyrian:target seq 493 >>>
[491/545] [DONE] install opentyrian:target
<<< bermuda:target seq 494 <<<
INSTALL      bermuda (target)
>>> bermuda:target seq 494 >>>
[492/545] [DONE] install bermuda:target
<<< hodesdl:target seq 495 <<<
ERROR: /home/xtgrrk/EmuELEC/sources/hodesdl/hodesdl-a971e54d53ac8c8e94894139d5b89c81d04b4d63: failed to determine git HEAD
ERROR: /home/xtgrrk/EmuELEC/sources/hodesdl/hodesdl-a971e54d53ac8c8e94894139d5b89c81d04b4d63: failed to determine git HEAD
GET      hodesdl (git)
    GIT PULL      hodesdl
Already up to date.
There is no commit 'a971e54d53ac8c8e94894139d5b89c81d04b4d63' on branch 'master' of package 'hodesdl'! Aborting!
*********** FAILED COMMAND ***********
. "${get_handler}"
**************************************
*********** FAILED COMMAND ***********
${SCRIPTS}/get "${PKG_NAME}"
**************************************
*********** FAILED COMMAND ***********
${SCRIPTS}/unpack "${PKG_NAME}" "${PARENT_PKG}"
**************************************
*********** FAILED COMMAND ***********
${SCRIPTS}/build "${1}" "${PARENT_PKG}"
**************************************
FAILURE: scripts/install hodesdl:target has failed!

The following log for this failure is available:
  /home/xtgrrk/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/.threads/logs/495.log

>>> hodesdl:target seq 495 >>>
[493/545] [FAIL] install hodesdl:target

The following log for this failure is available:
  /home/xtgrrk/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/.threads/logs/495.log

Parallel build failure - see log for details. Time of failure: Sun Mar  6 00:47:29 PST 2022
Makefile:12: recipe for target 'image' failed
make: *** [image] Error 1

Kelvfimer, 2022-03-06:
Hello

I fixed it editing the package.mk at /emuelec/package/sx905/packages/hodesdl/ you have to change the pkg number to dcf1366561582189e77ac5131d21fc56a26ba524

xtgrrk, 2022-03-06:
> 你好
> 
> 我修复了它在 /emuelec/package/sx905/packages/hodesdl/ 编辑 package.mk 您必须将 pkg 编号更改为 dcf1366561582189e77ac5131d21fc56a26ba524

Thank you for taking the time to reply to the question, thank you very much, the above problem has been solved perfectly, and now there is a new problem, which is frustrating to me. .

-------------------------------------------------------------------------------------------------------

4-libreelec-linux-gnueabi/10.3.0/../../../../aarch64-libreelec-linux-gnueabi/bin/ld.gold: fatal error: ../../../../../mamearcade2016_libretro.so: No space left on device
collect2: error: ld returned 1 exit status
make[3]: *** [mamearcade2016.make:444: ../../../../../mamearcade2016_libretro.so] Error 1
make[2]: *** [Makefile:364: mamearcade2016] Error 2
make[2]: Leaving directory '/home/xtgrrk/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/build/mame2016-bcff8046328da388d100b1634718515e1b15415d/build/projects/retro/mamearcade/gmake-linux'
make[1]: *** [makefile:1211: linux_x86] Error 2
make[1]: Leaving directory '/home/xtgrrk/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/build/mame2016-bcff8046328da388d100b1634718515e1b15415d'
FAILURE: scripts/build mame2016:target during make_target (package.mk)
*********** FAILED COMMAND ***********
make $PKG_MAKE_OPTS_TARGET OVERRIDE_CC=$CC OVERRIDE_CXX=$CXX OVERRIDE_LD=$LD AR=$AR $MAKEFLAGS
**************************************
*********** FAILED COMMAND ***********
${SCRIPTS}/build "${1}" "${PARENT_PKG}"
**************************************
FAILURE: scripts/install mame2016:target has failed!

The following log for this failure is available:
  /home/xtgrrk/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/.threads/logs/519.log

>>> mame2016:target seq 519 >>>
[542/545] [FAIL] install mame2016:target

The following log for this failure is available:
  /home/xtgrrk/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/.threads/logs/519.log

Parallel build failure - see log for details. Time of failure: Sun Mar  6 06:03:21 PST 2022
Makefile:12: recipe for target 'image' failed
make: *** [image] Error 1

xtgrrk, 2022-03-06:
I tried again and the prompt said it might be the disk space, I'll try again. Thank you very much for your answer.

closed 2022-03-07 by shantigilbert