EmuELEC/EmuELEC #841: Wifi driver QCA6174 (S905L)

331105363, 2022-03-09:
Please add wifi driver QCA6174 for S905L

I have the same problem as #611  

https://github.com/EmuELEC/EmuELEC/issues/611

Please add wifi driver QCA6174 for Amlogic, thank you very much!

SamuEDL, 2022-03-09:
What version of Emuelec is using?

331105363, 2022-03-09:
> What version of Emuelec is using?

4.4

331105363, 2022-03-09:
> What version of Emuelec is using?

EmuELEC-Amlogic.aarch64-4.3

331105363, 2022-03-09:
> What version of Emuelec is using?

The driver of this chip is added in Amlogic-ng, can it be added in Amlogic

SamuEDL, 2022-03-09:
> > What version of Emuelec is using?
> 
> The driver of this chip is added in Amlogic-ng, can it be added in Amlogic

If it's as you're commented, in the next version it will be solved since you will only have "amlogic-ng". If you can try Emuelec 4.4-test, to check that it works: https://github.com/EmuELEC/EmuELEC-tests/releases/tag/v4.4-TEST-02152022

331105363, 2022-03-09:
> If it's as you're commented, in the next version it will be solved since you will only have "amlogic-ng". If you can try Emuelec 4.4-test, to check that it works: [...]

Emuelec 4.4-test can not be used for s905L, I update Emuelec 4.4-test based on Emuelec 4.3, indicating file damage or insufficient space, 64GB USB flash drive

> If it's as you're commented, in the next version it will be solved since you will only have "amlogic-ng". If you can try Emuelec 4.4-test, to check that it works: [...]

Emuelec 4.4-test can not be used for s905L, I update Emuelec 4.4-test based on Emuelec 4.3, indicating file damage or insufficient space, 64GB USB flash drive

shantigilbert, 2022-03-09:
You can wait for 4.4 next test or final to see if it's fixed, otherwise, there will not be a fix for this.

closed 2022-03-09 by shantigilbert