EmuELEC/EmuELEC #845: advmame screen tearing and vsync

dewenhsu, 2022-03-13:
Odroid N2+, EmuELEC TEST0215 :
for advmame, there is a "**tearing**" effect
( horizontal tear line when playing vertical scroll game. eg. Brave Blade, Nostradamus  )
https://www.advancemame.it/doc-faq#7.5

The advmame author suggest enable "vertical sync" to fix the problem.
https://www.advancemame.it/doc-faq#7.6
you can turn on **vsync** from UI -> Video or add a line in advmame.rc
display_vsync yes 
https://www.advancemame.it/doc-advmame#8.4.10

Unfortunately, when you turn on vsync, it seems vsync not detect the correct refresh rate,
and the game runs too fast. 

advmame use the default framebuffer ( fb ) video driver,
https://www.advancemame.it/doc-advdev#1.2.1

"This driver is able to use the Linux fb API to **synchronize with the vertical sync** of the video mode, but generally **it isn't possible** because the low-end drivers don't support this feature. Anyway, **if you run the program as root** it can use the standard VGA registers to detect the vsync."
I tried the root permission as suggestion for advmame bin, but  still not work

Is there a way to let advmame detect the correct refresh rate for vsync,
so to fix tearing effect and game will not run too fast ?

The other way to fix tearing effect is to use **sdl video drive**r instead of **fb** driver 
add the following line in advmame.rc  to use sdl driver :
device_keyboard sdl
device_mouse sdl
device_video sdl
display_vsync yes
The only disadvantage is that sdl driver use "overlay" as video output :
device_video_output overlay
and the RGB effect ( scanline effect ) is broken with odd distorsion/moiree effects
To us sdl driver with correct RGB effect, you have to use "fullscreen" as video output
device_video_output fullscreen  ( add the line in advmame.rc )
but the "fullscreen" mode only support in sdl1, and the version EmuELEC used is sdl2.
https://sourceforge.net/p/advancemame/discussion/313511/thread/f2cdcb33/

Is it possible to compile advmame with **sdl1** so it can use sdl driver with fullscreen output, to fix tearing effect, enable vsync with normal speed, and normal RGB effect ?

I searched that advmame can be compiled with sdl1 with  --prefix, but I don't know how and will it be work on EmuELEC ?

"The configure script automatically detects all the available libraries and the optimization flags. You can use the --with-sdl-prefix and --with-sdl2-prefix option to search for the SDL and SDL2 libraries in a specific location."
https://www.advancemame.it/doc-build

If anyone can help ( compile sdl1 version ? hint on how to build sdl1 advmame ) 
thank you very much !

shantigilbert, 2022-03-13:
SDL1 does not work on emuelec.

Langerz82, 2022-03-14:
Try these settings in advmame.rc
```
display_vsync no
sync_fps 60
display_frameskip 0.9999
```
Set the sync fps to your display refresh rate. Theoretically it should emulate display frames just before the fps sync. I tested it briefly and noticed less tearing, but give it a go and let me know.

dewenhsu, 2022-03-14:
> display_vsync no
> sync_fps 60
> display_frameskip 0.9999

Thank you for the suggestion.
Not sure my NTSC LCD TV's frame rate ( it's 1080P HDMI TV 10 yrs ago ) 
so I set it to 60.
frameskip 0.9999 makes game ( Nostradamus ) background scroll more stutter,  
so I use 1 ( no skip ) or auto.  anyway, the tearing effect seems better ? but still exist.

shantigilbert, 2022-03-14:
> NTSC? The sync fps should be set to 50 then. NTSC runs at 50Hz by default.

PAL runs at 50hz, NTSC runs at 60hz. But it is not an issue with refresh rate, ADVMAME has had those issues for ever, I think is the FrameBuffer that is at fault. There is nothing we can actually do, except reduce it.

Langerz82, 2022-03-14:
Ah sorry my bad, I thought it was the other way around. Whats the brand and model of your TV? Are you sure it does not run at a even higher frequency by default?

dewenhsu, 2022-03-16:
> Ah sorry my bad, I thought it was the other way around. Whats the brand and model of your TV? Are you sure it does not run at a even higher frequency by default? edit: Sorry shanti I thought you were dewenhsu, ah ok.

I check the TV Manual, it's CHIMEI LED 46" TL-46Z8000D, HDMI support 640x480@60Hz, 720x400@60Hz, 800x600@60Hz, 1024x768@60Hz, 1920x1080@60Hz(native). It's an old TV ( about 10+yrs ) and seems nto support higher frequency.
[image]

I also think the tearing problem is difficult to solve unless you use SDL instead of FB, but you lost the RGB scanline. It's lucky Retroarch and other SA emu still support vsync.

Langerz82, 2022-04-19:
What if you do,
```
display_vsync no
sync_fps 61
display_frameskip 1
```
any difference? Just an idea.

dewenhsu, 2022-04-20:
> What if you do,
> 
> ```
> display_vsync no
> sync_fps 61
> display_frameskip 1
> ```
> 
> any difference? Just an idea.

It seems improved but not complete fix.
you can also use keyboard F8/F9 to try different frame rate,
but the tearing effect still can not complete disappear.

Langerz82, 2022-04-20:
Hey did you also set the native video to 640x480 ? Theoretically a smaller resolution will make the buffer complete faster in cases where it's not completing fast enough.

dewenhsu, 2022-04-21:
> Hey did you also set the native video to 640x480 ? Theoretically a smaller resolution will make the buffer complete faster in cases where it's not completing fast enough.

I tried different native video including 640x480, it seems improved but not complete fix the tearing effect.

I agree with the author said [https://www.advancemame.it/doc-faq#7.6](url)
The only way to remove the tearing effect is to enable the `**display_vsync**' option. 
Unfortunately, for unknown reason, vsync only works for **SDL** video driver, if used in FB driver, the game speed will run too fast.

The benefit for SDL driver :
1. under ES native video resolution, for example 1080P, it will automatic choose the best Magnify[x] video mode ( usually it choose x=2, 2x original game resolution ), and you can also adjust the x value from 1 to 4 freely. If you use FB driver, the x  is usually fixed to 4x and can not be adjust, and will slow down some game's speed due to high 4x resolution.
2. It is [integer] resized
3. You can enable Vsync to fix the tearing effect.  the vertical scroll is much smoother.

The drawback for SDL driver :
1. The RGB Effect ( scanline ) is not correct. ( you lost scanline effect )
2. The text font of config menu is blurred if magnify[x] is too low ( eg. Magnify[x]=1, original game resolution ),  but you can switch to 3x or 4x Magnify so you can read the text and switch back  after change.

I personally use SDL instead of FB. hope someday the author will update and fix the problem.

Langerz82, 2022-09-15:
This being fixed would be great, I noticed the hug tearing in wonderboy scroller. Anyway here are my SDL settings which seem to work ok:
```
display_rgbeffect none
display_vsync yes
device_video_clock 10-100 / 10-100 / 60-70
display_color auto
display_scanlines no
sync_fps 60
sync_speed 1
display_buffer yes
vertical/display_rol yes
device_video sdl
device_keyboard none
misc_smp yes
display_resize fractional
display_resizeeffect none
```

Langerz82, 2022-09-17:
I worked on this issue a bit but could not make any progress. I noticed that if you slow the synching down in the config if the fps is too high for the synch the tear it scrolls upward, if it's too slow the tear goes downwards. And by halfing the value after the decimal point and checking if it goes downwards (too slow) or upwards (too fast) you can find a sweet spot that the horizontal tearing becomes almost obsolete to the eye. The issue is first this number that occurs is different for each resolution, second I have found a reliable way of working out the exact value other than trial and error.

Langerz82, 2022-09-17:
Try copy/pasting this file in SSH into /emuelec/bin/advmame.sh with execute permissions. I'm not sure if the 640x480 timings will work for different devices, but it seems to work for mine with little to no screen tearing. For 720p I had to disable smp (misc_smp no) to make it scroll perfectly without tearing.

See:
https://gist.github.com/Langerz82/773b355c14d9f844328515861bf94f61

Langerz82, 2022-09-17:
I worked out what is causing the tears, the pixel clock in device_modeline rounding off to two digits causes the sync_fps to be off by the amount it ends up dropping and round up too.
So for example if its vsynch set too 60, and in modeline say the pixel clock is real value is 51.2222 but is rounded to 51.22 dropping off 0.0022 that has to come off the synch_fps value to avoid screen tearing which will be actually need to be: 60 - 0.0022 = "59.9978".
I'm looking at a way to fix it in advmame for the framebuffer mode.

Langerz82, 2022-09-18:
@dewenhsu 

For a fix that needs more testing see:
https://github.com/EmuELEC/EmuELEC/pull/992

Langerz82, 2022-09-18:
@shantigilbert I worked it out, it wasn't what I said earlier that was the issue lol, checking further I realized the vsync function was throwing errors and aborting, when it could be falling back on its other checks.

Langerz82, 2022-12-13:
This should be fixed and implemented as a patch in 4.6.

closed 2022-12-13 by Langerz82