EmuELEC/EmuELEC #866: [OGS] 4.4 "reset EE script and binaries to default" also resets config and themes

PikachuEXE, 2022-03-19:
### Env
EMUELEC version: `v4.4`
Update method: From `4.3`, put `tar` file into SD card `.update` folder

### Issues

#### 1. `RESET EMUELEC SCRIPTS AND BINARIES TO DEFAULT` also resets unexpected things
Such as timezone, Wifi settings, theme set

This has not happened in previous **test** versions (like `v4.4-TEST-02152022`)

### Notes
No issue found **just after** updated to `v4.4`

shantigilbert, 2022-03-19:
Did you do the reset before, or after updating? if it was before then this was a known issue since 4.3 had a problem when creating the backup file.

if it was after updating, then can you check `/storage/roms/backup` to see if there is a file named `restored....` and you can check the contents of the file to see if your settings are there.

PikachuEXE, 2022-03-20:
Yes I can find a file but I am not sure what to do with it
I have checked [Wiki - How to update](https://github.com/EmuELEC/EmuELEC/wiki/How-to-update)
[image]

shantigilbert, 2022-03-20:
Are you sure you selected `reset emuelec scripts and binaries to default` ? and not `Reset system to default config` ?

PikachuEXE, 2022-03-20:
To be sure I change some settings again like timezone & theme and start `reset emuelec scripts and binaries to default` again
**After** first binary reset, `reset emuelec scripts and binaries to default` would **not** erase my settings
But **before** first binary reset (upgraded from `4.3` to `4.4`), `reset emuelec scripts and binaries to default` **would** reset all the config

shantigilbert, 2022-03-20:
So like I said that was an issue with 4.3, you seemed to have run it before the update (as the tutorial says) since there was a bug in 4.3 and the script is from 4.3 that is why this happened. I will post a note about this, but this bug is fixed on 4.4

PikachuEXE, 2022-03-21:
OK any manual action to fix/workaround this issue?
Manually backing up the scripts from `4.4` and restore them after I revert back to 4.3 and update to 4.4?
If this is the way I need the location of the scripts to try them out.
Thanks.

shantigilbert, 2022-03-21:
> OK any manual action to fix/workaround this issue? Manually backing up the scripts from `4.4` and restore them after I revert back to 4.3 and update to 4.4? If this is the way I need the location of the scripts to try them out. Thanks.

Why would you want to do that? just copy your settings from 4.3 to 4.4 manually, basically everything in `/emuelec/configs` and your theme I guess

PikachuEXE, 2022-03-21:
OK I will try that later thx!

PikachuEXE, 2022-03-21:
OK restoring seems working...
I guess since it's 4.3 issue I will close this one

closed 2022-03-21 by PikachuEXE