EmuELEC/EmuELEC #868: No sound on amlogic 905w & S905x on Emuelec 4.4

planeta2325, 2022-03-20:
# Hello emuelec Team today install  4.4 EmuELEC-Amlogic-ng.aarch64-4.4-Generic on 2 model one with Amlogic 905x and other with 905w in the 2 boxes does not have volume.
i try erase the micro sd and reinstall and nothing i use the boot files with 281 number.

shantigilbert, 2022-03-20:
empty the contents of the file `/storage/.config/asound.conf`

qphoria, 2022-03-20:
I have audio but had to switch to audio 0,1 for my S905x3 boxes.

xgrind, 2022-03-20:
Same thing happens here, but it worked with emuelec 4.3

mavisguo, 2022-03-21:
Can amlogic s905m2 use Emuelec 4.4?

xgrind, 2022-03-25:
> empty the contents of the file `/storage/.config/asound.conf`

I did that and it fixed the problem with the audio. Thanks.

fantas21, 2022-03-26:
905w uses amlogic generic aarch64, not "amlogic-ng"

romanaksenov, 2022-04-10:
> empty the contents of the file `/storage/.config/asound.conf`
working, but video preview still without sound on my s905x3 box.

Langerz82, 2022-04-20:
In emulationstation go to settings audio device > and changing auto to one of the others and see if it makes any difference.

d-apps, 2022-04-30:
I found something, when I leave the system idle for some time in the emuelec menu and then when I press back button :O the frontend music works but just until I restart the system.

AlienatedDiego, 2022-04-30:
> Encontré algo, cuando dejo el sistema inactivo por un tiempo en el menú de emuelec y luego cuando presiono el botón Atrás, la música de la interfaz funciona, pero solo hasta que reinicio el sistema.
> 
> Intenté vaciar el contenido del archivo /storage/.config/asound.conf más tarde y el problema persiste.
> 
> @shantigilbert Editar: el código que reproduce la música cuando presiona cualquier botón para detener el protector de pantalla, corrige la música de la interfaz. Establecí la configuración del protector de pantalla para que comience después de 1 minuto, después de 1 minuto, el protector de pantalla se inicia, luego presiono cualquier botón y la música funciona.

AlienatedDiego, 2022-04-30:
this is correct, sound recovered in ES when waking up video screensaver and it happens only in 4.4 & 4.5 versions (S905w device).

alijani1, 2022-08-20:
I made my asound.conf file blank and no luck after reboot. Screen saver exit also did not have any effect,. then I tried setting audio device to 0,1 and still no audio on front end.   I also noticed that in some games there is no audio and in others there is.  games i can exit with start+select at same time like zaxxon have no audio. games that i need to press left and right analog control buttons on that bring up reatroarch do have sound.    No matter what I do i can  get navigation of video sounds on front end and some of the games.  I hope there is a better solution as this is marked as fixed and its not.

AlienatedDiego, 2022-08-21:
> I made my asound.conf file blank and no luck after reboot. Screen saver exit also did not have any effect,. then I tried setting audio device to 0,1 and still no audio on front end. I also noticed [...]

this is a problem for soc amlogic old s905x/w using image 4.4/4.5 NG versions, if you don't want to deal with this use older versions (non NG) or latest Amlogic-old test https://github.com/EmuELEC/EmuELEC-tests/releases/download/v4.6-TEST-07082022/EmuELEC-Amlogic-old.aarch64-4.6-TEST-07082022-Generic.img.gz

Langerz82, 2022-12-12:
If it's still an issue with 4.6 please re-open.

closed 2022-12-12 by Langerz82