EmuELEC/EmuELEC #876: Usb drive reading speed bug?

edouardlicn123, 2022-03-24:
Hello everyone.I am using a S922X machine(not a famous one)and some USB drives.I can promise that all my USB drive are working well.

I tried some 16G and 32G drives first with 4.4,and all failed with the "second step" of the first bootup,means that after the partition job finished and reboot,then going into the system again.

It shows that:

It failed to complete the bootup.First it run 2 jobs with a great time,more than 3 mins.then it will stop on some step.And after several trys it runs into the system but with many issues.And then I checked the files in the storage,it shows that many files are in 0 kb size,means that they are failed to write in but still have the file in it.

And when I change the STORAGE partition into FAT32 format,on the second step it shows "failed to mount Assets diretory","failed to mount Cores diretory",ect.And still feezed on some step.

Then I tried to search on google and fine a video descript the same problem with me. https://www.youtube.com/watch?v=d8YZsnFf2vI .But he is using another machine so I can not use the "slow emmc" dtb way.And I tried to use a 8G USB drive as he said,it finally works.I can go into the system without any problem.

What I want to say is that I think it is a bug and can anyone who can fix it?Or is there any way to solve?As you know not everyone like me have a 8G USB disk.

277383761, 2022-03-24:
友情帮顶,不保存是否怀孕

edouardlicn123, 2022-03-25:
Is there any one who could write a slowemmc version of S992X x88king dtb file?

closed 2022-04-18 by shantigilbert