EmuELEC/EmuELEC #877: emuelec4. 4 CSR4. 0 Bluetooth cannot connect

bruce0746, 2022-03-25:
CSR4.0 USB external Bluetooth. The chip is Qualcomm 8510a10. In the version emuelec3.1-4.3, the blue light will flash all the time after power on, and it can connect PS3 and PS4 controller normally. However, version 4.4 changed the working mode. After entering the system, the blue light of Bluetooth indicator flashes for ten seconds and then stops flashing. It looks like it doesn't work and can't connect to the Bluetooth game controller. I also tried to search for Bluetooth devices in the game controller settings in the main menu. During the search, the Bluetooth indicator flashes for about ten seconds, and then there is no response.

shantigilbert, 2022-04-03:
Please post dmesg logs

bruce0746, 2022-04-03:
/storage$ dmesg
[    0.000000@0]d Booting Linux on physical CPU 0x0
[    0.000000@0]d Linux version 4.9.269 (shantigilbert@ubuntu) (gcc version 10.3.0 (GCC) ) #1 SMP PREEMPT Wed Mar 16 18:30:53 CST 2022
[    0.000000@0]d Boot CPU: AArch64 Processor [411fd050]
[    0.000000@0]d CoreELEC dt-id: sm1_s905x3_2g
[    0.000000@0]d Machine model: Amlogic
[    0.000000@0]d 	07400000 - 07500000,     1024 KB, ramoops@0x07400000
[    0.000000@0]d __reserved_mem_alloc_size, start:0x0000000005000000, end:0x0000000005400000, len:4 MiB
[    0.000000@0]d 	05000000 - 05400000,     4096 KB, linux,secmon
[    0.000000@0]d __reserved_mem_alloc_size, start:0x0000000077800000, end:0x0000000078000000, len:8 MiB
[    0.000000@0]d 	77800000 - 78000000,     8192 KB, linux,meson-fb
[    0.000000@0]d 	78000000 - 80000000,   131072 KB, linux,ion-dev
[    0.000000@0]d 	75c00000 - 77800000,    28672 KB, linux,ion-fb
[    0.000000@0]d Reserved memory: created DMA memory pool at 0x0000000075c00000, size 0 MiB
[    0.000000@0]d 	75c00000 - 75c00000,        0 KB, linux,ppmgr
[    0.000000@0]d 	58800000 - 73800000,   442368 KB, linux,codec_mm_cma
[    0.000000@0]d 	75c00000 - 75c00000,        0 KB, linux,codec_mm_reserved
[    0.000000@0]d 	56800000 - 58800000,    32768 KB, linux,vm0_cma
[    0.000000@0]d cma: Reserved 8 MiB at 0x0000000075400000
[    0.000000@0]d On node 0 totalpages: 524288
[    0.000000@0]d   DMA zone: 8192 pages used for memmap
[    0.000000@0]d   DMA zone: 0 pages reserved
[    0.000000@0]d   DMA zone: 524288 pages, LIFO batch:31
[    0.000000@0]d psci: probing for conduit method from DT.
[    0.000000@0]d psci: PSCIv1.0 detected in firmware.
[    0.000000@0]d psci: Using standard PSCI v0.2 function IDs
[    0.000000@0]d psci: MIGRATE_INFO_TYPE not supported.
[    0.000000@0]d psci: SMC Calling Convention v1.1
[    0.000000@0]d percpu: Embedded 21 pages/cpu s47576 r8192 d30248 u86016
[    0.000000@0]d pcpu-alloc: s47576 r8192 d30248 u86016 alloc=21*4096
[    0.000000@0]d pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000@0]d Detected VIPT I-cache on CPU0
[    0.000000@0]d Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 516096
[    0.000000@0]d Kernel command line: init=/init console=ttyS0,115200 no_console_suspend earlyprintk=aml-uart,0xff803000 ramoops.pstore_en=1 ramoops.record_size=0x8000 ramoops.console_size=0x4000 otg_device=0 reboot_mode_android=normal logo=osd0,loaded,0x3d800000 vout=1080p60hz,enable panel_type=lcd_1 lcd_ctrl=0x00000000 hdmitx=,444,12bit hdmimode=1080p60hz frac_rate_policy=1 hdmi_read_edid=1 cvbsmode=576cvbs osd_reverse=0 video_reverse=0 irq_check_en=0 androidboot.selinux=permissive androidboot.firstboot=0 jtag=disable androidboot.hardware=amlogic androidboot.serialno=1234567890 mac=90:0e:b3:26:c3:01 androidboot.mac=90:0e:b3:26:c3:01 BOOT_IMAGE=kernel.img boot=LABEL=EMUELEC disk=LABEL=STORAGE console=ttyS0,115200 console=tty0 no_console_suspend        vout=1080p60hz,en logo=osd0,loaded,0x3d800000 frac_rate_policy=0 native_4k_gui=0 hdr_policy=1 quiet systemd.show_status=auto
[    0.000000@0]d fb: osd0
[    0.000000@0]d fb: loaded
[    0.000000@0]d fb: 0x3d800000
[    0.000000@0]d vout: 1080p60hz
[    0.000000@0]d vout: enable: 1
[    0.000000@0]d lcd: panel_type: lcd_1
[    0.000000@0]d lcd: lcd_ctrl: 0x00000000
[    0.000000@0]d vpp_axis_reverse: bootargs is 0
[    0.000000@0]d DI: di_read_canvas_reverse: bootargs is 0.
[    0.000000@0]d phlock_phase_config: bootargs is 0.
[    0.000000@0]d vout: 1080p60hz
[    0.000000@0]d vout: en: 1
[    0.000000@0]d fb: osd0
[    0.000000@0]d fb: loaded
[    0.000000@0]d fb: 0x3d800000
[    0.000000@0]d am_vecm: boot hdr_policy: 1
[    0.000000@0]d PID hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000@0]d Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes)
[    0.000000@0]d Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000@0]d Memory: 1366360K/2097152K available (13246K kernel code, 2072K rwdata, 10016K rodata, 5056K init, 1430K bss, 104104K reserved, 626688K cma-reserved)
[    0.000000@0]d Virtual kernel memory layout:
[    0.000000@0]d     modules : 0xffffff8000000000 - 0xffffff8008000000   (   128 MB)
[    0.000000@0]d     vmalloc : 0xffffff8008000000 - 0xffffffbebfff0000   (   250 GB)
[    0.000000@0]d       .text : 0xffffff8009080000 - 0xffffff8009d70000   ( 13248 KB)
[    0.000000@0]d     .rodata : 0xffffff8009d70000 - 0xffffff800a740000   ( 10048 KB)
[    0.000000@0]d       .init : 0xffffff800a740000 - 0xffffff800ac30000   (  5056 KB)
[    0.000000@0]d       .data : 0xffffff800ac30000 - 0xffffff800ae36008   (  2073 KB)
[    0.000000@0]d        .bss : 0xffffff800ae36008 - 0xffffff800af9bbdc   (  1431 KB)
[    0.000000@0]d     fixed   : 0xffffffbefe7fd000 - 0xffffffbefec00000   (  4108 KB)
[    0.000000@0]d     PCI I/O : 0xffffffbefee00000 - 0xffffffbeffe00000   (    16 MB)
[    0.000000@0]d     vmemmap : 0xffffffbf00000000 - 0xffffffc000000000   (     4 GB maximum)
[    0.000000@0]d               0xffffffbf00000000 - 0xffffffbf02000000   (    32 MB actual)
[    0.000000@0]d     memory  : 0xffffffc000000000 - 0xffffffc080000000   (  2048 MB)
[    0.000000@0]d SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000@0]d Preemptible hierarchical RCU implementation.
[    0.000000@0]d 	Build-time adjustment of leaf fanout to 64.
[... 1349 lines cut ...]
[   19.899665@3]- hdmitx: hw: set audio
[... cut ...]
[   32.022348@3]- RTW: rtl8188f_fill_default_txdesc(wlan0): SP Packet(0x0806) rate=0x0
[   44.023625@3]- RTW: rtl8188f_fill_default_txdesc(wlan0): SP Packet(0x0806) rate=0x0
[   50.468204@1]- RTW: rtl8188f_fill_default_txdesc(wlan0): SP Packet(0x0806) rate=0x0
[   62.022098@3]- RTW: rtl8188f_fill_default_txdesc(wlan0): SP Packet(0x0806) rate=0x0
[   62.500129@1]- fb: mem_free_work, free memory: addr:800000
[   74.022071@1]- RTW: rtl8188f_fill_default_txdesc(wlan0): SP Packet(0x0806) rate=0x0
[   86.022069@0]- RTW: rtl8188f_fill_default_txdesc(wlan0): SP Packet(0x0806) rate=0x0

bruce0746, 2022-04-21:
After continuous tests, it turns out that the Guo Hua PS3 gamepad Bluetooth controller is no longer supported after 4.3. I want to replace the Bluetooth device files in system version 4.4. Can you tell me which files need to be replaced?

zouplus, 2022-09-10:
I have the same problem on emuelec 4.5

zouplus, 2022-09-10:
Have you solved it

zouplus, 2022-09-10:
where is the log path?

Langerz82, 2023-06-01:
Try this:
https://github.com/EmuELEC/EmuELEC/issues/1083#issuecomment-1571479576

Langerz82, 2023-06-03:
Please post in this thread if you have any Bluetooth Controller pairing issues:
https://github.com/EmuELEC/EmuELEC/issues/1194

closed 2023-06-03 by Langerz82