EmuELEC/EmuELEC #878: [v4.4] There is a problem with font display displayed in EmuELEC v4.4.

british-choi, 2022-03-26:
EmuELEC Version : v4.4
Device : Odroid N2+

The font is displayed like the picture below.

Ports game list
[image]

File Manager
[image]

british-choi, 2022-04-02:
I found the commit causing the first problem. (Ports game list)
[emuelec-emulationstation]
[GLES20 renderer : New optimisations. Thanks to rtissera](https://github.com/EmuELEC/emuelec-emulationstation/commit/73fc3bbb3c4f7903ad68b5236f7d7bd9c2eeeae1)

shantigilbert, 2022-04-03:
> I found the commit causing the first problem. (Ports game list) [emuelec-emulationstation] [GLES20 renderer : New optimisations. Thanks to rtissera](https://github.com/EmuELEC/emuelec-emulationstation/commit/73fc3bbb3c4f7903ad68b5236f7d7bd9c2eeeae1)

Can you try and compile with GLES1 instead of GLES2 and test if the same thing happens?

british-choi, 2022-04-03:
@
In GLES1, the problem does not occur.
It only happens in GLES2

> > I found the commit causing the first problem. (Ports game list) [emuelec-emulationstation] [GLES20 renderer : New optimisations. Thanks to rtissera](https://github.com/EmuELEC/emuelec-emulationstation/commit/73fc3bbb3c4f7903ad68b5236f7d7bd9c2eeeae1)
> 
> Can you try and compile with GLES1 instead of GLES2 and test if the same thing happens?
> 
> `PKG_CMAKE_OPTS_TARGET=" -DENABLE_EMUELEC=1 -DDISABLE_KODI=1 -DENABLE_FILEMANAGER=1 -DGLES2=0`

In GLES1, the problem does not occur.
It only happens in GLES2

Langerz82, 2024-02-07:
I have noticed it only occurs when you launch a game or emulator then get back into emulationstation. Restarting ES fixes the issue. So I guess what it calls during a restart need to be called during the script end-game event and maybe it should fix the issue? I tried a few things but gave up as I am not sure how to fix it.

Langerz82, 2024-08-30:
Finally came up with a fix for this see:
https://github.com/EmuELEC/emuelec-emulationstation/pull/100

To test you need to point the emulationstation package file in EmuELEC, replace the following too:
```
PKG_NAME="emuelec-emulationstation"
PKG_VERSION="4803c74b1a1f97801a691afbcbb8df4053f27f57"
PKG_GIT_CLONE_BRANCH="EE-font-fix-6"
PKG_REV="1"
PKG_ARCH="any"
PKG_LICENSE="GPL"
PKG_SITE="https://github.com/Langerz82/emuelec-emulationstation"
PKG_URL="${PKG_SITE}.git"
PKG_DEPENDS_TARGET="toolchain SDL2 freetype freeimage vlc rapidjson ${OPENGLES} SDL2_mixer fping p7zip espeak"
PKG_SECTION="emuelec"
PKG_SHORTDESC="Emulationstation emulator frontend"
PKG_BUILD_FLAGS="-gold"
GET_HANDLER_SUPPORT="git"
```

Langerz82, 2024-09-01:
Fixed in latest version, will be included in Version 4.9 when it's available.

closed 2024-09-01 by Langerz82