EmuELEC/EmuELEC #879: Scummvm - controls not working

SzymonZy, 2022-03-27:
Odroid n2 scummvmsa controls not working event od starter from shell. When starter from emulationstation detected gamę startą but keyboard and mouse does not work. When started from shell keyboard entries arę passing to console(?) window making interface scroll up.

SzymonZy, 2022-04-02:
Can any one confirm that after upgrade from 4.3 to 4.4 scummvm still runs? 
In may case update broke only scummvm, all other emu I use work.

SzymonZy, 2022-04-03:
Downgraded to 4.3 - now scummvm run fine.

eviltroooper, 2022-04-16:
Is it possible to change the scummvm binary to the 2.5.1? Anyhow? Seems like read only.. 
I also tried to figure out the Problem, but no error on debug, and it’s also not working when I start scummvm via CLI, so I think the problem is scummvm binary it self, can someone check this please?

shantigilbert, 2022-04-16:
@eviltroooper if you have the binary place it in /emuelec/bin and it should take priority

eviltroooper, 2022-04-16:
Okay I try to find it. It seems I can’t download arm64 bin on scummvm..

eviltroooper, 2022-04-18:
Mhh only controller is working in scummvm now, any ideas why he can’t find my keyboard and mouse?

eviltroooper, 2022-04-18:
It seems the joystick is hardcoded, I run out of ideas.. 
I have no idea.. just switch back to 4.3…

eviltroooper, 2022-04-18:
I‘m out :( 
new scummvm binary will not work with 4.3.

i have no idea what happens between 4.3 and 4.4..
I don’t wanna use controller for scummvm..

shantigilbert, 2022-04-18:
I can confirm that for some reason Scummvm stopped working with keyboard and mouse, joystick works fine. But I have no idea why, so in the meantime use the Retroarch core instead.

eviltroooper, 2022-04-18:
Something changed in the compiling? 
i searched in scummvm Forum but nothing found here..

eviltroooper, 2022-04-22:
@shantigilbert 
Maybe you can ask via bugtracker, I think it makes no sense that I do that..
https://bugs.scummvm.org/report/1

eviltroooper, 2022-04-23:
@shantigilbert
https://forums.scummvm.org/viewtopic.php?t=16565

maybe this helps? They ask for /.configure output.

eviltroooper, 2022-04-25:
don't know if it helps.
but when i try to start the scummvm 2.60 (from emuelec 4.5) on emuelec 4.3 this error apears:

symbol lookup error: ./scummvm: undefined symbol: SDL_SetWindowMouseRect

shantigilbert, 2022-04-25:
> don't know if it helps. but when i try to start the scummvm 2.60 (from emuelec 4.5) on emuelec 4.3 this error apears:
> 
> symbol lookup error: ./scummvm: undefined symbol: SDL_SetWindowMouseRect

Try copying the old libsdl libraries from 4.3 to /emuelec/lib, then first try the new Scummvm and see if that fixes the issue, if not try the old scummvm

eviltroooper, 2022-04-26:
Hi, thanks, I will give it a try

eviltroooper, 2022-04-26:
with the 4.3 LIB and scummvm 2.6 i got this error 
./scummvm-2-6: undefined symbol: SDL_SetWindowMouseRect

with old scummvm it works

if i use the LIB from 4.5 i also got NO Mouse on scummvm 2.4

shantigilbert, 2022-04-26:
Maybe scummvm does not support sdl 2.20?

eviltroooper, 2022-04-28:
Is it possible to compile scummvm with old SDL version from emuelec 4.3?

shantigilbert, 2022-04-28:
This is definitely a SDL issue, as I compiled git version `b21fd0c9aa075915658d65c1c61e9ee2fb23d363` with SDL 2.0.9 and it works fine (if I use the SDL2 2.0.9 library in /emuelec/lib) but I am not going to revert SDL, This needs to be fixed in Scummv (I think, so all the other programs that use keyboard with SDL work fine) because if I compile the same exact version with only the library change or if I use the SDL2.20 library instead then the mouse/keyboard stops working.  

You can test it with this binary
[scummvm.zip](https://github.com/EmuELEC/EmuELEC/files/8586021/scummvm.zip)

eviltroooper, 2022-04-29:
Now it comes really strange, 
If I put the SDL2 2.0.9 library in /emuelec/lib it will stay on emulation station bootloop…
:(

eviltroooper, 2022-04-29:
I already opened a ticket von scummvm..
We will see

eviltroooper, 2022-04-29:
https://bugs.scummvm.org/ticket/13456#

It seems they want to move the problem to emuelec again 
:(
Is it possible that you also can comment there ?

thanks

shantigilbert, 2022-04-29:
This should not be considered a fix, but its a workaround. Delete all other libs in /emuelec/lib and scummvm binaries in /emuelec/bin then unzip this in `/emuelec` 

[scummvm_SDL2.0.9.zip](https://github.com/EmuELEC/EmuELEC/files/8594591/scummvm_SDL2.0.9.zip)

> Is it possible that you also can comment there ?

There is nothing I can say that you haven't already, they already said they will not look into the issue since its not an official build.

eviltroooper, 2022-04-30:
@shantigilbert 
Thanks..
And sorry for this shit.. :)

shantigilbert, 2022-05-22:
> @shantigilbert Thanks.. And sorry for this shit.. :)

I have been searching for more info on this, it seems the issue is only with EmuELEC. 

The patch that we use for SDL to enable Mali seems to be the issue, unfortunately this seems to only affect scummvm, so something changed between SDL2.0.9 and SDL 2.0.22 and Scummvm that the patch no longer detects mouse/keyboard  functions (again only for Scummvm) but SDL 2.0.22 on other devices that do not use this patch works fine. So as ugly as this might be, the workaround might be the only solution we have.

dajoho, 2023-12-15:
Hello,

I have spent many hours trying to debug this over the last few months. I have a Sunxi H6 SOC, with mali gpu and mali-patched `SDL 2.0.9`. ScummVM worked fine up to version `2.5.1` and the mouse/keyboard suddenly stopped working on `2.6.0` upwards. 3D games worked in `2.5.1` but were never hardware accelerated (they always used the software renderer), while the ScummVM menus always worked in high-resolution accelerated-mode using GLES2. I have not yet patched the latest SDL as my `2.0.9` has extra 'special-sauce' which also needs to be ported first.

After trawling through the commits, I discovered that ScummVM migrated between `2.5.1` and `2.6.0`, from `GLEW` to `GLAD`. Prior to the migration, code for 3D in the menu and 3D in games were separate from each other and now they have been re-factored to work together.

The first part of the migration: 

`30.12.2021` : https://github.com/scummvm/scummvm/commit/21b0d6ded18e6994e43c316759e4ffdb487f244d

Additionally, my SOC suffered from a seemingly unrelated bug that prevented GLSL version codes being parsed correctly. This was corrected at a later date in this commit:

`03.04.2022` : https://github.com/scummvm/scummvm/commit/bc4e7662cc23e2be6c45c54d216dd21bbb9fe5bb

Using this information, I bisected the code using `git bisect` and, upon each iteration, cherry-picked `bc4e7662cc23e2be6c45c54d216dd21bbb9fe5bb` to fix the GLSL bug, compiled ScummVM and tested on the SOC. After several nights compiling each bisected revision, I finally landed on this commit:  

`18.01.2022` : https://github.com/scummvm/scummvm/commit/a9418f8f6343151ff3bf98df74a4b44c714c1b3e

..which modified the GL/GLES detection in the configure script and was a general cleanup to make it work correctly together with the GLAD refactoring.

Comparing the generated `config.h` from `2.5.1` with one from after the GLAD-migration, I noticed that in `2.5.1` `USE_OPENGL_GAME` and `USE_OPENGL_SHADERS` were set to false, but in 2.6.0 the new `configure` detection code-tests set them, at least on my toolchain, to true. 

Going through the other changes in the `configure` script revealed these two lines, which can be manually set to mimic the behaviour of `2.5.1`:

`--disable-opengl-game-classic` and `--disable-opengl-game-shaders`

https://github.com/scummvm/scummvm/commit/a9418f8f6343151ff3bf98df74a4b44c714c1b3e#diff-90d08e583c4c9c6f391b2ae90f819f600a6326928ea9512c9e0c6d98e9f29ac2R951-R952

With these two configuration options set, mouse and keyboard are restored and I can compile right up to the latest `2.8.0pre` version of ScummVM without problems.

However, as in `2.5.1`, 3D games in the newer ScummVM versions are still restricted on my board to software rendering, which is sort of implied to be the correct behaviour in this comment: https://github.com/scummvm/scummvm/blob/a9418f8f6343151ff3bf98df74a4b44c714c1b3e/configure#L5526

I hope this helps you get your versions updated too. This was by far the worst bug-hunt I have ever had to deal with!

Dave

shantigilbert, 2023-12-15:
> Going through the other changes in the `configure` script revealed these two lines, which can be manually set to mimic the behaviour of `2.5.1`:
> 
> `--disable-opengl-game-classic` and `--disable-opengl-game-shaders`

Thank you so much for the thorough debugging!  I will test this hopefully today!

One question just out of curiosity, as EmuELEC does not run on the H6 (AFAIK), what distro are you using?

dajoho, 2023-12-15:
I'm not using a distro as such but rather a completely custom crosstool-toolchain and root-filesystem for this particular board (THEA500 Mini). It just happens to share a similar mali-patched sdl2 and hardware with some of the boards which are compatible with EmuELEC.

shantigilbert, 2023-12-15:
> THEA500 Mini

Awesome, never heard of that one, looks really cool!

shantigilbert, 2023-12-16:
@dajoho This worked perfectly!

closed 2023-12-16 by shantigilbert

dajoho, 2023-12-16:
Good to hear! One thing you may want to try (if you haven't already): I saw you added ` --disable-opengl-game`, which you may be able to remove. On my hardware only the other two flags were necessary. This should allow you to play Grim Fandango and EFMI in software mode, which is still pretty acceptable, at least on my SoC.

shantigilbert, 2023-12-18:
I tested Grim Fandango and it works fine, it was one of the games I tested, I also tested Indiana Jones Fate of Atlantis, Sam and Max and other random ones, all of them seem to work fine with gamepad and keybard+mouse... But I will try them again without the switch.

commit 131697f2 (2023-12-16): Scummvmsa: Bump to 1fa6fcd branch-2.8 / libretro to 0db091d

Thanks to @dajoho this fixes #879