EmuELEC/EmuELEC #880: IR configuration not working anymore on 4.4

pmsobrado, 2022-03-27:
Hi, since I discovered this wonderful project, I've been using my remote to shutdown the system thanks to this config from the official wiki:

[https://github.com/EmuELEC/EmuELEC/wiki/how%2Dto%2Duse%2DRC%2Dfom%2Dtvbox%3F](https://github.com/EmuELEC/EmuELEC/wiki/how-to-use-RC-fom-tvbox%3F)

But for some reason, on build 4.4 this does not work anymore. I have tested with CoreELEC 20, on which version 4.4 is based, and this configuration works there. I did not enable BL301 injection in CoreELEC's system settings, in case that's relevant.

So, EE 4.3 and CE 20 do accept the config of the wiki page posted above, but EE 4.4 seems not to.

One thing that may be worth mention is that on EE 4.3, the command `ir-keytable -p NEC,RC-5,RC-6,JVC,SONY -t` does output codes when I press buttons, as described on the wiki page posted above, but CE 20 and EE 4.4 does not.

Maybe one of you guys using remotes could test this? Thanks in advance.

miwasp, 2022-03-28:
I never used a remote in this way until now. So i grabbed my Tanix TX3 and tested a few things.
First i installed 4.3, got the remote.conf for this device, put it in /storage/.config and rebooted.
The command `ir-keytable -p NEC,RC-5,RC-6,JVC,SONY -t` does not work:
`/sys/class/rc/: No such file or directory
No devices found`
But I was able to switch the Tanix on and off with the remote.

Then i upgraded this device to 4.4 and it behaves exactly the same. I can still switch the Tanix on and off with the remote and `ir_keytable` still gives the same error.

pmsobrado, 2022-03-28:
That's even stranger.. I don't use remote.conf, but rc_maps.cfg and rc_keymaps folder files. Where did you grab your remote.conf? I can try that as well... I will also test updating from a working 4.3 to 4.4 and see what happens.

pmsobrado, 2022-03-28:
> I never used a remote in this way until now. So i grabbed my Tanix TX3 and tested a few things. First i installed 4.3, got the remote.conf for this device, put it in /storage/.config and rebooted. [...]

That's even stranger.. I don't use remote.conf, but rc_maps.cfg and rc_keymaps folder files. Where did you grab your remote.conf? I can try that as well... I will also test updating from a working 4.3 to 4.4 and see what happens. Thanks for the feedback!

miwasp, 2022-03-28:
> That's even stranger.. I don't use remote.conf, but rc_maps.cfg and rc_keymaps folder files. Where did you grab your remote.conf? I can try that as well... I will also test updating from a working 4.3 to 4.4 and see what happens. Thanks for the feedback!

I got the remote.conf from here [https://github.com/CoreELEC/remotes](https://github.com/CoreELEC/remotes)

pmsobrado, 2022-03-28:
> I got the remote.conf from here https://github.com/CoreELEC/remotes

Thanks, I ended up creating my own remote.conf with the help of this wiki:

https://github.com/CoreELEC/remotes/wiki/10.-Walk-through%3A-Create-remote.conf

Apparently that was it, on EE 4.4 remote.conf in /storage/.config folder seems to work! Thanks @miwasp for guiding me in the right direction. Just two comments though:

1. I think the command `ir-keytable -p NEC,RC-5,RC-6,JVC,SONY -t` does NOT work when there is a remote.conf file present. But on a fresh install, it should work. I guess this is not device dependant, but you have a different device than mine, so who knows.

2. One of the things I tried today was configuring the remote on a fresh 4.3 install (only this config, no other) and the upgrade to 4.4. Well, the update process crashed and the installation got corrupted. I write this here because that's the second time it happens to me. Maybe 4.3 -> 4.4 is broken for my device? In any case, that's a different issue and since I'm starting fresh from 4.4, it doesn't bother me much.

Thanks again!

closed 2022-03-28 by pmsobrado