EmuELEC/EmuELEC #881: V4.3 There is such an error in the compilation process, and I ask experts to help me solve it.

xtgrrk, 2022-03-28:
HTTP request sent, awaiting response... 404 Not Found
2022-03-28 08:20:43 ERROR 404: Not Found.

Cannot get pcre2 sources : https://ftp.pcre.org/pub/pcre/pcre2-10.33.tar.bz2
Try later!
FAILURE: scripts/install pcre2 has failed!
[259/488] [FAIL] install pcre2

The following logs for this failure are available:
  stdout: /home/emuelec/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4.3/.threads/logs/262/stdout
  stderr: /home/emuelec/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4.3/.threads/logs/262/stderr

parallel: This job failed:
package_worker 1 262 488 'install pcre2'
Parallel build failure - see log for details. Time of failure: Mon Mar 28 08:20:43 PDT 2022
Makefile:12: recipe for target 'image' failed
make: *** [image] Error 1

british-choi, 2022-03-28:
Url is changed (ftp.pcre.org site is not working)
refer below link
https://raw.githubusercontent.com/EmuELEC/EmuELEC/dev/packages/addons/addon-depends/pcre2/package.mk

shantigilbert, 2022-03-29:
Duplicate: https://github.com/EmuELEC/EmuELEC/issues/774

closed 2022-03-29 by shantigilbert