EmuELEC/EmuELEC #882: Missing savestate support with some cores in 4.4

miwasp, 2022-03-29:
The following cores don't have savestate support in EmuELEC 4.4:
- flycast_32b
- mupen64plus_32b
- mupen64plus_next_alt
- parallel_n64_32b
- pcsx_rearmed_32b
- uae4arm_32b
- vice_x64dtv

Those cores never had an .info file until now. Since retroarch version 1.10.1 this is handled differently.
Quote out of issue #13703 (https://github.com/libretro/RetroArch/issues/13703)

> We added an option to automatically hide/disable certain features if the core doesn't support them (mostly related to savestates at the moment, including rewind and runahead), and that feature is powered by the info files.

So if a core does not have an .info file then it does not have savestate support. I created .info files for these cores and uploaded them to discord (https://discord.com/channels/570131631260696576/570131631713550337/958309082056491028).

I'm not sure how to integrate this in the build process, so no PR for now.

dewenhsu, 2022-03-29:
for mame2016, 
edit mame2016_libretro.info
savestate = "true"
to get savestate working

shantigilbert, 2022-04-08:
https://github.com/EmuELEC/EmuELEC/commit/63e18ad00b047a0a85cdcfaf481cb2cf74b007a9

closed 2022-04-18 by shantigilbert