EmuELEC/EmuELEC #883: The storage partition is too small to use the options in the download menu

leszekm1, 2022-04-02:
On any brand new vanilla install go to the updates/download menu.  Download and install the 17 themes there.  After all themes are installed you will not be able to install the arcade bezels from the bezels menu because you will run out of storage during the downlaod of the bezels zip file.

shantigilbert, 2022-04-03:
This is not a bug, its intended only 2GB is available, rest of the storage is for ROMS on the third partition (EEROMS)

closed 2022-04-03 by shantigilbert

leszekm1, 2022-04-03:
Then perhaps a warning that if you install the themes you will not be able to install the bezels?

shantigilbert, 2022-04-03:
Bezels install to roms partition, so not sure what your setup is. You can also create a folder called `themes` in the roms partition, and themes will download there. But honestly, who installs 17 themes?

leszekm1, 2022-04-03:
I understand.  Bezels do install into roms, but **duing the download** the ZIP file is stored temporarily on the storage partition so the download fails.  This repros on **any brand new install** of emulec.  There is nothing special about my install.  I do agree that it is ridiculous to install all themes, but it's the very first thing that pops up on the download/update menu so that is what I did on my first install.  I then was unable to install any bezels so I had to reset emuelec to "initial settings" in order to install the bezels.

shantigilbert, 2022-04-03:
Got it, I was not aware (or did not remember) it goes to tmp first, will change it for the next version. :)

leszekm1, 2022-04-03:
Awesome! Thank you so much. :)

shantigilbert, 2022-04-03:
But you can also create the `themes` folder on the roms partition, and they will take no space on the main one.

leszekm1, 2022-04-03:
That is good to know.  TY.  Just to double check my understanding - If I create the "themes" folder on the roms partition then when I install the themes from the update/download menu they will install on the roms partition instead of storage - correct?

shantigilbert, 2022-04-03:
Correct

leszekm1, 2022-04-03:
Roger! TY!