EmuELEC/EmuELEC #893: Radxa Zero crashes upon starting 3d games

thefunkdoctor, 2022-04-12:
I have had no issues running 4.3 for awhile on my radxa zero

Attempted a fresh 4.4 install and everything seems to work the same except games wont load

specifically psp/n64/saturn or any 3d game seems to freeza the console once its about to load in game

i can usually start the game, get through the menus

but once i go to choose a level it freezes mid loading

any help is appreciated

shantigilbert, 2022-04-13:
It will be nearly impossible to help without logs

thefunkdoctor, 2022-04-27:
i recently picked up a radxa zero 2gb version
it works flawlessly with emuelec 4.4 and 4.5 while the 1gb version freezes on those versions, so perhaps its just the lack of ram

shantigilbert, 2022-04-27:
> i recently picked up a radxa zero 2gb version it works flawlessly with emuelec 4.4 and 4.5 while the 1gb version freezes on those versions, so perhaps its just the lack of ram

1GB might not be enough for 1080p on 3D emulators. But another thing to consider would be heat?

thefunkdoctor, 2022-04-27:
both units i tested had heatsinks

they also never went above 66 degrees in desktop testing so i dont imagine they would overheat emulating. Seeing as my 1gb model will run all day on emuelec 4.3
ive even let it play through a few anime episodes on the media player no problem

shantigilbert, 2022-04-27:
Then no idea, sorry. I do not have a 1gb device to test this, nor do you provide any type of logs.

thefunkdoctor, 2022-04-28:
2022-04-28 15:05:17	INFO	EmulationStation - v31, built Apr 18 2022 - 10:52:28
2022-04-28 15:05:17	INFO	Parsing XML file "/storage/.emulationstation/resources/genres.xml"...
2022-04-28 15:05:17	DEBUG	TextureData::load :/scroll_gradient.png
2022-04-28 15:05:17	DEBUG	ImageIO::loadFromMemoryRGBA32
2022-04-28 15:05:17	INFO	Window::init
2022-04-28 15:05:17	INFO	Creating window...
2022-04-28 15:05:17	INFO	Created window successfully.
2022-04-28 15:05:17	INFO	GL vendor:   ARM
2022-04-28 15:05:17	INFO	GL renderer: Mali-G31
2022-04-28 15:05:17	INFO	GL version:  OpenGL ES 3.2 git.c8adbf9.ad00b04c1b60847de257177231dc1a53
2022-04-28 15:05:17	INFO	GL shading:  OpenGL ES GLSL ES 3.20
2022-04-28 15:05:17	INFO	GL exts:     GL_ARM_rgba8 GL_ARM_mali_shader_binary GL_OES_depth24 GL_OES_depth_texture GL_OES_depth_texture_cube_map GL_OES_packed_depth_stencil GL_OES_rgb8_rgba8 GL_EXT_read_format_bgra GL_OES_compressed_paletted_texture GL_OES_compressed_ETC1_RGB8_texture GL_OES_standard_derivatives GL_OES_EGL_image GL_OES_EGL_image_external GL_OES_EGL_image_external_essl3 GL_OES_EGL_sync GL_OES_texture_npot GL_OES_vertex_half_float GL_OES_required_internalformat GL_OES_vertex_array_object GL_OES_mapbuffer GL_EXT_texture_format_BGRA8888 GL_EXT_texture_rg GL_EXT_texture_type_2_10_10_10_REV GL_OES_fbo_render_mipmap GL_OES_element_index_uint GL_EXT_shadow_samplers GL_OES_texture_compression_astc GL_KHR_texture_compression_astc_ldr GL_KHR_texture_compression_astc_hdr GL_KHR_texture_compression_astc_sliced_3d GL_EXT_texture_compression_astc_decode_mode GL_EXT_texture_compression_astc_decode_mode_rgb9e5 GL_KHR_debug GL_EXT_occlusion_query_boolean GL_EXT_disjoint_timer_query GL_EXT_blend_minmax GL_EXT_discard_framebuffer GL_OES_get_program_binary GL_OES_texture_3D GL_EXT_texture_storage GL_EXT_multisampled_render_to_texture GL_OES_surfaceless_context GL_OES_texture_stencil8 GL_EXT_shader_pixel_local_storage GL_ARM_shader_framebuffer_fetch GL_ARM_shader_framebuffer_fetch_depth_stencil GL_ARM_mali_program_binary GL_EXT_sRGB GL_EXT_sRGB_write_control GL_EXT_texture_sRGB_decode GL_EXT_texture_sRGB_R8 GL_EXT_texture_sRGB_RG8 GL_KHR_blend_equation_advanced GL_KHR_blend_equation_advanced_coherent GL_OES_texture_storage_multisample_2d_array GL_OES_shader_image_atomic GL_EXT_robustness GL_EXT_draw_buffers_indexed GL_OES_draw_buffers_indexed GL_EXT_texture_border_clamp GL_OES_texture_border_clamp GL_EXT_texture_cube_map_array GL_OES_texture_cube_map_array GL_OES_sample_variables GL_OES_sample_shading GL_OES_shader_multisample_interpolation GL_EXT_shader_io_blocks GL_OES_shader_io_blocks GL_EXT_tessellation_shader GL_OES_tessellation_shader GL_EXT_primitive_bounding_box GL_OES_primitive_bounding_box GL_EXT_geometry_shader GL_OES_geometry_shader GL_ANDROID_extension_pack_es31a GL_EXT_gpu_shader5 GL_OES_gpu_shader5 GL_EXT_texture_buffer GL_OES_texture_buffer GL_EXT_copy_image GL_OES_copy_image GL_EXT_shader_non_constant_global_initializers GL_EXT_color_buffer_half_float GL_EXT_color_buffer_float GL_EXT_YUV_target GL_OVR_multiview GL_OVR_multiview2 GL_OVR_multiview_multisampled_render_to_texture GL_KHR_robustness GL_KHR_robust_buffer_access_behavior GL_EXT_draw_elements_base_vertex GL_OES_draw_elements_base_vertex GL_EXT_buffer_storage 
2022-04-28 15:05:17	INFO	 ARB_texture_non_power_of_two: MISSING
2022-04-28 15:05:18	DEBUG	ImageIO::loadFromMemoryRGBA32
2022-04-28 15:05:18	DEBUG	Glyph texture cache full, creating a new texture cache for ubuntu_condensed.ttf 91pt
2022-04-28 15:05:18	DEBUG	TextureData::load :/frame.png
2022-04-28 15:05:18	DEBUG	ImageIO::loadFromMemoryRGBA32
2022-04-28 15:05:18	ERROR	Resource path not found: :/splash.xml
2022-04-28 15:05:18	DEBUG	TextureData::load :/logo.png
2022-04-28 15:05:18	DEBUG	ImageIO::loadFromMemoryRGBA32
2022-04-28 15:05:18	INFO	Parsing XML file "/storage/.emulationstation/resources/mamenames.xml"...
2022-04-28 15:05:19	INFO	Parsing XML file "/storage/.emulationstation/resources/mamebioses.xml"...
2022-04-28 15:05:19	INFO	Parsing XML file "/storage/.emulationstation/resources/mamedevices.xml"...
2022-04-28 15:05:19	INFO	Loading system config file /storage/.emulationstation/es_systems.cfg...
2022-04-28 15:05:19	WARNING	  Unknown platform for system "amstradgx4000" (platform "amstradgx4000" from list "amstradgx4000")
2022-04-28 15:05:19	WARNING	System "imageviewer" has no games! Ignoring it.
2022-04-28 15:05:19	WARNING	System "arcade" has no games! Ignoring it.
2022-04-28 15:05:19	WARNING	System "amstradcpc" has no games! Ignoring it.
2022-04-28 15:05:19	WARNING	System "atari2600" has no games! Ignoring it.
2022-04-28 15:05:19	WARNING	System "amstradgx4000" has no games! Ignoring it.
2022-04-28 15:05:19	WARNING	System "mame" has no games! Ignoring it.
2022-04-28 15:05:19	WARNING	System "atari7800" has no games! Ignoring it.
2022-04-28 15:05:19	WARNING	System "atari5200" has no games! Ignoring it.
2022-04-28 15:05:19	WARNING	System "atarist" has no games! Ignoring it.
2022-04-28 15:05:19	DEBUG	Warning : could not find file "./core/fonts/ubuntu_bold.ttf" (which resolved to "/etc/emulationstation/themes/es-theme-alekfull-EmueELEC/_inc/menu/core/fonts/ubuntu_bold.ttf") 
2022-04-28 15:05:19	DEBUG	Warning : could not find file "./core/fonts/ubuntu_medium.ttf" (which resolved to "/etc/emulationstation/themes/es-theme-alekfull-EmueELEC/_inc/menu/core/fonts/ubuntu_medium.ttf") 
[... cut ...]
2022-04-28 15:06:16	DEBUG	TextureData::load /etc/emulationstation/themes/es-theme-EmuELEC-fundamental/snes/console.png
2022-04-28 15:06:16	DEBUG	TextureData::load /etc/emulationstation/themes/es-theme-EmuELEC-fundamental/snes/logo.svg
2022-04-28 15:06:16	DEBUG	ImageIO::loadFromMemoryRGBA32
2022-04-28 15:06:16	DEBUG	TextureData::load /etc/emulationstation/themes/es-theme-EmuELEC-fundamental/setup/console.png
2022-04-28 15:06:16	DEBUG	ImageIO::loadFromMemoryRGBA32
2022-04-28 15:06:16	DEBUG	TextureData::load /etc/emulationstation/themes/es-theme-EmuELEC-fundamental/setup/logo.svg

thefunkdoctor, 2022-04-28:
http://ix.io/3WvX

is this correct

thefunkdoctor, 2022-04-29:
Apr 29 06:51:42 EMUELEC systemd[1]: Finished Wait for Kernel Time Synchronisation.
Apr 29 06:51:42 EMUELEC systemd[1]: Reached target System Time Synchronized.
Apr 29 06:51:42 EMUELEC systemd[1]: Started Discard unused blocks once a week.
Apr 29 06:51:42 EMUELEC systemd[1]: Reached target Timer Units.
Apr 29 06:51:42 EMUELEC systemd[1]: Startup finished in 3.955s (kernel) + 16.291s (userspace) = 20.247s.
Apr 29 06:51:42 EMUELEC nmbd[3443]: [2022/04/29 06:51:42.966508,  0] ../../lib/util/become_daemon.c:135(daemon_ready)
Apr 29 06:51:42 EMUELEC nmbd[3443]:   daemon_ready: daemon 'nmbd' finished starting up and ready to serve connections
Apr 29 06:51:42 EMUELEC nmbd[3443]: [2022/04/29 06:51:42.990728,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Apr 29 06:51:42 EMUELEC nmbd[3443]:   query_name_response: Multiple (2) responses received for a query on subnet 192.168.0.28 for name WORKGROUP<1d>.
Apr 29 06:51:42 EMUELEC nmbd[3443]:   This response was from IP 192.168.0.10, reporting an IP address of 192.168.0.10.
Apr 29 06:51:43 EMUELEC python[4004]: Traceback (most recent call last):
Apr 29 06:51:43 EMUELEC python[4004]:   File "/usr/lib/python3.9/site-packages/dbus/bus.py", line 177, in activate_name_owner
Apr 29 06:51:43 EMUELEC python[4004]:   File "/usr/lib/python3.9/site-packages/dbus/bus.py", line 361, in get_name_owner
Apr 29 06:51:43 EMUELEC python[4004]:   File "/usr/lib/python3.9/site-packages/dbus/connection.py", line 652, in call_blocking
Apr 29 06:51:43 EMUELEC python[4004]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'org.bluez': no such name
Apr 29 06:51:43 EMUELEC python[4004]: During handling of the above exception, another exception occurred:
Apr 29 06:51:43 EMUELEC python[4004]: Traceback (most recent call last):
Apr 29 06:51:43 EMUELEC python[4004]:   File "/usr/bin/batocera/batocera-bluetooth-agent", line 446, in <module>
Apr 29 06:51:43 EMUELEC python[4004]:     obj = bus.get_object("org.bluez", "/org/bluez")
Apr 29 06:51:43 EMUELEC python[4004]:   File "/usr/lib/python3.9/site-packages/dbus/bus.py", line 241, in get_object
Apr 29 06:51:43 EMUELEC python[4004]:   File "/usr/lib/python3.9/site-packages/dbus/proxies.py", line 250, in __init__
Apr 29 06:51:43 EMUELEC python[4004]:   File "/usr/lib/python3.9/site-packages/dbus/bus.py", line 182, in activate_name_owner
Apr 29 06:51:43 EMUELEC python[4004]:   File "/usr/lib/python3.9/site-packages/dbus/bus.py", line 277, in start_service_by_name
Apr 29 06:51:43 EMUELEC python[4004]:   File "/usr/lib/python3.9/site-packages/dbus/connection.py", line 652, in call_blocking
Apr 29 06:51:43 EMUELEC python[4004]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.bluez was not provided by any .service files
Apr 29 06:51:43 EMUELEC systemd[1]: bluetooth-agent.service: Main process exited, code=exited, status=1/FAILURE
Apr 29 06:51:43 EMUELEC systemd[1]: bluetooth-agent.service: Failed with result 'exit-code'.
Apr 29 06:51:43 EMUELEC systemd[1]: bluetooth-agent.service: Scheduled restart job, restart counter is at 5.
Apr 29 06:51:43 EMUELEC systemd[1]: Stopped EmuELEC Bluetooth Agent.
Apr 29 06:51:43 EMUELEC systemd[1]: bluetooth-agent.service: Failed to get cgroup ID on cgroup /sys/fs/cgroup/unified/system.slice/bluetooth-agent.service, ignoring: Function not implemented
Apr 29 06:51:43 EMUELEC systemd[1]: Started EmuELEC Bluetooth Agent.
Apr 29 06:51:44 EMUELEC python[4045]: Traceback (most recent call last):
Apr 29 06:51:44 EMUELEC python[4045]:   File "/usr/lib/python3.9/site-packages/dbus/bus.py", line 177, in activate_name_owner
Apr 29 06:51:44 EMUELEC python[4045]:   File "/usr/lib/python3.9/site-packages/dbus/bus.py", line 361, in get_name_owner
Apr 29 06:51:44 EMUELEC python[4045]:   File "/usr/lib/python3.9/site-packages/dbus/connection.py", line 652, in call_blocking
Apr 29 06:51:44 EMUELEC python[4045]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'org.bluez': no such name
Apr 29 06:51:44 EMUELEC python[4045]: During handling of the above exception, another exception occurred:
Apr 29 06:51:44 EMUELEC python[4045]: Traceback (most recent call last):
Apr 29 06:51:44 EMUELEC python[4045]:   File "/usr/bin/batocera/batocera-bluetooth-agent", line 446, in <module>
Apr 29 06:51:44 EMUELEC python[4045]:     obj = bus.get_object("org.bluez", "/org/bluez")
Apr 29 06:51:44 EMUELEC python[4045]:   File "/usr/lib/python3.9/site-packages/dbus/bus.py", line 241, in get_object
Apr 29 06:51:44 EMUELEC python[4045]:   File "/usr/lib/python3.9/site-packages/dbus/proxies.py", line 250, in __init__
Apr 29 06:51:44 EMUELEC python[4045]:   File "/usr/lib/python3.9/site-packages/dbus/bus.py", line 182, in activate_name_owner
Apr 29 06:51:44 EMUELEC python[4045]:   File "/usr/lib/python3.9/site-packages/dbus/bus.py", line 277, in start_service_by_name
Apr 29 06:51:44 EMUELEC python[4045]:   File "/usr/lib/python3.9/site-packages/dbus/connection.py", line 652, in call_blocking
Apr 29 06:51:44 EMUELEC python[4045]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.bluez was not provided by any .service files
Apr 29 06:51:44 EMUELEC systemd[1]: bluetooth-agent.service: Main process exited, code=exited, status=1/FAILURE
Apr 29 06:51:44 EMUELEC systemd[1]: bluetooth-agent.service: Failed with result 'exit-code'.
[... cut ...]
Apr 29 06:52:33 EMUELEC emulationstation[4708]: round-trip min/avg/max = 13.604/13.604/13.604 ms
Apr 29 06:52:44 EMUELEC kernel: asoc-aml-card auge_sound: S/PDIF[1] Playback disable
Apr 29 06:52:44 EMUELEC kernel: spdif_b keep clk continuous
Apr 29 06:52:44 EMUELEC kernel: audio_ddr_mngr: frddrs[0] released by device ff642000.audiobus:spdif_b
Apr 29 06:52:44 EMUELEC kernel: Console: switching to colour frame buffer device 240x67

this was a log today on a fresh sd card 

i select a psp game and then it freezes when attempting to start

Langerz82, 2023-02-08:
the file we usually need with emulator crashing issues is in /emuelec/logs/emuelec.log . Anyway this issue may be outdated as it's for 4.4, If it's still an issue open up another thread.

closed 2023-02-08 by Langerz82