EmuELEC/EmuELEC #900: 720p resolution error on S905w after 4.5 clean install

marcus-abreu, 2022-04-20:
My S905w (tx3 mini) worked fine with 720p resolution until yesterday, when I did a clean install to 4.5 version. I just can't change the resolution to 720p without glitches.
Thanks.

Langerz82, 2022-04-22:
What glitches do you mean exactly? Is it a video change warning comes up? You can override it by just selecting yes.

marcus-abreu, 2022-04-22:
Thanks for the reply. It´s hard to describe, so I took some snapshots:
[image]
It happens after reboot the system.

AlienatedDiego, 2022-04-22:
it is correct, 720p is broken in 4.5 for Amlogic old(s905w) devices

Langerz82, 2022-04-23:
That looks like your resolution setting is not matching the emulationstation resolution setting. Make sure emulationstation resolution matches the output.
If you have a resolution set in: /flash/config.ini look for the key hdmimode and make sure it's commented out (prefix it with a #). You may have to remount the flash directory to make changes by doing:
```
mount -o rw,remount /flash
```

Also make sure there is no file: /storage/.config/EE_VIDEO_MODE if it's there just remove it.

Also in emulationstation you may have to set a custom resolution and put in: 1280x720p60hz instead of 720p60hz.

Once that is done be sure to restart your device.

Let me know how it goes.

Langerz82, 2022-04-23:
Hey, try this PR and see if it fixes your issue. mount_romfs.sh had a syntax error which may have been breaking the scripts and not allow setres.sh to run properly.

See:
https://github.com/EmuELEC/EmuELEC/pull/904

AlienatedDiego, 2022-04-23:
hi, I did the checks of your first answer with no success. I have no code knowledge, if there is any easy way to get the mount_romfs.sh file I could use it in emuelec/bin to check. thanks for your answers. regards

Langerz82, 2022-04-23:
ok so if your experiencing the issue can someone SSH in and type in the command:
```
emuelec_autostart.sh
```
If no errors are appearing please run the command:
```
bash -x emuelec_autostart.sh
```
And paste all the output in a gist here:
https://gist.github.com/
and provide the link so I can check the debug output? thanks.

AlienatedDiego, 2022-04-23:
EMUELEC:~ # emuelec\_autostart.sh  
rm: can't remove '/storage/.cache/services/bluez.conf': No such file or director                                                                                       y  
find: /media/\*/roms/: No such file or directory  
Using path  
find: /media/\*/roms/: No such file or directory  
find: /media/\*/roms/: No such file or directory  
mv: can't rename '/storage/roms': Device or resource busy  
mount: mounting /dev/mmcblk1p3 on /storage/roms failed: Device or resource busy

Langerz82, 2022-04-23:
ok so it looks like it's not generating any errors.

try going into EmulationStation, setting the resolution to 720p60hz and restarting the device so the bug occurs again.
While it's bugged, SSH in and type the command:
```
setres.sh 720p60hz
```
If the EmulationStation display is still incorrect, try:
```
setres.sh 1280x720p60hz
```
Also use a controller to move with the dpad so EmulationStation tries to refresh. If 1280x720p60hz works for it, we may have to add by default in ES, instead of just 720p60hz.

AlienatedDiego, 2022-04-23:
hey Josh it didn't work.

I think considering that there are more issues besides resolutions for this device and that using the same 4.5 ng image works fine for S905x3 and OdroidN2+ and that an alternative version for Amlogic_old with older kernel is in the kitchen, maybe we should wait until it is ready to try it. Shanti may know the origin of this and other problems.
Thank you for your interest in this matter.

Langerz82, 2022-04-23:
It looks like the ES resolution being picked up is 720p which is correct, however the actual display mode in linux is not changing properly and staying at full HD or more.
Try and run this command and post the output:
```
cat /sys/class/amhdmitx/amhdmitx0/rawedid | /usr/bin/edid-decode | grep -E '^ *VIC' | sed 's/ *VIC  *[0-9]*\://g' | sed 's/^ *//g' | sed 's/ * / /g' | cut -d'.' -f1 | sed 's/i /i/g' | sed 's/ /p/g' | sed -e 's/$/hz/g'
```
The supported resolutions should show up.

AlienatedDiego, 2022-04-23:
1920x1080p50hz  
1920x1080p60hz  
1920x1080p24hz  
1920x1080p25hz  
1920x1080p30hz  
1920x1080i50hz  
1920x1080i60hz  
1280x720p50hz  
1280x720p60hz  
720x576p50hz  
720x576p50hz  
720x480p59hz  
720x480p59hz  
1440x576i50hz  
1440x576i50hz  
1440x480i59hz  
1440x480i59hz  
640x480p59hz

Langerz82, 2022-04-23:
Resolutions look fine, what about this command I posted earlier:
```
bash -x emuelec_autostart.sh
```
Can you run that, it will produce lots of output.

AlienatedDiego, 2022-04-23:
https://pastebin.com/RGkf6fu9

Langerz82, 2022-04-23:
@AlienatedDiego
According to that output you still have 1080p60hz set in your config.
```
++ PAT='^ee_videomode=(.*)'
+++ grep -oE '^ee_videomode=(.*)'
+++ cat /storage/.config/emuelec/configs/emuelec.conf
++ EES=ee_videomode=1080p60hz
++ EES=1080p60hz
++ echo 1080p60hz
+ DEFE=1080p60hz
+ '[' 1080p60hz == Custom ']'
+ setres.sh 1080p60hz
```
Are you sure it was set to 720p and saved in ES?

AlienatedDiego, 2022-04-23:
sorry mi fault...a second

AlienatedDiego, 2022-04-23:
720p https://pastebin.com/Zs7Cm3Ag

Langerz82, 2022-04-23:
@AlienatedDiego 
Hey, and that output was while your were in ES and the display was screwed up right?

AlienatedDiego, 2022-04-23:
I did a new check with setres.sh using a controller to refresh but nothing

[image]

[image]

Langerz82, 2022-04-23:
@AlienatedDiego that is in a game, is EmulationStation the same and corrupt too?

AlienatedDiego, 2022-04-23:
correct.while the problem is displayed on screen and 720p in ES.  
NO.it is the screensaver

Langerz82, 2022-04-23:
On the TV do you have an option were you can see the output resolution on the current HDMI signal? It should be an info button on the TV remote control or something similar.

Langerz82, 2022-04-23:
@AlienatedDiego 
Hey can you also run this following command in SSH:
```
bash -x setres.sh 720p60hz
```
and post output? thanks for your patience.

AlienatedDiego, 2022-04-23:
yes of course, in all cases the TV shows the change of resolution and indeed it shows "720p".

no problem.

Langerz82, 2022-04-23:
Wait so when you press info on the tv remote it says it's displaying in 720p? That's real strange maybe ES is not checking the actual resolution dimensions when starting the screen.

Langerz82, 2022-04-23:
@AlienatedDiego 
hey try this command and see what it does to the output:
```
fbset -fb /dev/fb0 -g 1280 720 1280 720 32
```

AlienatedDiego, 2022-04-23:
mmm does nothing at all.

the tv shows the current resolution and refresh rate of the selected hdmi channel, if i restart emuelec it will show 1080p60hz on the splashscreen (default resolution) when loading the system it changes to the chosen resolution in ES, 720p or whatever. In short, the behavior of the TV is the same in the rest of the devices and versions of emuelec. I don't think this is the case.

please excuse me if I don't explain myself well, I use a translator...

AlienatedDiego, 2022-04-23:
the question is...what do these device(s) differ from the rest. taking into account that all of them are now using the same version 4.5NG and the resolutions work fine in the rest of the devices that I have tried ... I mean that the code is probably fine but these old amlogics work differently than the rest, surely for this reason Shanti is thinking of a specific release for them

Langerz82, 2022-04-23:
I have one idea, not sure if it will work though:
Use SSH, and run the command:
```
nano /emuelec/bin/setres.sh
```
copy and paste this into the file:
https://gist.githubusercontent.com/Langerz82/f9d2b43052b9fe56f287e8cc2c6c645c/raw/79615e4087466d4c6caceffc35bbe6a38866f7ea/setres.sh

save, exit then restart the device.

Langerz82, 2022-04-23:
if it does not work. In that file at the bottom change:
```
[[ "$EE_DEVICE" == "Amlogic-ng" ]] && fbfix
```
to just:
```
fbfix
```
and re-try. I forgot that ng devices suffer from a bug that the buffer is not usually set correctly without that call when SDL is creating it's screen.

AlienatedDiego, 2022-04-23:
ok , the first attempt failed, it took me a while to understand that save(enter) ctrl+x (save)..reboot and it didn't work.  
I can't edit the line from my browser, can you do it for me? Thanks for your patience

Langerz82, 2022-04-23:
run this command:
```
sed -i "s/[[ "$EE_DEVICE" == "Amlogic-ng" ]] && fbfix/fbfix/g" /emuelec/bin/setres.sh
```
will change the line to correct fbfix.

AlienatedDiego, 2022-04-23:
It didn't work. But I can't be 100% sure I did it right. sorry.

[image]

Langerz82, 2022-04-23:
ah no, try and just run these commands:
```
rm /emuelec/bin/setres.sh
wget https://gist.githubusercontent.com/Langerz82/f9d2b43052b9fe56f287e8cc2c6c645c/raw/4f313c9aab5f461aec2b99069684da113e5fd458/setres.sh /emuelec/bin
chmod +x setres.sh
```
then reboot.

AlienatedDiego, 2022-04-23:
yes...i deleted it because i realized it was wrong.thank you

[image]

Langerz82, 2022-04-24:
now reboot did it work?

AlienatedDiego, 2022-04-24:
ok. new attemp

AlienatedDiego, 2022-04-24:
nothing... tell me if you see something wrong, I close ssh before trying

[image]

Langerz82, 2022-04-24:
run each line separately.

AlienatedDiego, 2022-04-24:
[image]es correcto?

Langerz82, 2022-04-24:
go into the directory with:
```
cd /emuelec/bin
```
then list the files in the directory with
```
ls -la
```
you might have to remove old files with
```
rm setres.sh*
```
then run:
```
wget https://gist.githubusercontent.com/Langerz82/f9d2b43052b9fe56f287e8cc2c6c645c/raw/4f313c9aab5f461aec2b99069684da113e5fd458/setres.sh /emuelec/bin
```
wait till finish, then put:
```
chmod +x /emuelec/bin/setres.sh
```

AlienatedDiego, 2022-04-24:
[image]

Langerz82, 2022-04-24:
yep good now reboot

AlienatedDiego, 2022-04-24:
It didn't work but I appreciate all the effort Josh!

Langerz82, 2022-04-24:
@AlienatedDiego I know your probably too tired, but if your still up for it, can you run this command again when you have 720p running?
```
bash -x emuelec_autostart.sh
```
I just want to make sure it's in-fact calling the modified setres.sh you downloaded.

AlienatedDiego, 2022-04-24:
No problem, I'm just sorry I can't be of more help. Thank you for your patience.

https://pastebin.com/VM7awD7Z

Langerz82, 2022-04-24:
I can see that it is, setting 720p mode and fixing the frame buffer with the modified setres.sh. And the resolution is still corrupt in  EmulationStation right? I think shantigilbert may have a better idea of why it still has an error, I'm kinda out of ideas.

```
++ EES=ee_videomode=720p60hz
++ EES=720p60hz
++ echo 720p60hz
+ DEFE=720p60hz
+ '[' 720p60hz == Custom ']'
+ setres.sh 720p60hz
The framebuffer device was opened successfully.
vinfo.yoffset=720
FIX: setting vinfo.yoffset=0.
```

AlienatedDiego, 2022-04-24:
yeah i'm running it now and it's still corrupt, let shanti dig into the problem, maybe he knows where to look. I also take this opportunity to thank you for your contributions with the resolutions and autoconfig of independent systems. really helpful. 

bedtime but available if I can help with anything.

Langerz82, 2022-04-24:
@AlienatedDiego just wondering if you have it set to 720p then you completely shutdown system and pull the plug out. For the first boot does 720p show ok?

Langerz82, 2022-04-24:
I think I have worked it out. mount_romfs restarts the emustation service before setres.sh takes place. so it's trying to run in 720p even though the display is at it's default of 1080p. I have to make some changes and release a PR fix since there are changes to the autostart file.

AlienatedDiego, 2022-04-24:
yes, I have completely turned it off by removing the cable, unfortunately 720p is still broken.

the guy who started the thread must be freaking out

I was remembering a similar issue with broken resolutions 720p and below a long time ago when still on 32bit releases that was discussed on discord, the broken resolution effect was similar, I'll look into it

AlienatedDiego, 2022-04-24:
Oh. Vale.eso tiene sentido

Langerz82, 2022-04-24:
Try this PR and see if it's fixed. You will have to recompile the build.
See:
https://github.com/EmuELEC/EmuELEC/pull/905

AlienatedDiego, 2022-04-24:
Oh ok
that makes sense. Unfortunately I can't do it, as I told you I don't have code knowledge nor do I know how to compile by myself, I only occasionally look for system failures and report them. I'm sorry

quote Shanti to compile a test and I'll try it out asap

Langerz82, 2022-04-24:
Ok so though about this some more, I need someone with the bug to replicate the issue and type a few commands in SSH to see if it fixes it.
So make sure emulationstation you have set the resolution to 720p and that the screen is stuffed as shown like the bug.
Then use a PC to SSH in and do the following commands one at a time:
```
setres.sh 720p60hz
fbfix
systemctl restart emustation
```
And let me know if the screen is still stuffed up.

AlienatedDiego, 2022-04-24:
Same, no change. commands are always entered while the bug is showing (ES 720p60hz)

[image]

Langerz82, 2022-04-24:
Are you sure EmulationStation is set to 720p60hz? In ssh also do:
```
get_ee_setting ee_videomode
```
720p60hz should return.

AlienatedDiego, 2022-04-24:
yep sure.

EMUELEC:~ # get\_ee\_setting ee\_videomode  
720p60hz

Langerz82, 2022-04-24:
Hey man do you want to Team Viewer so I can try and debug the issue?

AlienatedDiego, 2022-04-24:
no problem, I'll contact you through the forum

[36 more replies]

closed 2023-05-02 by Langerz82