EmuELEC/EmuELEC #903: Hi ，Geekers！My S905m box boot by hdd 、usb drive、ssd slow than sdcard

zouplus, 2022-04-22:
Hi ，Geekers！My S905m box boot by hdd 、usb drive、ssd slow than sdcard，I try edit “aml_autoscript” ：

setenv upgrade_step 2
setenv usbdtb ‘if fatload usb 0 ${dtb_mem_addr} dtb.img; then else store dtb read $dtb_mem_addr; fi’
setenv bootfromusb ‘usb start 0; if fatload usb 0 ${loadaddr} kernel.img; then run usbdtb; setenv bootargs ${bootargs} bootfromusb; bootm; fi’
setenv bootcmd run bootfromusb; fi; run storeboot’
saveenv
run storeargs
run bootfromusb

but it may not work，all usb device seem boot slower than sd card；the hdd boot process is thant ： loading sometimes，and pause above 30 sec，then contionue loading emuelec normally。

So Is there some way to reslove it；thank you every much！！！

the usb is 3.0

Langerz82, 2022-04-24:
lol @ calling us Geekers. Are you using an older HDD that spins? They are slower than most microSD's.

Langerz82, 2022-04-24:
sorry I accidentally closed the issue.

zouplus, 2022-04-24:
thank you so much！
I had saw that tutorial ，I will try it but I thingk it will not useful for this issue。I have two usb device，one is 128gb sansum usb disk,one is old hdd divice （it is usb 3.0， support stata);both of devices boot seem have that 2 steps,and need above 30sec;I had try my way to resolve it ,I had try s905m/s905l/s905x,all the results are same.. boot seem have 2 step ,one for 30sec（after 30sec hdd have a clearly sound）,one for loading normally boot（after 30 sec boot is fast as same as sd card above 4-7sec show loading image）

zouplus, 2022-04-24:
set config delay not work. I only use one storage device ，I make three partitions，fat32 2gb named emuelec，ext4 2gb named storage ,fat32 100gb named eerom ，I make folder roms and create a file named eluelecrom；all step is right，all step are right however it is still boot slow when use usb device ，in system and play game usb device work normally。

shantigilbert, 2022-04-24:
USB is slower than SD card, there is no fix for that.

closed 2022-04-24 by shantigilbert

zouplus, 2022-04-24:
> USB is slower than SD card, there is no fix for that.

but sd boot only 3-7 sec；usb devices need 30+3-7sec？does it normally？

zouplus, 2022-04-24:
@shantigilbert

shantigilbert, 2022-04-24:
That depends on your USB drive, device, USB controller, etc, but that sounds about right, USB is way slower than SD.  Why not just use SD card to boot and USB to load  ROMS?