EmuELEC/EmuELEC #906: [4.3][S905M2 rev d] Start netplay host with relay doesn't work and cannot update to 4.5

d-apps, 2022-04-24:
@shantigilbert Sorry for opening this issue, I thought about explaining my situation to you on Discord but it will be a big post and probably confusing so I thought that it is better If I create an issue.

Me and my friend are having problems with our internet provider and we really need to use retroarch's relay option to start netplay as host. We can enter in other people's rooms (work ok) but we can't play with each other because start net play host relayed is not working:

First, we both installed emuELEC 4.3 version (for no retroarch/core issues when connecting), I have box D905 (**S905M2 rev d**), I used **gxl_p281_1g.dtb** and installation was fine.

[image]

I tried to start netplay with relay but the room is never connectable, I set Sao Paulo because I live in Brazil but nobody is able to connect, they always get stuck in loading game screen (Tested with multiple friends, emuelec to emuelec and pc to my box)

[image]

So I tried to test exact same retroarch's version (1.9.8) on my laptop, set the relay option to on and server Sao Paulo, my friend was able to connect in every game we couldn't on emuelec. (me on my laptop and he is on his box with 4.3).

After that I thought that I could update my emuelec version 4.3 to 4.5 to test and see if it was fixed, but I can't update:

I tried all dtb images and the only one who boots in emuelec splash is **gxl_p281_1g.dtb** (the same dtb I used to install 4.3 which works fine) but when I start the system it freezes in emuelec splash screen for some time and restart the system, the installation does not start.

Is there anything you can do to help me with that? Maybe checking the dtb file for this soc so I will be able to update to 4.5?

Thank you very much.

shantigilbert, 2022-04-24:
Not sure why you even have to post here when I already asked for the information you claimed "fixed it" but you refuse to provide it, and even here you do not provide it. There is nothing I can do, as I cannot reproduce this issue as I already told you.

shantigilbert, 2022-04-24:
Please add logs, emuelec,log at least, right after you get the error.

d-apps, 2022-04-24:
I uploaded two logs, the first when I start netplay as host and the other when I try to connect to his room (both relay).

[as host.log](https://github.com/EmuELEC/EmuELEC/files/8550730/as.host.log)
[as guest.log](https://github.com/EmuELEC/EmuELEC/files/8550729/as.guest.log)

One other thing is that I tried to connect him directly through retroarch, imported the rom before trying to connect, and I got the error: **Failed to initialize netplay**.

d-apps, 2022-04-30:
@shantigilbert 
I am not sure what I did but I managed to install 4.5 on my S905M, I changed sd card with 16gb (with 4.3 installed) to a 64gb I have and it installed, was strange but at least it worked.

I know that the main problem is netplay hosting relay not working but to avoid confusion I am closing this issue and I gonna test everything with 4.5.

Thank you.

closed 2022-04-30 by d-apps