EmuELEC/EmuELEC #911: [4.5][S905M2 rev d][psx] error while loading shared libraries: libGLESv2.so: wrong ELF class: ELFCLASS64

d-apps, 2022-05-01:
Every game I try to run I get this error message.

**error while loading shared libraries: libGLESv2.so: wrong ELF class: ELFCLASS64**

Before this version I had 4.3 installed and psx games worked fine, I had to update because of netplay issues.

[psx emuelec.log](https://github.com/EmuELEC/EmuELEC/files/8599649/psx.emuelec.log)

shantigilbert, 2022-05-01:
https://github.com/EmuELEC/EmuELEC/wiki/How-to-update

https://github.com/EmuELEC/EmuELEC/issues/604

AlienatedDiego, 2022-05-02:
I reproduce the bug, it seems to affect pcsx and parallel 32B cores on amlogic(old) devices.
tested on s905w clean install 4.5 ng

miwasp, 2022-05-02:
shanti replied on discord:

copy this file [https://github.com/CoreELEC/opengl-meson/blob/aml64_buildroot_master/lib/eabihf/m450/r7p0/fbdev/libMali.so](https://github.com/CoreELEC/opengl-meson/blob/aml64_buildroot_master/lib/eabihf/m450/r7p0/fbdev/libMali.so) to /emuelec/lib32/libMali.m450.so

Can you please test this?

AlienatedDiego, 2022-05-02:
thanks, i'll do that asap, i'll need a fresh install for testing.

AlienatedDiego, 2022-05-02:
confirmed.it works

d-apps, 2022-05-02:
> /emuelec/lib32/libMali.m450.so

I can confirm too, copy the file fixed the problem, so I think this was a valid bug for this soc...

hobapli, 2022-05-04:
> > Hi friends i have same issue with n64 parallel core.
> > "retroarch32: error while loading shared libraries: libGLESv2.so: wrong ELF class: ELFCLASS64 exit 127"
> > I copied and pasted the file(libMali.so)you said into the folder, but it did not overwrite any file. there is no file such as libmali.m450.so in lib32 folder.
> > My device is also s905w. Still doesnt work.. What i'm doing wrong? regards
> 
> You did something wrong, maybe when renaming the file or didn't copy to the right directory.
> 
> I did it and n64 parallel core is working, before that is was not working too.

Do i have to rename libmali.so to libmali.m450.so and paste?

Langerz82, 2023-02-08:
Closing issue as not relevant anymore.

closed 2023-02-08 by Langerz82