EmuELEC/EmuELEC #913: [4.5][S905M2 rev d] Mupen64Plus Next is not working: Segmentation fault

d-apps, 2022-05-05:


Langerz82, 2022-05-09:
I'm not so sure s905m chipsets are supported on 4.5. What dtb file are you using?

d-apps, 2022-05-09:
> I'm not so sure s905m chipsets are supported on 4.5. What dtb file are you using?

But the message about 4.4/4.5 is:

"IMPORTANT: S905 (GXBB, p201) for the moment is no longer supported. If you have one of those devices **(s905 no letter after the 5)** DO NOT UPDATE, stay in 4.3."

Mine there is a letter "**M**" after "**5**" so I assume that it should work.

I used **gxl_p281_1g** dtb.

Most of other cores are working.

shantigilbert, 2022-05-18:
> Most of other cores are working.

But Yabasanshiro is not one of them.

closed 2022-05-18 by shantigilbert

d-apps, 2022-05-18:
> > Most of other cores are working.
> 
> But Yabasanshiro is not one of them.

Sorry but I didn't understand why you closed this.

It won't be fixed in future updates?

shantigilbert, 2022-05-18:
No, Yabasanshiro does not work on older devices, there is nothing for me to fix