EmuELEC/EmuELEC #914: [4.5][S905M2 rev d] Mupen64Plus Next is not working: Segmentation fault

d-apps, 2022-05-05:
The core Mupen64Plus Next is not working on emuelec 4.5 with S905M2

Error: 

**/usr/bin/emuelecRunEmu.sh: line 441:  9943 Segmentation fault      ${RABIN} $VERBOSE -L /tmp/cores/${EMU}.so --config ${RACONF} "${ROMNAME}"
exit 139**

[n64 emuelec.log](https://github.com/EmuELEC/EmuELEC/files/8628144/n64.emuelec.log)

jlsegb, 2022-11-13:
Same issue in [4.6 TEST]. None of the n64 games run due to this.

Langerz82, 2023-05-02:
Are you running the Retroarch Core with 1080p resolution? If not change it to run at 1080p. s905m resolution is locked to 1080p.

Langerz82, 2023-06-10:
I tried a few things to fix this issue. It looks like mupen64plus-next is exhausting too much memory on 1gb ram devices.

Langerz82, 2023-06-12:
Does Mupen64Plus 32-bit work for you? In the gameslist press SELECT, Advanced options, and change the core too it.

Mupen64Plus-Next does not work for 1GB devices since EmuELEC 4.4. Even when creating 32-bit versions its screwed, and the source would have to be modified to use less memory which I doubt will be done.

leosmeira, 2023-08-03:
> Here is the amlogic mupen64plus-next 32-bit executable. You can try it, but I think you will find mupen64plus 32-bit on 4.6 runs better.
> 
> [mupen64plus_next_32b_libretro.zip](https://github.com/EmuELEC/EmuELEC/files/11719629/mupen64plus_next_32b_libretro.zip)

How can I compile and generate 32bit emulator (Retroarch and libretro cores)?

Langerz82, 2023-12-14:
@leosmeira In EE 4-6 Retroarch 32-bit should compile automatically and some libretro cores. If you need additional libretro cores in 32-bit the packages need to be added to lib32 and all there dependencies need to be 320bit as well. They are contained in the package section:
EmuElec/packages/lib32/emuelec/libretro

Anyway I'm going to close this issue because there is no way to really fix this issue on devices with low memory, and mupen64plus 32-bit runs much better than the -next version anyway, and doesnt have memory constraints.

closed 2023-12-14 by Langerz82