EmuELEC/EmuELEC #916: ps3 controller ppsspp 、drastic work not fine

zouplus, 2022-05-09:
1.emuelec controller setting ,all button can work(I try 3.9 & 4.5)
2.ppsspp the center menu button (whitch mapping hotkey button) at 3.9 work fine,but 4.5 cannot be recognized
3.drastic cannot controll menu : up / down / left / right ；I had config the extern setting in drastic menu up / down / left / down,I am sure the setting is work,because the left ,right ,...A B... can controlle the games

Langerz82, 2022-05-09:
Did you do a clean install of 4.5 or upgrade?
Also try in EmulationStation re-mapping your game controller again. From what I understand PPSSPP uses the config from gamecontrollerdb.txt.
Also please provide more info such as your device, and the name of the game controller.

zouplus, 2022-05-09:
> Did you do a clean install of 4.5 or upgrade?
> Also try in EmulationStation re-mapping your game controller again. From what I understand PPSSPP uses the config from gamecontrollerdb.txt.
> Also please provide more info such as your device, and the name of the game controller.

my controller is ps3 controller and I do clean install of 4.5，I fomate usb flash；I try many times to mapping controller the menu button
 still not work at ppsspp but ee work okey；my box is 900v22d，controller is ps3 controller

zouplus, 2022-05-09:
and the box get error when boot！
；It hanppend boot with usb ，but tf card is okey！the box is 900v 22d ，s905l3a （905x2）

zouplus, 2022-05-10:
[image]
[image]
[image]

--------------
the ps button not work at ppsspp at 4.5 but 3.9 work fine（only that button），and the ps3 controller at 3.9 default mapping  
is right，but 4.5 should remapping the controller。

Langerz82, 2022-06-06:
I cant even get the dpad working in drastic, not for the menu screen below or for the actual game.
What controller do you use that works? can you provide the files in /storage/.config/emulationstation/scripts/drastic/config ?

zouplus, 2022-07-26:
sorry i am so late to reply, you can use keyboard to set drastic，the different controller may not work，you can press “M” key to set.

7Ji, 2022-08-28:
S905L3A/L3B (S905L, version 3, step A/B) is not supported and none of our existing DTBs are for these SoCs. You can boot them with dtbs of S905X2, but many components on the SoC do not work as intended --- they are NOT some simple reskined S905X2 or either S905X3 (they're more like downgraded X3, not X2), not as simple as S905L downgraded from S905X, and do expect major breaks when you force an incompatible dtb onto a different SoC.

As long as we are still using the 4.9 kernel, devices equiped with these SoCs (E900V3c, CM118, etc) can never be supported and don't expect any support as no one of the team has such hardware

closed 2022-08-28 by 7Ji

Langerz82, 2022-08-28:
What version had drastic? It seems we dont have it included anymore.

Langerz82, 2022-08-28:
I forgot you had to install drastic from the settings menu.

Jus a note on these issues:
1. PPSSPP the control.ini did not change, therefore it is most likely a source bug and should be raised in there github.
2. Drastic is closed source so you would have to raise the issue with them.