EmuELEC/EmuELEC #918: Unable to boot EmuELEC 4.5 on Phicomm N1(S905D) box

alifare, 2022-05-11:
From the official NOTE, S905D is supported by EmuELEC 4.5, even if S905 is not.

> IMPORTANT: S905 (GXBB, p201) for the moment is no longer supported. If you have one of those devices (s905 no letter after the 5) DO NOT UPDATE, stay in 4.3.

So I downloaded `EmuELEC-Amlogic-ng.aarch64-4.5-Generic.img.gz` and flashed it into a USB disk. Unfortunately, however, I failed to boot EmuELEC 4.5 from USB on my Phicomm N1(S905D) box, even if I replaced the files(dtb.img, aml_autoscript and others) with some patch packages downloaded from Internet, which is said to work well on older EmuELEC versions like 4.3.

So I am asking is there any way to boot EmuELEC 4.5 on Phicomm N1(S905D), or can any one give me any working directions to fix this problem?

I am willing to do it myself, and I have rich experience in Linux and programming skills. But I know nothing about S905D and EmuELEC.

AlienatedDiego, 2022-05-11:
Duplicate/same issue https://emuelec.discourse.group/t/unable-to-boot-emuelec-4-5-on-phicomm-n1-s905d/1137

maskedeken, 2022-05-12:
> From the official NOTE, S905D is supported by EmuELEC 4.5, even if S905 is not.
> 
> > IMPORTANT: S905 (GXBB, p201) for the moment is no longer supported. If you have one of those devices (s905 no letter after the 5) DO NOT UPDATE, stay in 4.3.
> 
> I downloaded `EmuELEC-Amlogic-ng.aarch64-4.5-Generic.img.gz` and flashed it into a USB disk. Unfortunately, however, I failed to boot EmuELEC 4.5 from USB on my Phicomm N1(S905D) box, even if I replaced the files(dtb.img, aml_autoscript and others) with some patch packages downloaded from Internet, which is said to work well on older EmuELEC versions like 4.3.
> 
> So I am asking is there any way to boot EmuELEC 4.5 on Phicomm N1(S905D), or can any one give me any working directions to fix this problem?
> 
> I am willing to do it myself, and I have rich experience in Linux and programming skills. But I know nothing about S905D and EmuELEC.

@alifare you can fix this with my patches. Please refer to [https://www.right.com.cn/forum/thread-8226482-1-1.html](https://www.right.com.cn/forum/thread-8226482-1-1.html)

alifare, 2022-05-17:
> 

Thanks for your help. I will give it a try

shantigilbert, 2022-05-17:
> @alifare you can fix this with my patches. Please refer to https://www.right.com.cn/forum/thread-8226482-1-1.html

Since those are binaries, What did you change on the source code to make it work?

alifare, 2022-05-17:
> @alifare you can fix this with my patches. Please refer to https://www.right.com.cn/forum/thread-8226482-1-1.html

I tried to switch from Android TV to EmuELEC, but failed. My television looks like this:
[image]
[image]

maskedeken, 2022-05-18:
> 

check if aml_autoscript.zip exists on boot parition of your usb disk, if not, just create an empty file named 'aml_autoscript.zip', and try again

maskedeken, 2022-05-18:
> > @alifare you can fix this with my patches. Please refer to https://www.right.com.cn/forum/thread-8226482-1-1.html
> 
> Since those are binaries, What did you change on the source code to make it work?

I just extracted the uncompressed kernel and ramdisk from the original KERNEL, since Phicomm N1 can only boot an uncompressed kernel

alifare, 2022-05-18:
> 

Yes, aml_autoscript.zip exists on boot partition because I copied ALL files from your patches. Actually, I was confused why there is an empty file like that, but I still copied it to the boot partition.

Maybe something happened after running Android TV and files or partitions in USB were changed?

maskedeken, 2022-05-18:
> Yes, aml_autoscript.zip exists on boot partition because I copied ALL files from your patches. Actually, I was confused why there is an empty file like that, but I still copied it to the boot [...]

try to run 'adb reboot update' ?

alifare, 2022-05-18:
> 

From where ?

maskedeken, 2022-05-18:
adb connect <your tv box ip>

then 

adb reboot update

alifare, 2022-05-18:
> adb connect xx.xx.xx.xx (your tv box ip)
> 
> then
> 
> adb reboot update

I understant the command `adb reboot update` is to update reboot command in my android TV box. 

Will the update affect my TV box? Can I still use my box to watch TV?

maskedeken, 2022-05-18:
'adb reboot update' just forces your box to boot from usb, it won't affect your tv box

alifare, 2022-05-18:
> 
If that is the case, I have to run `adb reboot update` each time when I switch from TV box to EmuELEC ?

maskedeken, 2022-05-18:
> > 
> 
> If that is the case, do I have to run `adb reboot update` each time when switching from TV box to EmuELEC ?

just give it a try

alifare, 2022-05-19:
> > > 
> > 
> > 
> > If that is the case, do I have to run `adb reboot update` each time when switching from TV box to EmuELEC ?
> 
> just give it a try

It still doesn't work, so sad

alifare, 2022-05-19:
> > > > > 
> > > > 
> > > > 
> > > > If that is the case, do I have to run `adb reboot update` each time when switching from TV box to EmuELEC ?
> > > 
> > > 
> > > just give it a try
> > 
> > 
> > It still doesn't work after running command `adb reboot update`, so sad [image]
> 
> how did you run it successfully for the first time?

The fresh burnt EmuELEC 4.5 (in USB disk) works, and I even played some games in it.

But after I switched from EmuELEC to Android TV by running `REBOOT FROM NAND`, and switched back from Android TV to EmuELEC by running `LibreELEC`, it faild to boot and my television showde this :
[image]

shantigilbert, 2022-05-19:
This is clearly not a bug on EmuELEC so please continue this on the forum there is already a discussion opened there : 
https://emuelec.discourse.group/t/unable-to-boot-emuelec-4-5-on-phicomm-n1-s905d/1137

closed 2022-05-19 by shantigilbert