EmuELEC/EmuELEC #922: Auto safe is not persistent

LeducH, 2022-05-16:
I want to turn off the auto load/save option. I did it in the setting and also in retroarch but the options keep turn on. Is there another place where this can be disabled.

closed 2022-05-16 by shantigilbert

LeducH, 2022-05-16:
what template are you referring to?

AlienatedDiego, 2022-05-16:
https://github.com/EmuELEC/EmuELEC/blob/dev/.github/ISSUE_TEMPLATE/bug_report.md

shantigilbert, 2022-05-16:
In this case, this is not a bug.

https://github.com/EmuELEC/EmuELEC/issues/604

please post general questions to Discord or the Forum

LeducH, 2022-05-17:
How is this not a bug? 

That is clearly a wrong if  the auto safe is set to off in  emulationstation and in retroarch and it is still autoloading. 

Please elaborate the how this feature works.

LeducH, 2022-05-17:
> do not reproduce this problem, I do not have oga but it seems some configuration. try: reset scripts and binaries (danger zone) now check the options: auto save/load disabled save incremental games disabled game manager (captures) deactivated
> 
> ask on the OdroidGo/Super channel on Discord or on the forum, maybe someone can help.
> 
> If you are sure that the error exists and you want to open a bug report, you must do it from a clean installation and follow the instructions in the template, providing all the information required to try to reproduce it.

tested the reset still keep the auto load/save on even tough it is off