EmuELEC/EmuELEC #924: Auto safe is not persistent

LeducH, 2022-05-17:
**Describe the bug**
The auto load/save in the emulation station option is not persistent in retroarch. 

**To Reproduce**
Steps to reproduce the behavior:
1. Install the `EmuELEC-OdroidGoAdvance.aarch64-4.5-odroidgo2.img.gz`
2. got to `game settings` 
3. turn on `incremental savestates` and `show save state manager` to `if not empty`
4. make sure `auto save/load` is off 
5. start a game 
6. open the retroarch toggle menu 
7. go to savestate settings 
8. turn off auto load/save  
9. save configuration 
10. restart game 
 
**Expected behavior**
Game neither auto saves nor loads. 
Either the global settings in emulation station or in retroach setting is persisted. setting it off still keeps the option on in retroarch 

also after reset in emulecsettings

**EmuELEC version:**
 - Version `4.5`
 
**Device (please complete the following information):**
 - SOC: [e.g. S095x3]
 - Brand: Odroid 
 - Model Go Super
 - Version https://www.hardkernel.com/shop/odroid-go-super-dim-gray/

**Additional context**
related to #922

miwasp, 2022-05-17:
Just tested this on an OGS and it works as expected on my side.
If i try to reproduce your steps and set "auto save/load" to "off" in step 4 i don't have to do step 8 because its already "off" there too. If i set it to "on" in ES, it is "on" in retroarch too.

What exactly do you mean in steps 6-8?
> 6. open the retroarch toggle menu
Do you mean the regular retroarch menu (Hotkey + X) or is "toggle menu" something special?
> 7. go to savestate settings
Do you mean "Settings->Saving"?
> 8. turn off auto load/save
There is no such setting in retroarch that i know of. There are "Auto Save State" and "Load State Automatically"

LeducH, 2022-05-17:
> Just tested this on an OGS and it works as expected on my side. If i try to reproduce your steps and set "auto save/load" to "off" in step 4 i don't have to do step 8 because its already "off" there [...]

fixed the steps refer to original post

miwasp, 2022-05-17:
Ok, so we are talking about the same things. But as i already said, i cannot reproduce this.
If i switch "off" the setting in ES (step 4), the two settings in Retroarch are automatically also set to "off" (step 8).

LeducH, 2022-05-17:
consider 

```
EMUELEC:~/.config/emuelec/configs # cat  emuelec.conf  | grep -i auto
# Audio device to use, comment for auto, other options 0,0 (default) 0,1 (for some AV outputs)
#ee_audio_device=auto
# Enable Advance mame auto gamepad configuration 1,0 (default 1)
advmame_auto_gamepad=1
## Set the audio device (auto, hdmi, jack, speakers, headphones) mostly for OdroidGoAdvance
## Automatically check for updates at start (0,1)
## Automatically select shaders for all systems
## Set ratio for all emulators (auto,4/3,16/9,16/10,custom)
## Set autosave/load savestate for all emulators (0,1)
pcenginecd.autosave=0
```
```
EMUELEC:~/.config/retroarch # cat retroarch.cfg | grep -i savestate
menu_savestate_resume = "true"
savestate_auto_index = "false"
savestate_auto_load = "true"
savestate_auto_save = "true"

```

LeducH, 2022-05-17:
I needed to write the line 

```
global.autosave=0
```

into `/storage/.config/emuelec/configs/emuelec.conf` so it will be consistent. This leads me to believe that some script is not properly setting this option. 

since it was also missing after a fresh install in the file as mentioned above.

miwasp, 2022-05-17:
If you set "auto save/load" to "on" in ES, you get a line

> global.autosave=1

in `/storage/.config/emuelec/configs/emuelec.conf`

If you set "auto save/load" to "off" again in ES, this line is removed from `/storage/.config/emuelec/configs/emuelec.conf`

I have set "auto save/load" to "off" currently, i don't have any "global.autosave" in emuelec.conf and it is working as it should.

AlienatedDiego, 2022-05-17:
hmm i think i could reproduce it

it seems that using the game manager to start "new game" the game will create an auto state file and use it next time even if the auto save/load option is disabled in ES

 somehow the savegame manager activates the auto s/l not sure but if it should work this way if you don't want to deal with just disable the savegame manager in ES

LeducH, 2022-05-17:
> hmm i think i could reproduce it
>
> it seems that using the game manager to start "new game" the game will create an auto state file and use it next time even if the auto save/load option is disabled [...]

Good observation. I would also that the savegame manager does brings in another moving part. Therefore it could be beneficial to add  

```
global.autosave=0
```

if the manager is on. 

@miwasp Turn on the save game manager

shantigilbert, 2022-05-17:
> > hmm i think i could reproduce it
> > it seems that using the game manager to start "new game" the game will create an auto state file and use it next time even if the auto save/load option is disabled in ES
> > somehow the savegame manager activates the auto s/l not sure but if it should work this way if you don't want to deal with just disable the savegame manager in ES
> 
> Good observation. I would also say that the savegame manager brings in another moving part. Therefore it could be beneficial to add
> 
> ```
> global.autosave=0
> ```
> 
> if the manager is on. depending it is off
> 
> @miwasp Turn on the save game manager and test again

There is no point on doing this, if the value is not present it is considered off.

I have not tested this personally, but you seem you are using the save manager, which in turn will enable autosave on as intended since it needs it to load the save you have chosen, to me it seems is working as it should, but you want the "auto save setting" to override this,  I am not sure that is possible, because if its disabled then the save manager will not "auto load" the save you have chosen, those 2 settings conflict, so either use one or the other.

LeducH, 2022-05-17:
> There is no point on doing this, if the value is not present it is considered off.
>
> I have not tested this personally, but you seem you are using the save manager, which in turn will enable autosave [...]

Your assumption is too strong. It might be that save manager includes auto save/load , but logically these two can exists separately. If you assume that you want to use  the save manger and picking a save state slot but you do not want to autoload and save --- especially when you do a incremental save. Then there is to reason that the manager has to imply auto save. 

Therefore the argument that it works just like designed might be right. Maybe I missed the project planning of this feature so I might be wrong. However my line of argument is still a valid and reproducible ( ref. @AlienatedDiego ) as I described in my opening post .  At the end, this might just be a suggestion instead of a bug report.  Nevertheless, my arguments still stand and can be of help for people who want to add this feature or create a pull request.

shantigilbert, 2022-05-17:
> logically these two can exists separately. If you assume that you want to use the save manger and picking a save state slot but you do not want to autoload and save --- especially when you do a incremental save. Then there is to reason that the manager has to imply auto save.

Basically the save manager is a way to load whatever save state you have chosen when it loads the game, sure they could work alongside, but then it would make no sense... the save manager NEEDS auto save/load to work, you cannot select a save slot and not auto load it that is a Retroarch function (or in your case limitation) and it is exactly how it should work. Again these are 2 features that work fine on their own, but not together, so choose one of the other.

LeducH, 2022-05-17:
> Basically the save manager is a way to load whatever save state you have chosen when it loads the game, sure they could work alongside, but then it would make no sense... the save manager NEEDS auto [...]

Please provide the feature of how the to save manager is defined. This definition will clear up the discussion. 

Moreover, you use the term `auto load` ambiguously. Concretely, autoload/save is defined in retroarch with save state slot `-1`. Thus if the save state manager loads slot `-1`. You use the word auto load as 'automatically loading the chosen slot in emuelec'. 
These are not the same. As I outlined, autosave should mean to load in slot `-1` as defined by retroarch if you want to use retroarch terminology.  

I am willing to accept the definition you mentioned, given that it is defined in this way.   Therefore please provide the feature definition of the save state manager (@shantigilbert ).

AlienatedDiego, 2022-05-17:
> > > lógicamente estos dos pueden existir por separado. Si asume que desea usar el administrador de guardado y elegir un espacio de guardado de estado, pero no desea cargar y guardar automáticamente, especialmente cuando realiza un guardado incremental. Entonces hay una razón por la que el administrador tiene que implicar el guardado automático.
> > 
> > 
> > Básicamente, el administrador de guardado es una forma de cargar cualquier estado de guardado que hayas elegido cuando carga el juego, seguro que podrían funcionar junto, pero entonces no tendría sentido... el administrador de guardado NECESITA guardar/cargar automáticamente para funcionar, no puedes seleccione una ranura para guardar y no la cargue automáticamente, es una función de Retroarch (o, en su caso, una limitación) y es exactamente como debería funcionar. Nuevamente, estas son 2 características que funcionan bien por sí solas, pero no juntas, así que elija una de las otras.
> 
> Proporcione la característica de cómo se define el administrador para guardar. Esta definición aclarará la discusión.
> 
> Además, utiliza el término de `auto load`forma ambigua. Concretamente, autoload/save se define en retroarch con save state slot `-1`. Por lo tanto, si el administrador de estado de guardado carga la ranura `-1`. Usas la palabra carga automática como 'cargar automáticamente la ranura elegida en emuelec'. Estos no son lo mismo. Como describí, el guardado automático debería significar cargar y guardar en la ranura `-1`según lo define retroarch si desea usar la terminología retroarch.
> 
> Estoy dispuesto a aceptar la definición que mencionas, dado que se define de esta manera. Por lo tanto, proporcione la definición de función del administrador de estado de guardado (@shantigilbert).

I support all points of view, for that reason I followed your problem... but I think this is not the point, please, you can say the same thing without imposing your ideas or demanding anything from the developer in this way, dealing with a minor issue in my opinion.
it is about contributing/collaborating with ideas.

shantigilbert, 2022-05-17:
> Moreover, how did you get to conclusion that the save state manager "needs" a auto save/load slot?

Well I already explained this... so no point in doing it again. Anyways this is a no fix, as there is no bug is working as intented.

closed 2022-05-17 by shantigilbert

LeducH, 2022-05-18:
> I support all points of view, for that reason I followed your problem... but I think this is not the point, please, you can say the same thing without imposing your ideas or demanding anything from [...]

I was making my point clear using a line of argument in accordance to retroarch. I was not imposing my view rather I ask for an reasonable and logical explanation. The argument 'it was designed this way' is rather thin and I would say it can have more explanation. Moreover, I did not demand anything rather I ask for an explanation in basis of reason and retroarch. It is quite fair to ask questions especially if there is conflicting logical viewpoints. But I guess this goes into a 'works as designed'. 

and as @shantigilbert mentioned I am supposed to create a pull request with the appropriate changes.

AlienatedDiego, 2022-05-18:
it's okay. the problem exists. Locating it thanks to your report is a first step.

the dev's explanation is logical, there is a conflict with RA and ES in the use of the saved game manager in ES and the auto s/l being separate options, if it is not used it works fine and when it is activated the exception appears.

I don't know if there is a way to make both things work at the same time, but perhaps the easiest solution in my opinion is to "join them" again. if you want to know what i mean try a 4.3 image

anyway nothing can be done now, when tests are done for future releases it will be time to come back to this if there are no other more important problems :)
[image]
[image]

ddzzhen, 2022-06-13:
Same problem as  yours , I tried the  4.5 yesterday ,and when  enter retroarch and change  the hotkeys, there is no where to save the profile（no options to choose ，and `F2` is not useful），and when  I back to  the  game, the hotkeys I just  changed  is back to default.