EmuELEC/EmuELEC #929: Rockchip 3228a is supported, developer never cared.

AstaFatherSatan, 2022-05-22:
It will run theres a rockchip 2020 port of emuelec unofically floating around and there is libreelec which is based on openelec which emuelec is based on. There is armbian for the rk3228a/rk322x and rk3229 available ontop of lakka. Its up to you, and you dont want to bother. Its a shame. It is compatible. You dont want to port it.

AstaFatherSatan, 2022-05-22:
https://github.com/EmuELEC/EmuELEC/commit/0e939189579e2700e970e2eaaa89630cad2dd173
https://github.com/knaerzche/LibreELEC.tv/releases/tag/RK322x-le92-0bca75f
https://forum.armbian.com/topic/12656-csc-armbian-for-rk322x-tv-boxes/
https://drive.google.com/file/d/1oRv1aglURg_EYcHsUAjOamfxomz27ZJU/view
https://bit.ly/3qVLkFt

shantigilbert, 2022-05-22:
No, sorry.

closed 2022-05-22 by shantigilbert