EmuELEC/EmuELEC #931: Add mrp emulator

xiayang0521, 2022-05-30:
**Is your feature request related to a problem? Please describe.**
Is it possible to add a  mrp emulator in Emuelec?

**Describe the solution you'd like**
The sources are:
https://github.com/Yichou/mrpoid2018
https://github.com/vmrp/vmrp

**Additional context**
The platform has excellent application driver performance, and can run games and various applications smoothly in harsh hardware environment. Very small space consumption, about 1/6 of the occupied space of the common java virtual machine . The file is only 1/3 the size of the same java game. The program is written in standard ANSI C, and the developed products have full independent property rights. Since 2003, the platform has been adapted to almost all mobile phone hardware solutions, and has highly reliable business capability and application scalability.

shantigilbert, 2022-05-30:
Sorry, I have no idea what this is, nor do I speak Chinese.  what does it emulate? and why should it be added to EmuELEC?

xiayang0521, 2022-05-30:
> 

OK，thank you！

closed 2022-05-30 by shantigilbert

xiayang0521, 2022-05-30:
> Sorry, I have no idea what this is, nor do I speak Chinese. what does it emulate? and why should it be added to EmuELEC?

It emulates mrp games like java game.  I think it may be possible to be added in emuelec as java simulator has been added.

7Ji, 2022-06-10:
Ahh, mrp games, the bizzare child of Chinese phone market for those low spec phones that can't support the "heavy" load of Java VM. These games were written poorly yet priced high as hell with payment method limiting to sending payment SMS only. And all those kids spending tens of Yuan just to play some games on their parents' phone or buying some powerful items through micro-transaction with purposedly set-up traps, they would definitely get their asses kicked when their parents checked their phone bills. Unwilling to get my ass kicked, as those mrp games were hard to pirate, I prefered Java games that I could easily downloaded for free (i.e. pirated) from d.cn (the site is still alive but became a pure Android games site, how time flows).

But support of it in EmuELEC is impossible, as:
1. mrpdroid/vmrp is not a libretro core
2. mrpdroid/vmrp is not a standalone emulator with Linux support
3. mrpdroid is written as an Android application with built on top of Android API and I/O methods so it can't be ported to be a libretro core or standalone emulator with Linux support either.
4. vmrp is written as a Windows/web app so it can't be ported to be a libretro core or standalone emulator with Linux support either.
5. The control method you need does not play well with the control method you would have on EmuELEC with a controller. You would need a dpad to perfectly play this game and the best way to emulate it is on PC with numpad. This is also the reason why some mature emulators for old pc-like platforms were not included, the games just shouldn't be played with a controller.

xiayang0521, 2024-04-12:
> Ahh, mrp games, the bizzare child of Chinese phone market for those low spec phones that can't support the "heavy" load of Java VM. These games were written poorly yet priced high as hell with payment method limiting to sending payment SMS only. And all those kids spending tens of Yuan just to play some games on their parents' phone or buying some powerful items through micro-transaction with purposedly set-up traps, they would definitely get their asses kicked when their parents checked their phone bills. Unwilling to get my ass kicked, as those mrp games were hard to pirate, I prefered Java games that I could easily downloaded for free (i.e. pirated) from d.cn (the site is still alive but became a pure Android games site, how time flows).
> 
> But support of it in EmuELEC is impossible, as:
> 
> 1. mrpdroid/vmrp is not a libretro core
> 2. mrpdroid/vmrp is not a standalone emulator with Linux support
> 3. mrpdroid is written as an Android application built on top of Android API and I/O methods so it can't be ported to be a libretro core or standalone emulator with Linux support either.
> 4. vmrp is written as a Windows/web app so it can't be ported to be a libretro core or standalone emulator with Linux support either.
> 5. The control method you need does not play well with the control method you would have on EmuELEC with a controller. You would need a numpad to perfectly play this game and the best way to emulate it is on PC with numpad. This is also the reason why some mature emulators for old pc-like platforms were not included, the games just shouldn't be played with a controller.

多谢7ji大神回复！原来如此！