EmuELEC/EmuELEC #932: [4.5][s922x][gamecube] No gamepad working - GCPadNew.ini faulty (?)

microrache, 2022-05-30:
I've updated EmuELEC from 4.3 to 4.5 on my AmLogic S922X device (Odroid N2+). Everything is working except I can't get any gamepad to work in Dolphin.

In 4.3 I could manually edit `/storage/.config/emuelec/configs/dolphin-emu/GCPadNew.ini` to configure my controllers. In 4.5 this file is generated automatically by `/usr/bin/set_dolphin_joy.sh`. Unfortunately the script creates a (presumably) faulty GCPadNew.ini in my case:

I have an Xbox 360 Wireless Adapter connected to my Odroid N2+. For that adapter I have two Xbox 360 Wireless Controllers. When I switch on only one controller, I get the following GCPadNew.ini-file after starting a GameCube ROM:

```
[GCPad2]
Device = evdev/1/Xbox 360 Wireless Receiver
Buttons/A = Button 1
Buttons/B = Button 0
Buttons/Start = Button 7
Buttons/X = Button 3
Buttons/Y = Button 2
Buttons/Z = Button 5
C-Stick/Dead Zone = 25.000000000000000
C-Stick/Down = Axis 4+
C-Stick/Left = Axis 3-
C-Stick/Modifier = Control_L
C-Stick/Modifier/Range = 50.000000000000000
C-Stick/Right = Axis 3+
C-Stick/Up = Axis 4-
D-Pad/Down = Button 14
D-Pad/Left = Button 11
D-Pad/Right = Button 12
D-Pad/Up = Button 13
Main Stick/Dead Zone = 25.000000000000000
Main Stick/Down = Axis 1+
Main Stick/Left = Axis 0-
Main Stick/Modifier = Shift_L
Main Stick/Modifier/Range = 50.000000000000000
Main Stick/Right = Axis 0+
Main Stick/Up = Axis 1-
Triggers/L = Axis 2+
Triggers/L = Axis 2+
Triggers/R = Axis 5+
Triggers/R = Axis 5+
```

AFAIK there are two errors here: 
- [GCPad2] should be [GCPad1]
- `evdev/1/Xbox 360 Wireless Receiver` should be `evdev/0/Xbox 360 Wireless Receiver` (zero instead 1)

As a result there is no controller recognition in Dolphin except Hotkey+Start to escape Dolphin.

When I switch on both controllers, I get the following GCPadNew.ini:

```
[GCPad2]
Device = evdev/1/Xbox 360 Wireless Receiver
Buttons/A = Button 1
Buttons/B = Button 0
Buttons/Start = Button 7
Buttons/X = Button 3
Buttons/Y = Button 2
Buttons/Z = Button 5
C-Stick/Dead Zone = 25.000000000000000
C-Stick/Down = Axis 4+
C-Stick/Left = Axis 3-
C-Stick/Modifier = Control_L
C-Stick/Modifier/Range = 50.000000000000000
C-Stick/Right = Axis 3+
C-Stick/Up = Axis 4-
D-Pad/Down = Button 14
D-Pad/Left = Button 11
D-Pad/Right = Button 12
D-Pad/Up = Button 13
Main Stick/Dead Zone = 25.000000000000000
Main Stick/Down = Axis 1+
Main Stick/Left = Axis 0-
Main Stick/Modifier = Shift_L
Main Stick/Modifier/Range = 50.000000000000000
Main Stick/Right = Axis 0+
Main Stick/Up = Axis 1-
Triggers/L = Axis 2+
Triggers/L = Axis 2+
Triggers/R = Axis 5+
Triggers/R = Axis 5+
[GCPad3]
Device = evdev/2/Xbox 360 Wireless Receiver
Buttons/A = Button 1
Buttons/B = Button 0
Buttons/Start = Button 7
Buttons/X = Button 3
Buttons/Y = Button 2
Buttons/Z = Button 5
C-Stick/Dead Zone = 25.000000000000000
C-Stick/Down = Axis 4+
C-Stick/Left = Axis 3-
C-Stick/Modifier = Control_L
C-Stick/Modifier/Range = 50.000000000000000
C-Stick/Right = Axis 3+
C-Stick/Up = Axis 4-
D-Pad/Down = Button 14
D-Pad/Left = Button 11
D-Pad/Right = Button 12
D-Pad/Up = Button 13
Main Stick/Dead Zone = 25.000000000000000
Main Stick/Down = Axis 1+
Main Stick/Left = Axis 0-
Main Stick/Modifier = Shift_L
Main Stick/Modifier/Range = 50.000000000000000
Main Stick/Right = Axis 0+
Main Stick/Up = Axis 1-
Triggers/L = Axis 2+
Triggers/L = Axis 2+
Triggers/R = Axis 5+
Triggers/R = Axis 5+
```

Again counting starts with [GCPad2] instead of [GCPad1] and `evdev/1` instead of `evdev/0`. But to my understanding at least my second controller should work inside Dolphin, but again this is not the case.

I do not need to mention that manually editing GCPadNew.ini has no effect as the file is replaced everytime I start a game. Editing the automatic controller recognition in set_dolphin_joy.sh is also not an option as this file is read-only. So what can I do to get any of my controllers to work in Dolphin?

Langerz82, 2022-06-04:
For now:
In SSH modify /emuelec/configs/emuelec.conf look for if there or add:
```
dolphin_auto_gamepad=0
```
to disable dolphin standalone auto detection.

I am looking into why this issue occurs for you. It's very strange that GCPad1 is skipped or somehow removed.

Can you connect one xbox controller and in SSH do commnd:
```
cat /proc/bus/input/devices
```
and post in a new gist with link here: https://gist.github.com/

Langerz82, 2022-06-04:
Also try running the auto joy script manually with debugging and post all output in a gist, to do run in SSH:
```
bash -x /usr/bin/set_dolphin_joy.sh
```
Make sure also dolphin_auto_gamepad=1 when running the debug.

Langerz82, 2022-06-06:
@microrache hello you there? I would like to work on this issue but without logs I cannot proceed.

Langerz82, 2022-06-07:
@microrache 

Your device:
```
OSMC Remote Controller  USB Keyboard Mouse
```
Is getting picked up as a valid controller, then failing when it tries to create the config since there is no SDL gamecontroller map for the device. I assume you do not use this device for your games? let me know, cheers.

microrache, 2022-06-08:
> Your device:
> 
> ```
> OSMC Remote Controller  USB Keyboard Mouse
> ```
> 
> Is getting picked up as a valid controller, then failing when it tries to create the config since there is no SDL gamecontroller map for the device. I assume you do not use this device for your games? let me know, cheers.

Ah, good to know. No, I don't use it for games. It's just a media center IR-remote-control since I use my device with dual boot for both, CoreELEC and EmuELEC and I need that remote control to switch between these two OSs. Is it possible to somehow disable or "blacklist" this device within EmuELEC controller configuration?

7Ji, 2022-06-08:
An easy workaround of this instead of digging through codes and trying fo implement a whole order fix mechanism other can simply unplugging the remote is to unbind the remote from the corresponding driver
````
echo 0003:2017:1688.0004 > /sys/bus/hid/drivers/hid-generic/unbind
echo 0003:2017:1688.0005 > /sys/bus/hid/drivers/hid-generic/unbind
````
You may need to replace hid-generic if it's using another driver (``ls /sys/bus/hid/drivers/*`` to find which driver is driving your 0003:2017:1688.0004/5)
You can add these lines to /storage/.config/autostart.sh (create it if not exists, remember to add execute permission and #!/bin/sh header line) so they will be automatically executed after bootup

Langerz82, 2022-09-27:
Test the fix with this PR build.
https://github.com/EmuELEC/EmuELEC/pull/936

Langerz82, 2022-12-12:
This should be fixed. If it's still an issue re-open.

closed 2022-12-12 by Langerz82