EmuELEC/EmuELEC #939: SDL2 Mali patch is missing mouse cursor / keyboard issues

shantigilbert, 2022-06-13:
The Mali patch used to add FBDEV support (https://github.com/EmuELEC/EmuELEC/blob/dev/packages/multimedia/SDL2/patches/Amlogic/SDL2-git-2.0.22-001-enable_mali.patch) is missing the mouse cursor, so some games that use SDL2 with mouse (like ScummVM 3D games, Heroes of Might and Magic, etc) do now show the cursor on screen, the mouse works but the cursor is invisible. 

Recently we also noticed that the keyboard/mouse does not work at all on some programs https://github.com/EmuELEC/EmuELEC/issues/879 (ScummVM) using the newest SDL2 (it works fine on SDL2.0.9) so something changed that also needs fixing. 

If you anyone can offer some help on this issue it will be greatly appreciated, a bounty might be opened if required.

Langerz82, 2022-09-22:
As discuss setting fb1 to a higher value like 64 makes the cursors work again. Do I get the bounty for the fix lol?

Langerz82, 2022-09-22:
In setres.sh:
```
BUFF=64
fbset -fb /dev/fb1 -g $BUFF $BUFF $BUFF $BUFF $BPP
```
FIXED!! Haha

shantigilbert, 2022-09-23:
> As discuss setting fb1 to a higher value like 64 makes the cursors work again. Do I get the bounty for the fix lol?

Please check on these: 

Scummvm with a 3d game
half life
Julius/Augustus (Caesar 3)

if the cursor appears then yes this would be resolved and you would get the bounty.

shantigilbert, 2022-09-23:
Also make sure you are not confused as the HOMM issue was fixed here https://github.com/EmuELEC/EmuELEC/commit/25c190a2647660963d74f365dc220ecad9165586, so try other games like I mentioned

Langerz82, 2022-09-23:
My latest PR doesn't change the fb1 buffer anymore. Missing text was occurring in ES due to setting it too low. When I get a chance I'll test cheers.

dajoho, 2023-12-15:
I have detailed my debugging of this bug here: https://github.com/EmuELEC/EmuELEC/issues/879#issuecomment-1858164600

shantigilbert, 2026-10-12:
Potentially fixed by #1688