EmuELEC/EmuELEC #941: Boot loop rebooting kernel panic

favarese91, 2022-06-15:
hi guys, i installed emuelec on a 128gb sd card for a 4gb ram hk1 series tv box.  installed emuelec I configured everything and loaded the roms.  i played a couple of times and then i used the tv box to watch movies.  after 1 week, I turn on the tv box to play games with emuelec and it gives me this error in the picture.  to premise that before it worked perfectly, then no longer used and at the next power on it gives me this error in the screen and the boot is in loop, how can I solve?  I used the official version and it is 4.5

[image]

7Ji, 2022-06-15:
The SYSTEM image (the file SYSTEM you can see under drive EMUELEC on Windows) on your SD card is corrupted and some files in it inclucding ``/usr/lib/libzstd.so.1`` is corrupted
Coupld of reasons this could happen:
1. Faulty SD card
2. Naughty Android behaviours, either by Android itself or some apps, they mount SD card as read+write to do some "helpful" modification and failed some write operation which results in corrupted SD card
3. Unstable power supply, the stock power adpater for HK1 (5V2A DC5525) is very crappy, and is notorious for fry SD cards/USB drives that connect to the drive for very long time. I use a second-hand 5V5A power adapter bundled with a Cisco switch and I suggest you get any adapter better than the stock one.

The actual reason could be a mix of the above

You can still try to 'fix' the drive but I doubt if it's still fixable, the first ``EMUELEC`` partition can be fixed on Windows, and you can extract the SYSTEM image from a latestet update tarboll to replace the corrupted one. The second ``STORAGE`` partition can only be fixed on Linux. Be noticed these operations are not supported

Best way is to reburn the EmuELEC image to the SD card, and get a better power adapter, and to hope the box won't fry the card anymore...

favarese91, 2022-06-15:
thanks for the quick reply?  how can i repair partitions?  from both windows and linux?  i have both operating systems

> The SYSTEM image (the file SYSTEM you can see under drive EMUELEC on Windows) on your SD card is corrupted and some files in it inclucding `/usr/lib/libzstd.so.1` is corrupted Coupld of reasons this [...]

7Ji, 2022-06-15:
Since you have Linux I'll explain the Linux way
Suppose your sdcard is ``/dev/sda``, run these 3 commands to fix all 3 partitions on it:
1. ``dosfsck -w -r -l -a -v -t /dev/sda1``  
``-w`` is for write to disk immediately, ``-r`` if for interactive (so you will be asked whether some errors should be fixed), ``-l`` is to list he filenames processed (so you can know if the ``SYSTEM`` image is fixed), ``-a`` is for automatical fix (so you won't be asked too many questions), ``-v`` is for verbose log, ``-t`` is for marking unreadable clusters as bad (if at least one is marked as bad, then this installation is out of luck and you should re-burn the image)
2. ``e2fsck -f /dev/sda2``  
``-f`` for fix issues
3. ``dosfsck -w -r -l -a -v -t /dev/sda3`` , by default ``EEROMS`` is formatted as FAT32 so this should be used, but if you've modified it to other FS, then you should use other corresponding commands

If all 3 commands succeccds, then you can plug the SD card and try to boot it
Then if it still failed to boot, you can try to download the corresponding .tar file from the latest release, and extract some stuffs under ``TARGET`` and ``3rdparty/bootloader`` to replace corresponding files under the first partition. These are the files map:
1. ``TARGET/KERNEL`` => ``kernel.img``
2. ``TARGET/SYSTEM`` => ``SYSTEM``
3. ``3rdparty/bootloader/aml_autoscript`` => ``aml_autoscript``
4. ``3rdparty/bootloader/Generic_cfgload`` => ``cfgload``
5. ``3rdparty/bootloader/device_trees/[You device tree]``=> ``dtb.img``

An alternative way to do this is to dd the first part in a full image to the first partition, you're a Linux user and I know you can figure it out by yourself : )

This is not guaranteed to work, and if the system still can't be booted, then you should re-burn the image

7Ji, 2022-06-18:
Since this is not an issue with EmuELEC itself, but an issue with your hardware, there's nothing we could do to prevent it from happening in EmuELEC. I'm closing this issue as it is unfixable.

closed 2022-06-18 by 7Ji