EmuELEC/EmuELEC #945: Ghost/Phantom/Duplicated joypad inputs

dyorgio, 2022-06-19:
Hello, I have using EmuELEC with two diff hardwares: TX-mini and GT-king.

Both are generating duplicated input to my arcade buttons (any, directions and action buttons) sometimes.

I tested using jstest and see that:

```.log
# Example 1
Event: time 1655576476.476594, type 3 (EV_ABS), code 16 (ABS_HAT0X), value -1
Event: time 1655576476.476594, type 3 (EV_ABS), code 0 (ABS_X), value 0
Event: time 1655576476.476594, -------------- SYN_REPORT ------------
Event: time 1655576476.556599, type 3 (EV_ABS), code 16 (ABS_HAT0X), value 0
Event: time 1655576476.556599, type 3 (EV_ABS), code 0 (ABS_X), value 128
Event: time 1655576476.556599, -------------- SYN_REPORT ------------
Event: time 1655576476.557599, type 3 (EV_ABS), code 16 (ABS_HAT0X), value -1
Event: time 1655576476.557599, type 3 (EV_ABS), code 0 (ABS_X), value 0
Event: time 1655576476.557599, -------------- SYN_REPORT ------------
Event: time 1655576476.561599, type 3 (EV_ABS), code 16 (ABS_HAT0X), value 0
Event: time 1655576476.561599, type 3 (EV_ABS), code 0 (ABS_X), value 128
Event: time 1655576476.561599, -------------- SYN_REPORT ------------

# Example 2
Event: time 1655576856.388740, type 3 (EV_ABS), code 16 (ABS_HAT0X), value 1
Event: time 1655576856.388740, type 3 (EV_ABS), code 0 (ABS_X), value 255
Event: time 1655576856.388740, -------------- SYN_REPORT ------------
Event: time 1655576856.392740, type 3 (EV_ABS), code 16 (ABS_HAT0X), value 0
Event: time 1655576856.392740, type 3 (EV_ABS), code 0 (ABS_X), value 128
Event: time 1655576856.392740, -------------- SYN_REPORT ------------
Event: time 1655576856.393740, type 3 (EV_ABS), code 16 (ABS_HAT0X), value 1
Event: time 1655576856.393740, type 3 (EV_ABS), code 0 (ABS_X), value 255
Event: time 1655576856.393740, -------------- SYN_REPORT ------------
Event: time 1655576856.519740, type 3 (EV_ABS), code 16 (ABS_HAT0X), value 0
Event: time 1655576856.519740, type 3 (EV_ABS), code 0 (ABS_X), value 128
Event: time 1655576856.519740, -------------- SYN_REPORT ------------

# Example 3
Event: time 1655577018.163691, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90003
Event: time 1655577018.163691, type 1 (EV_KEY), code 306 (BTN_C), value 1
Event: time 1655577018.163691, -------------- SYN_REPORT ------------
Event: time 1655577018.324690, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90003
Event: time 1655577018.324690, type 1 (EV_KEY), code 306 (BTN_C), value 0
Event: time 1655577018.324690, -------------- SYN_REPORT ------------
Event: time 1655577018.328691, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90003
Event: time 1655577018.328691, type 1 (EV_KEY), code 306 (BTN_C), value 1
Event: time 1655577018.328691, -------------- SYN_REPORT ------------
Event: time 1655577018.336691, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90003
Event: time 1655577018.336691, type 1 (EV_KEY), code 306 (BTN_C), value 0
Event: time 1655577018.336691, -------------- SYN_REPORT ------------

# Example 4
Event: time 1655577136.366192, type 3 (EV_ABS), code 17 (ABS_HAT0Y), value -1
Event: time 1655577136.366192, type 3 (EV_ABS), code 1 (ABS_Y), value 0
Event: time 1655577136.366192, -------------- SYN_REPORT ------------
Event: time 1655577136.502190, type 3 (EV_ABS), code 17 (ABS_HAT0Y), value 0
Event: time 1655577136.502190, type 3 (EV_ABS), code 1 (ABS_Y), value 128
Event: time 1655577136.502190, -------------- SYN_REPORT ------------
Event: time 1655577136.504191, type 3 (EV_ABS), code 17 (ABS_HAT0Y), value -1
Event: time 1655577136.504191, type 3 (EV_ABS), code 1 (ABS_Y), value 0
Event: time 1655577136.504191, -------------- SYN_REPORT ------------
Event: time 1655577136.505191, type 3 (EV_ABS), code 17 (ABS_HAT0Y), value 0
Event: time 1655577136.505191, type 3 (EV_ABS), code 1 (ABS_Y), value 128
Event: time 1655577136.505191, -------------- SYN_REPORT ------------
```

In all cases two last events are ghost/phantom/duplicated (not real, see the u_seconds[millis])

I'm now developing a executable to grab joypad input device and filter it to a uinput device 😋 

But maybe someone knows wtf is happening here 🤓 

This happen in EmuELEC 3.9, 4.2 and 4.5.

Joypad deviceGUID="03000000c4100000c082000001010000"
```properties
input_device = "Bencao74 MGames Joystick Player 1"
input_driver = "udev"
input_vendor_id = "4292"
input_product_id = "33472"
```

shantigilbert, 2022-06-19:
This is most likely your gamepad  causing this, test with another gamepad

dyorgio, 2022-06-19:
Hi @shantigilbert, tks for replying 😃.

I also believe that, but I'm trying to solve it in software layer, because there is another kind of problems that can be solved filtering /dev/input events...

I'm already filtering /dev/input/event4 to /dev/input/event7 (theses are my real(4) and virtual(7) event devices).

Now I'm stucked on get my virtual device listed as a /dev/input/js* one (I can test it with evtest, but not with jstest and SDL/ES cannot recognize it).

Do you have some trick/magic command, to do it?

shantigilbert, 2022-06-19:
> Do you have some trick/magic command, to do it?

No idea, sorry. But as this is not an EmuELEC issue I will close this, you might want to continue this on the forum or Discord

closed 2022-06-19 by shantigilbert