EmuELEC/EmuELEC #952: some openssl error when configuring qt

Pzy2002, 2022-07-06:
# BEFORE YOU OPEN AN ISSUE MAKE SURE IT IS NOT ALREADY RESOLVED IN THE WIKI https://github.com/EmuELEC/EmuELEC/wiki

# MAKE SURE YOUR DEVICE IS A [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices)  

# PLEASE DO NOT POST GENERAL HELP QUESTIONS! EITHER USE THE FORUM (https://emuelec.org) OR USE DISCORD FOR THAT (https://discord.gg/cbgtJTu), THIS IS ONLY FOR BUG/ISSUES REPORTING!

# DELETE EVERYTHING ABOVE THIS LINE (INCLUDING THIS LINE) BEFORE SUBMITTING TO MAKE SURE YOU'VE READ THIS!

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**EmuELEC version:**
 - Version [e.g. 22]
 - v4.5

**Device (please complete the following information):**
 - SOC: [e.g. S095x3]
 - Brand: [e.g. Beelink]
 - Model [e.g. mini g1]
 - Version [e.g. 22]
 - (not modified, use demo default)

**Additional context**
Add any other context about the problem here.
crosscompile at ubuntu 20.04 lts（vmware，64-bit），with the newest dev branch（2022/07/07）
it occurs error when compiling qt （openssl issue），but i had already tested many openssl version such as ubuntu default（with libssl-dev） ，openssl 3.0，openssl 1.1f，and openssl 1.02
weird.
[image]
[image]

Pzy2002, 2022-07-07:
i found it might caused i installed libraries via apt and sourcefile and so on.
and also use apt downloaded an arm-gcc and aarch64-gcc compiler.
might system choose them as default.

Pzy2002, 2022-07-07:
but a new issue occurs
[image]

Pzy2002, 2022-07-07:
> but a new issue occurs [image]

it could use -a (modify image script)  option to skip
but sed command occurs another error

Pzy2002, 2022-07-07:
./scripts/image: line 245: ${INSTALL}/usr/lib/kernel-overlays/base/lib/modules/${MODVER}/modules.order: ambiguous redirect

british-choi, 2022-07-08:
> **Describe the bug** ssl error when configure qt
> 
> **To Reproduce** Steps to reproduce the behavior: run compile as sampile in readme.md
> 
> **Expected behavior** shuold pass qt compile
> 
> **Screenshots** below provided
> 
> **EmuELEC version:**
> 
> * v4.5
> 
> **Device (please complete the following information):**
> 
> * (not modified, use demo default)
> * Host (ubuntu ver.20.04_lts  x86_64 with vmware)
> * target: (demo) Amlogic-ng one
> 
> **Additional context** Add any other context about the problem here. crosscompile at ubuntu 20.04 lts（vmware，64-bit），with the newest dev branch（2022/07/07） it occurs error when compiling qt （openssl issue），but i had already tested many openssl version such as ubuntu default（with libssl-dev） ，openssl 3.0，openssl 1.1f，and openssl 1.02 weird. !

https://github.com/EmuELEC/EmuELEC/tree/dev/packages/sx05re/tools/qt-everywhere/package.mk
PKG_DEPENDS_TARGET="pcre2 zlib openssl"

Plz add `openssl` in depends target
and try rebuild qt-everywhere

7Ji, 2022-07-08:
> ./scripts/image: line 245: INSTALL/usr/lib/kernel−overlays/base/lib/modules/{MODVER}/modules.order: ambiguous redirect

Don't build in a directoty whose path contains non-ascii characters (in this case Chinese characters since you are using a Chinese env, and I guess it's 桌面), and it is not an issue to fix where it could be simply omitted by building in a pure ascii path.

7Ji, 2022-07-08:
@Pzy2002 
Both of your PR were closed in CoreELEC repo as they are not done in the correct way, but I encourage you to do it here in the EmuELEC repo in a correct way:
1. Fork the EmuELEC repo
2. Create a new branch qt-everywhere-fix
3. Add the openssl as dependency for qt-everywhere in its PKG_DEPENDS_TARGET
4. Commit the change, and push the commit to your forked repo
5. Open a Pull Request here to merge the fix from yours:qt-everywherefix branch into emuelec:dev branch, describing the issue and how it's fixed

: ) PR are always welcomed as long as they are done in the correct way