EmuELEC/EmuELEC #962: Version 4.5 bug, one I resloved,one I have no idea

zouplus, 2022-07-26:
bug---------------->01
Version:4.5
device cpu: S905l3a 、S905l3b （I try three kind of devices）
happend：boot screent
there is many error log such as
"/dev/kmsg not permitted"
"failed to get xxxx .service"

It seem 4.5 and 4.4 use the system.d version over 242(the linux core version maybe low,so if there is no necessary，you can use 242 version，or try high linux version）

tips you can close issue:
https://github.com/EmuELEC/EmuELEC/issues/916

bug---------------->02
Version：4.5
device cpu: S905l3a 、S905l3b （I try three kind of devices）
happend: if use multiple controllers ，that you cannot change the input order，for example：player 1、player 2、player 3、player 4；if change controller order port 4 to port 1，the Retroarch seem still use random port ，some time：port 1 use controller 1、port 4 also use controller 1
but the function work fine at ee3.9
So I consider to read RA doc to resolve it

----------------------------
 suggest：
the es for emuelec you can add code for dealy excute function that can help devloper to excute some function when ui animation finished。（or and some other API on es animtion finished ！！！ hahaha）。why need delay？because if some function with subthread，that system font will render error！！！

7Ji, 2022-07-26:
For bug 01, are you refering to console spam during splash screen? This is not considered a bug yet as S905L version 3 (whether step a or step b) is not officially supported and you're probably using the wrong dtb (There's not even dtbs for them, you're probably using dtb for S905X2 which is actually incompatible with your device). 

Main cause would be the improper SDIO speed. This is not reproducible for other supported devices and should be considered as "won't fix", or at least not fixable here as this is more or less a kernel issue since the 4.9 kernel we're using doesn't even support the device you're using.

7Ji, 2022-07-26:
A workaround for bug 01 would be to use a USB boot drive, and this is more nature as devices like E900V22C, E900V22D, CM311 do not have microSD card slot AT ALL. The SD card slot you're using is most likely soldered on afterwards, not by the manufacturer, the device was never planned to be booted on with a SD card. And no, these 3 devices will never be properly supported without a proper dtb for S905L3A/B

Fix for controller ordering is mostly covered the the following PR:
https://github.com/EmuELEC/EmuELEC/pull/936
A workaround for now would be to manually assign controllers to players in the ES menu, not in the RA.

zouplus, 2022-07-26:
A workaround for now would be to manually assign controllers to players in the ES menu, not in the RA:
START(Menu) -> Controller & Bluetooth Settings -> P1's Controller -> Assign -> P2's Controller ...
----------------------------------------------
I tryed ,but still not work, but when I use ee3.9 on es menu,it work fine

zouplus, 2022-07-26:
by the way , when "reset system to default" , the use interface will hold and not reboot .

zouplus, 2022-07-26:
may i use chinese to communication？My English so bad.

7Ji, 2022-07-26:
https://github.com/EmuELEC/EmuELEC/issues/963#issuecomment-1195235873
> but if instead of version of 4.5 "packages/sysutils/systemd" content with “EE 3.9” ，it may fix.(the screen not show at less)

We won't downgrade packages in EmuELEC unless the package causes severe problems. The systemd used in 3.9 is version 242 which is very outdated and suited only for the older non-ng images, and we're using 251.2 for 4.5 now.

If you're willing to tinker with that package, you can swap things half by half and find out which exactly unit is causing the console spam. But I doubt it's any of them; it could probably be systemd itself which as I said we won't downgrade.

And after all, the screen only "not show", that does not actually fix any problem, which I don't think is a valid fix. But anyway, if you find a way to fix this in systemd 251.2, PR is always welcomed : )

7Ji, 2022-07-26:
> but when I use the 242 version of system.d .... there is no error log at screen at less

Then we must wait for Langerz82's joystick order fix. He's the only one working on this and as you can see that PR is just a draft and far from complete.

> may i use chinese to communication？My English so bad.

Any language other than English is not recommended here as other future readers can't get any idea what the problem is and how it's fixed. The lead dev shanti is a Mexican and I'm a Chinese (I don't know exactly other devs' background), but both of us use English on github and discord.

zouplus, 2022-07-26:
Maybe I can compare with the version which one is work fine for the controller setting at es

shantigilbert, 2022-07-26:
These 2 issues are related to dtb,  this is a "won't fix" as there are dtbs for those devices. I suggest that if 3.9 works fine for you you keep using that, unless someone can provide a working dtb. As far as I know these issues are not present on supported hardware

zouplus, 2022-07-27:
but the version of 3.9 flycastSA doest not work,And when play “Metal Slug 6”（use flycast retro core）,the end of stage one is Serious frame loss. but if use 4.5 there is no problem（use flycast retro core x64）

7Ji, 2022-07-27:
So, as shanti saied we won't "fix" this as it isn't even an issue found on our supported hardware.

Feature wise a PR to add dtb for S905L version 3 step a/b is still welcomed, but it is more suggested to be submitted to upstream CoreELEC's CoreELEC repo or linux-amlogic repo

> but the version of 3.9 flycastSA doest not work,And when play “Metal Slug 6”（use flycast retro core）,the end of stage one is Serious frame loss. but if use 4.5 there is no problem（use flycast retro core x64）

Addtionally as a workaround, you can try to bump flycastSA in your 3.9 local branch, this should be doable and not that diffucult, as you could already partially downgrade the systemd and compile a full image.

Closing this, as it is not considered an issue we won't fix it.

closed 2022-07-27 by 7Ji