EmuELEC/EmuELEC #963: Version 4.5 bug, one I resloved,one I have no idea

zouplus, 2022-07-26:
bug---------------->01
Version:4.5
device cpu: S905l3a 、S905l3b （I try three kind of devices）
happend：boot screent
there is many error log such as
"/dev/kmsg not permitted"
"failed to get xxxx .service"

It seem 4.5 and 4.4 use the system.d version over 242(the linux core version maybe low,so if there is no necessary，you can use 242 version，or try high linux version）

tips you can close issue:
https://github.com/EmuELEC/EmuELEC/issues/916

bug---------------->02
Version：4.5
device cpu: S905l3a 、S905l3b （I try three kind of devices）
happend: if use multiple controllers ，that you cannot change the input order，for example：player 1、player 2、player 3、player 4；if change controller order port 4 to port 1，the Retroarch seem still use random port ，some time：port 1 use controller 1、port 4 also use controller 1
but the function work fine at ee3.9
So I consider to read RA doc to resolve it

----------------------------
 suggest：
the es for emuelec you can add code for dealy excute function that can help devloper to excute some function when ui animation finished。（or and some other API on es animtion finished ！！！ hahaha）。why need delay？because if some function with subthread，that system font will render error！！！

7Ji, 2022-07-26:
Duplicate of https://github.com/EmuELEC/EmuELEC/issues/962, close and we should discuss this in there only

closed 2022-07-26 by 7Ji

zouplus, 2022-07-26:
Okay

zouplus, 2022-07-26:
For bug 01, are you refering to console spam during splash screen? This is not considered a bug yet as S905L version 3 (whether step a or step b) is not officially supported and you're probably using the wrong dtb (There's not even dtbs for them, you're probably using dtb for S905X2 which is actually incompatible with your device).

Main cause would be the improper SDIO speed. This is not reproducible for other supported devices and should be considered as "won't fix", or at least not fixable here as this is more or less a kernel issue since the 4.9 kernel we're using doesn't even support the device you're using.
------------------------------------------------

but if instead of version of 4.5 "packages/sysutils/systemd"  content with “EE 3.9” ，it may fix.(the screen not show at less)

zouplus, 2022-07-26:
sorry, may use the chinese to response those question,my English so bad,you can read my content via browser “translate” tool...

zouplus, 2022-07-26:
sorry, may I use the chinese to response those question,my English so bad,you can read my content via browser “translate” tool...

zouplus, 2022-07-26:
A workaround for now would be to manually assign controllers to players in the ES menu, not in the RA.
----------------------------
I manually config the players in ES menu ; but it not work at 4.5;3.9 work fine

7Ji, 2022-07-26:
I said... You opened two issues with the same content, this issue (963) is closed, and we should discuss in the early one (962). Yet you are reading in 962 yet replying in 963... This just makes things more strange and people need to jump between the two issues to read the whole discussion :/