EmuELEC/EmuELEC #964: Bezels not showing on most systesm in 4.6test

TimeWarpGamer, 2022-07-30:
Used update --> Bezels for a whole bunch of systems on 4.6test, and bezels and configs seemed to all install correctly, but the only systems that actually display bezels are Atomiswave, Naomi, Dreamcast, N64, Sega Saturn, and Vectrex.  I have about 40ish systems set up, and none of the other bezels showed properly.  To futher test, I copied the system folder (i.e. arcade, ngpc) from Bezels onto a different SD card with 4.5 installed, and bezels showed up perfectly.  So it appears that there's some sort of broad issue with RA correctly applying bezels/configs in 4.6test.

7Ji, 2022-07-30:
You must set Bezel=Yes for the whole system or for certain systems for bezels to be displayed for some systems instead of Auto, at least on 4.6Test and newer dev builds 
Global: Main screen -> START -> Game settings -> Enable RA Bezels -> Yes
Per-system: System gamelist -> SELECT -> Advanced System Options -> Bezel -> Yes
Perhaps you enabled this in your old 4.5 install and forgot? Or maybe the default setting was changed between 4.5 and 4.6test?

TimeWarpGamer, 2022-07-30:
Thought I had checked the bezel toggle in ES and the config in RA for
overlay, but will check again tonight to be certain.  Thanks!

shantigilbert, 2022-07-30:
> You must set Bezel=Yes globally or for certain systems for bezels to be displayed for some systems instead of Auto, at least on 4.6Test and newer dev builds

This should not be the case if the system has Bezels set to "Yes" as it should take priority over the global. If it is then this should be considered a bug, I will check it later when I get some time.

7Ji, 2022-07-30:
I meant either global on + per-system auto/on, or global off + per-system on

TimeWarpGamer, 2022-07-30:
Just checked again.  Since NGPC is one I copied over to 4.5 and made bezels
work no problem I looked at that.  Global RA bezel is on.  RA display
overlay is on.  Even went into advanced system setting and changed bezel
from auto to yes.  Still no bezels showing up.

TimeWarpGamer, 2022-07-30:
Not sure if it’s a related RA config issue, but I also noticed in FB Neo
that vertical arcade games (ie 1942) are not displaying in proper “skinny”
aspect ratio, even though RA says it’s using “core provided”.  Oddly, even
when I tried to manually set rez to 3:4 aspect ratio, nothing happened.

shantigilbert, 2022-07-30:
> Not sure if it’s a related RA config issue, but I also noticed in FB Neo that vertical arcade games (ie 1942) are not displaying in proper “skinny” aspect ratio, even though RA says it’s using “core provided”. Oddly, even when I tried to manually set rez to 3:4 aspect ratio, nothing happened.

Lets keep the issue to one thing at a time please

TimeWarpGamer, 2022-07-30:
Sorry, will do in the future :-)

TimeWarpGamer, 2022-07-30:
Might be a corrupt sd card on my end...Will try another.

TimeWarpGamer, 2022-07-31:
New SD works fine and I no longer have the issues described above.  Sorry for the confusion.  Closing this one.

closed 2022-07-31 by TimeWarpGamer