EmuELEC/EmuELEC #969: freej2me cannot save

25minutes, 2022-08-10:
**Describe the bug**
freej2me in EmuELEC v4.5 cannot save. This seems to be an issue of EmuELEC's build only, because Retropie build actually works (see details below).

**To Reproduce**
Steps to reproduce the behavior:
1. Load any jar game with save feature (I use Heroes Lore, a RPG game, as an example).
2. In emuelec.log, we can see error "javax.microedition.rms.RecordStoreException: Problem Creating Record Store Path rms/Heroes Lore".

**Expected behavior**
According to author's github, save should work (https://github.com/hex007/freej2me/issues/69)
I also try it on Retropie (install via https://github.com/zerojay/RetroPie-Extra/blob/master/scriptmodules/libretrocores/lr-freej2me.sh), and verify that save works. 

**EmuELEC version:**
 - Version 4.5

**Device (please complete the following information):**
 - SOC: S095x2

shantigilbert, 2022-08-10:
Did you create the folders as they say on that issue? `/storage/roms/bios//freej2me/rms` and `/storage/roms/bios/freej2me/config` ??

25minutes, 2022-08-10:
freej2me code might have changed. Based on my testing on Retropie, the rms folder (where save is stored) is created in user's home directory. Actually, I don't need manually create it as it's automatically created.

But on EmuELEC, even if I manually create the rms folder in both bios folder and home directory, save still doesn't work. BTW, config folder is automatically created on EmuELEC. With default setting, it's created under freej2me roms folder. I've also tried to create rms folder at the same level as config folder, but it's not working either.

shantigilbert, 2022-08-11:
Please try the attached core: 

.jar goes into `/storage/roms/bios`
.so goes into `/tmp/cores`
Overwrite the old ones

saves will be saved in `/emuelec/configs/freej2me/rms/`

```
EMUELEC:/emuelec/configs/freej2me/rms/Heroes Lore # ls -la
total 5
drwxr-xr-x    2 root     root          1024 Aug 10 19:38 .
drwxr-xr-x    3 root     root          1024 Aug 10 19:36 ..
-rw-r--r--    1 root     root            14 Aug 10 19:36 _c
-rw-r--r--    1 root     root           440 Aug 10 19:38 _k
-rw-r--r--    1 root     root            30 Aug 10 19:38 _o
```

[freej2me.zip](https://github.com/EmuELEC/EmuELEC/files/9305289/freej2me.zip)

closed 2022-08-11 by shantigilbert

haha496744, 2022-09-07:
@shantigilbert 
EmuELEC version:
Version 4.5
I followed your method,
Please try the attached core:

.jar goes into /storage/roms/bios
.so goes into /tmp/cores
Overwrite the old ones

 but now I can't open the jar, it was working normally before and can't be saved

[image]

commit 890e6c7b (2022-08-11): freej2me: Move save folder to fix saves

Fixes #969

Also we make sure we are using the provided jar