EmuELEC/EmuELEC #975: RetroArch vibration strength preference is not honoured

gingerbeardman, 2022-08-13:
**Describe the bug**

RetroArch vibration strength preference is not honoured

**To Reproduce**
Steps to reproduce the behavior:
1. Start a game with rumble support
2. Witness rumble in a reproducible way
3. Change `Settings > Input > Haptic Feedback/Vibration > Vibration Strength` to less than 100%
4. Reproduce rumble in game and check whether the intensity of the vibration has changed

**Expected behavior**
- vibration strength preference should be honoured 
- vibration strength/intensity should change accordingly

**Screenshots**
n/a

**EmuELEC version:**
 - Version 4.5 stable

**Device (please complete the following information):**
 - SOC: RK3326
 - Brand: GameForce
 - Model: Chi
 - Version: beige

**Additional context**

i just got rumble to work in the PS1 game I'm playing by editing the libretro-database .dat files, generating updated .rdb files and copying them onto my Chi <https://github.com/libretro/libretro-database/issues/64#issuecomment-1213464021>
Default rumble in RA on Chi is *fierce*! I thought it could be changed at *Settings > Input > Haptic Feedback/Vibration > Vibration Strength* but it makes no difference

Langerz82, 2022-08-28:
You might want to raise this with retroarch as it is an input setting in retroarch that is not working.
see:
https://github.com/libretro/RetroArch/issues?q=is%3Aissue+vibration

gingerbeardman, 2022-08-28:
> You might want to raise this with retroarch as it is an input setting in retroarch that is not working.
> see:
> https://github.com/libretro/RetroArch/issues?q=is%3Aissue+vibration

I'm sorry @Langerz82 can you clarify what you mean? 

The link you posted takes me to a multitude of seemingly unrelated issues. 

How did you come to the conclusion that it's the RetroArch setting that is broken?

Langerz82, 2022-08-28:
Sorry I should have been more specific. It's an issue within retroarch that the vibration is not turning down right? Retroarch is a package of EmuELEC but it's developers are on the Retroarch github, you should lodge the issue there (as a new issue if one is not present). Retroarch configurations do not persist outside of Retroarch to EmulationStation and the other emulators.

This is the version it's occurring on if you are using EE v4.5:
https://github.com/libretro/RetroArch/tree/7bdcc8bfe179aa0eee2e60858c4d5b806d34445d

gingerbeardman, 2022-08-30:
> It's an issue within retroarch that the vibration is not turning down right? 

This is an assumption. This is what I want you to explain.

How do you know it's ReotrArch and not EmuELEC not hooking the value of the slider?

(this sort of EmuELEC problem has happened in the past with screen brightness and various other settings)

Langerz82, 2022-08-30:
EmuELEC does not hook into retroarch, it merely pass on settings through config files. I'm a dev of this project I know a bit about it lol. If you dont believe me I dunno what to tell you lol.

shantigilbert, 2022-08-30:
> This is an assumption.

Sure, but I already told you we are not sure the motors are PWM or "on/off" A thing you can do is to try another distro, and see if the rumble work as you want there. Lakka is a good choice since its 100% Retroarch. 

> I feel it's an incorrect assumption, given my experience with similar EmuELEC bugs. (this sort of EmuELEC problem has happened in the past with screen brightness and various other settings)

Can you elaborate? this was a Retroarch issue and it was fixed IN Retroarch, EmuELEC does not control what Retroarch do other than the settings, and AFAIK there are no settings in EmuELEC to affect vibration/rumble.

Langerz82, 2022-08-31:
Not sure if it's even retroarch as it probably would have been reported by now. I think it might be the driver that connects the physical device to retroarch. If you open in SSH /tmp/joypads and find your controllers config file it should which list input_driver= it uses.

gingerbeardman, 2023-02-15:
Here we go

[image]

Langerz82, 2023-02-16:
You could try manually changing the percentage in /storage/.config/retroarch.cfg to input_rumble_gain = "50". If that does not work I suggest asking GameForce support if they can look into the issue. I looked for alternative input drivers you could use but could not find any for GameForce controls.

You could maybe try and load it with this file no guarantee it will work though, but it's worth a try:
https://gist.github.com/Langerz82/216e3dba45406f41e2dbbce739c858db

gingerbeardman, 2023-02-16:
The percentage can be set through RetroArch menu system, the issue is that it is not honoured.

closed 2024-09-02 by Langerz82

gingerbeardman, 2024-09-02:
Shame