EmuELEC/EmuELEC #977: [4.5/4.6]RetroArch font show box temporary solution

zouplus, 2022-08-16:
**Describe the bug**
When I just selected Chinese and Korean as the language in RA, the font still displayed normally, but after saving and exiting, restart RA, and the font became a box.

**To Reproduce**
Steps to reproduce the behavior:
1. Retroarch-->user--->language(font right)
2. After saving and re entering, the font becomes box

**EmuELEC version:**
- Version [4.5]
- Version [4.6 test]
- 
**Device (please complete the following information):**
- X96max plus

---------------------------------The temporary solutions are as follows:
Don't adopt the style of ozone, use XMB instead
A. Find the font you need and copy it to the storage partition
B. Select the font as follows:
1. Enter [setting \ user interface \] to open [show advanced settings];
2. Enter [setting \ user interface \ menu] and switch the system skin to [XMB];
3. Enter [setting \ user interface \ appearance], open [menu font], find and select the font in the first step directory;
At this step, the font setting is completed, but the system interface is still in English, and it needs to be further switched to other languages

4. Enter [setting \ user], open [language] to set your language, and you are done.

zouplus, 2022-08-16:
and the ES on EE4.5 may should set xmb also

shantigilbert, 2022-08-16:
Duplicate #534 #431

closed 2022-08-16 by shantigilbert

zouplus, 2022-08-16:
omg...