EmuELEC/EmuELEC #978: Emuelec 4.6-TEST-07082022

Sidordroid, 2022-08-16:
In Emuelec 4.6 -TEST, the video player control does not work - it is not paused by pressing the Right mini-joystick / "R3" button. Because of this, I rolled back to version 4.5. Fix the video player control if possible

haha496744, 2022-09-07:
@Sidordroid 
In the config file there is a mpv folder
Create a new input.conf file
The r1 of my bluetooth controller is the screenshot key, which corresponds to the s key on the keyboard. I changed the screenshot key to the q key, and the q key on the keyboard is quit
You can modify the keys you want in this way. This has a default keyboard profile
[input.zip](https://github.com/EmuELEC/EmuELEC/files/9502631/input.zip)

Langerz82, 2022-12-12:
Is this an issue with 4.6 release as well? If not want me to close it?

closed 2023-01-14 by shantigilbert