EmuELEC/EmuELEC #986: Emuelec 4.6-TEST-07082022  Bluetooth handle

haha496744, 2022-09-05:
# BEFORE YOU OPEN AN ISSUE MAKE SURE IT IS NOT ALREADY RESOLVED IN THE WIKI https://github.com/EmuELEC/EmuELEC/wiki

# MAKE SURE YOUR DEVICE IS A [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices)  

# PLEASE DO NOT POST GENERAL HELP QUESTIONS! EITHER USE THE FORUM (https://emuelec.org) OR USE DISCORD FOR THAT (https://discord.gg/cbgtJTu), THIS IS ONLY FOR BUG/ISSUES REPORTING!

# DELETE EVERYTHING ABOVE THIS LINE (INCLUDING THIS LINE) BEFORE SUBMITTING TO MAKE SURE YOU'VE READ THIS!

The Bluetooth handle can work normally only when it is connected at power on
When the Bluetooth handle is connected again, it will not work, and the key will not work
It can be recognized as a handle in the emulec interface, but it cannot be recognized in the key mapping in retroarch

7Ji, 2022-09-05:
> # DELETE EVERYTHING ABOVE THIS LINE (INCLUDING THIS LINE) BEFORE SUBMITTING TO MAKE SURE YOU'VE READ THIS!
And you surely didn't read that at all, you need to delete things **ABOVE** that line and fill in the remaining infos:

>**Describe the bug**
>A clear and concise description of what the bug is.
>
>**To Reproduce**
>Steps to reproduce the behavior:
>1. Go to '...'
>2. Click on '....'
>3. Scroll down to '....'
>4. See error
>
>**Expected behavior**
>A clear and concise description of what you expected to happen.
>
>**Screenshots**
>If applicable, add screenshots to help explain your problem.
>
>**EmuELEC version:**
> - Version [e.g. 22]
>
>**Device (please complete the following information):**
> - SOC: [e.g. S095x3]
> - Brand: [e.g. Beelink]
> - Model [e.g. mini g1]
> - Version [e.g. 22]
>
>**Additional context**
>Add any other context about the problem here.

And please, even you can't follow the template, at least describe your issues clearly
> Bluetooth handle

What do you even mean by that? Bluetooth game controller? 

What device you're using? Don't tell me you are using S905L3A/B boxes (E900V3C/D, CM131), they are **not** supported at all and the bluetooth not functioning properly is expected since we don't even have a dtb for it.

haha496744, 2022-09-06:
> And you surely didn't read that at all, you need to delete things **ABOVE** that line and fill in the remaining infos:
>
>
> And please, even you can't follow the template, at least describe your issues [...]

Sorry for that, I've reverted to version 4.5    。 big brother ， are  you  chinese？

closed 2022-09-06 by shantigilbert