EmuELEC/EmuELEC #987: About the Chinese content display error when playing Chinese games on the free2jme emulator

haha496744, 2022-09-06:
EmuELEC version:

Version [e.g. 4.5]

SOC: [e.g. S905x3]
Brand: [e.g. x96max+]
Model [e.g. x96max+]
Version [e.g. 4.5]
When using the free j2me emulator to play java Chinese games, Chinese content is replaced by small squares.
[image]

7Ji, 2022-09-07:
Someone reported the same issue about Chinese font displaying on freej2me's repo and he has managed to fix that but leaved no solution:
https://github.com/hex007/freej2me/issues/160
@25minutes
Sorry for any inconvinience, but since you said this is not a problem of free2me in that issue, could you elaborate on how to solve this on EmuELEC?

haha496744, 2022-09-07:
> Someone reported the same issue about Chinese font displaying on freej2me's repo and he has managed to fix that but leaved no solution: [hex007/freej2me#160](https://github.com/hex007/freej2me/issues/160) @25minutes Sorry for any inconvinience, but since you said this is not a problem of free2me in that issue, could you elaborate on how to solve this on EmuELEC?
@7Ji 
I think it may be related to the emuelec system font, because Chinese cannot be displayed normally on RA

asakous, 2022-09-12:
@haha496744 which game you are testing. give me the game link . maybe I can help.

haha496744, 2022-09-21:
> @haha496744 which game you are testing. give me the game link . maybe I can help.

@asakous  几乎所有的中文游戏都会出现这种问题，您有解决方式吗？

asakous, 2022-09-22:
> > @haha496744 which game you are testing. give me the game link . maybe I can help.
> 
> @asakous 几乎所有的中文游戏都会出现这种问题，您有解决方式吗？
@haha496744  我需要1個rom. 我網上找的中文遊戲都跑不了，但英文遊戲可以。只要跟我說哪個網站哪個中文遊戲名我會下載來試試。

haha496744, 2022-09-22:
> > > @haha496744 which game you are testing. give me the game link . maybe I can help.
> > 
> > @asakous 几乎所有的中文游戏都会出现这种问题，您有解决方式吗？
> @haha496744  我需要1個rom. 我網上找的中文遊戲都跑不了，但英文遊戲可以。只要跟我說哪個網站哪個中文遊戲名我會下載來試試。
> 

@asakous  事实上是中文名的游戏运行不了，改成英文名还是可以，问题是游戏内部的中文字体不能正常显示，等下我给你附件吧，你自己下载试试

haha496744, 2022-09-22:
@asakous 
[java.zip](https://github.com/EmuELEC/EmuELEC/files/9624874/java.zip)

asakous, 2022-09-23:
since EmuELEC only included liberation-fons . these fonts are  not suitable for chinese games. so unless some on introduce a chinese font package for EmuELEC . otherwise there no way to fix it. 
現在的EmuELE版本沒有中文字型所以你看不了中文字。除非有人上個PR放中文字型進4.6的版本，如 droidsans 。
底下的版本是我放了 DroidSansFallback.ttf 後的圖
[image]

asakous, 2022-09-23:
@haha496744 
another game image
[image]

7Ji, 2022-09-23:
These Chinese fonts should be free to be included
https://wiki.archlinux.org/title/Localization/Chinese
But as always CJK (Chinese, Japanese, Korean) fonts are very chunky and take a lot of disk space (avg. ~5MiB for a single font with one variant), and inclusion of a single font is not enough (at lease regular and bold variant of a font should come in pair). I don't think many non-CJK users would like to see the image "bloated" with multiple MiBs of fonts they'll never use...

haha496744, 2022-09-23:
@asakous @7Ji 
thank for help，java游戏好像不能存储

shantigilbert, 2022-09-28:
> These Chinese fonts should be free to be included https://wiki.archlinux.org/title/Localization/Chinese But as always CJK (Chinese, Japanese, Korean) fonts are very chunky and take a lot of disk [...]

Maybe we could find a way to call these fonts from the EEROMS partition if a user has them there? I am not 100% how these fonts work, but its just an idea.

wang80919, 2022-12-19:
> Maybe we could find a way to call these fonts from the EEROMS partition if a user has them there? I am not 100% how these fonts work, but its just an idea.

thanks, we need this.

> since EmuELEC only included liberation-fons . these fonts are not suitable for chinese games. so unless some on introduce a chinese font package for EmuELEC . otherwise there no way to fix it. 現在的EmuELE版本沒有中文字型所以你看不了中文字。除非有人上個PR放中文字型進4.6的版本，如 droidsans 。 底下的版本是我放了 DroidSansFallback.ttf 後的圖 [image]

How to complete this operation manually.

wang80919, 2022-12-19:
> Maybe we could find a way to call these fonts from the EEROMS partition if a user has them there? I am not 100% how these fonts work, but its just an idea.

thanks we need this.

w553491775, 2023-02-15:
So,do you know how to solve it?